carla-simulator/carla #8046: Vehicle Physics in Carla

AOOOOOA, 2024-08-09:
Hi,
I have a question regarding vehicle physics in CARLA. It's known that CARLA utilizes Nvidia PhysX in Unreal Engine 4 and Chaos in Unreal Engine 5 for physics simulation. My inquiry is: Do all vehicles provided in CARLA share a uniform physics and dynamics model, or do they have distinct physical properties based on their individual sizes and characteristics?

RodrigoVillarias, 2024-08-09:
Hello, yes, each vehicle depending on the size and model has different physical properties, right now we are trying to study the real models in the real models for differences in weight, traction, braking type, etc.. adapting it to chaos

closed 2024-08-16 by MarcelPiNacy-CVC

AOOOOOA, 2024-08-21:
> Hello, yes, each vehicle depending on the size and model has different physical properties, right now we are trying to study the real models in the real models for differences in weight, traction, braking type, etc.. adapting it to chaos

I'm interested in how different vehicle types are set up in CARLA using the PhysX model. I searched through the CARLA code but couldn't find the specific implementation in UE4 or the NVIDIA PhysX SDK. How is the configuration for different vehicle models achieved? Thanks.