carla-simulator/carla #8051: cant launch scenario runner

Turuu1124, 2024-08-13:
Hello,
OS: windows11
carla version: 0.9.15
python: 3.7 anaconda prompt

i am having a problem getting scenario runner working, i cant get it to open and it keeps giving me different errors as shown below, guided by this [thread](https://github.com/carla-simulator/carla/issues/7692) i tried setting PYTHONPATH as shown in the images. i feel like i have tried everything and nothing seems to be working, any help would be greatly appreciated.

[image]
[image]
[image]
[image]
[image]
[image]

PatrickPromitzer, 2024-08-19:
Hi,
you can try installing numpy version 1.22
```
path/to/python -m pip install numpy==1.22
```

Turuu1124, 2024-08-19:
thank you for your input, the issue has been fixed.

closed 2024-08-19 by Turuu1124