carla-simulator/carla #8052: Set traffic lights into customized map

SExpert12, 2024-08-14:
I have used this code given on Carla document to generate 30 vehicles and I am using customized map. How to make the vehicles to obey the traffic lights?
Do I need to use both traffic manager and traffic light configuration to follow traffic light for the multiple vehicles.

spawn_delay = 20
counter = spawn_delay

# Set max vehicles (set smaller for low hardward spec)
max_vehicles = 200
# Alternate between spawn points
alt = False

spawn_points = world.get_map().get_spawn_points()
while True:
    world.tick()

    n_vehicles = len(world.get_actors().filter('*vehicle*'))
    vehicle_bp = random.choice(blueprints)

    # Spawn vehicle only after delay
    if counter == spawn_delay and n_vehicles < max_vehicles:
        # Alternate spawn points
        if alt:
            vehicle = world.try_spawn_actor(vehicle_bp, spawn_point_1)
        else:
            vehicle = world.try_spawn_actor(vehicle_bp, spawn_point_2)

        if vehicle: # IF vehicle is succesfully spawned
            vehicle.set_autopilot(True) # Give TM control over vehicle

            # Set parameters of TM vehicle control, we don't want lane changes
            traffic_manager.update_vehicle_lights(vehicle, True)
            traffic_manager.random_left_lanechange_percentage(vehicle, 0)
            traffic_manager.random_right_lanechange_percentage(vehicle, 0)
            traffic_manager.auto_lane_change(vehicle, False)

            # Alternate between routes
            if alt:
                traffic_manager.set_path(vehicle, route_1)
                alt = False
            else:
                traffic_manager.set_path(vehicle, route_2)
                alt = True

            vehicle = None

        counter -= 1
    elif counter > 0:
        counter -= 1
    elif counter == 0:
        counter = spawn_delay

Thanks

SExpert12, 2024-08-18:
I checked again all issues. and working fine for 1 traffic lights but not working for all traffic lights at the intersection area. How to install traffic lights in intersection area and make it work for all four approach?

Anyone can help please?

Thanks

DnDare, 2024-08-23:
Hi, 
Have you added in the BP_TrafficLightGroup all the BP_TrafficLights  you want to control?
This can be added at the editor, 

[image]

SExpert12, 2024-08-28:
Yes. I did that. I want to allow both left turn and straight-going vehicles to cross the intersection simultaneously, but the left turn vehicles stays in the intersection area. That is because the only one light is green in the traffic group. So how to allow the two movements to clear the intersection simultaneously. 

Thanks for help.

DnDare, 2024-09-06:
I think that it is not possible to have simultaneous movements from different directions/states with one traffic light controller, at this moment. I'll try to get a confirmation.

I don't think it is the approach you want, but maybe, adding a second traffic light controller at the same place, you might be able to synchronize them , if they both have the same number of directions/states  somehow.
This would be similar to this [issue](https://github.com/carla-simulator/carla/issues/7963) you already now.

SExpert12, 2024-09-09:
Hi,
What I want is once I allowed the vehicles to enter in the intersection area from approach number 1, it can exit from any three approach except the one from which enters which is in this case approach number 1.
Are you suggesting me to add two traffic light group if I need the vehicles to exit from any two approaches and three traffic light group if vehicles exit from any three approaches?

I hope I am able to explain clearly what I want.

Thanks.

DnDare, 2024-09-13:
Hi, yes I think you explained clearly,

And yes that's what I'd try, to duplicate the traffic light group, the as many times you want to synchronize different directions/approaches.

I'm not sure if it will desynchronize in long simulations.

And we are not sure if there are alternatives, we should review the trafficLights Blueprints in depth. Maybe we will at some point in the future,

SExpert12, 2024-09-17:
Okay. Thanks

closed 2024-09-17 by SExpert12