carla-simulator/carla #8056: make import failed

ShahzebTUMBachelor, 2024-08-17:
CARLA version: 0.9.15
Platform/OS: Ubuntu 22.04
Problem you have experienced: make import failed

I tried to import a map (.fbx and .xodr file) created in RoadRunner with the make import command but it failed. I am getting this error:

BuildPythonAPI.sh: Success!
/media/shahzeb/Volume/carla/Util/BuildTools/Import.py:353: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if key is not 'tiles':
/media/shahzeb/Volume/carla/Util/BuildTools/Import.sh: line 52: 14394 Segmentation fault      (core dumped) /usr/bin/env python${PY_VERSION_LIST[0]} ${CARLA_BUILD_TOOLS_FOLDER}/Import.py ${ARGS}
make: *** [Util/BuildTools/Linux.mk:14: import] Error 139
My python versions:
python --version
Python 2.7.18

python3 --version
Python 3.10.12

PatrickPromitzer, 2024-08-19:
Hi,
open the file
```
path/to/Carla/Util/BuildTools/Import.py
```
and change the ```is not```to ```!=``` in line 353

After that, it should work.

```python
if key is not 'tiles':
```
to

```python
if key != 'tiles':
```

ShahzebTUMBachelor, 2024-08-27:
@PatrickPromitzer your proposal did not work for me. I still get the same error. 
/media/shahzeb/Volume/carla/Util/BuildTools/Import.sh: line 53:  8345 Segmentation fault      (core dumped) /usr/bin/env python${PY_VERSION_LIST[0]} ${CARLA_BUILD_TOOLS_FOLDER}/Import.py ${ARGS}
make: *** [Util/BuildTools/Linux.mk:14: import] Error 139

However it works if I run import.py directly with python. I am really wondering what the problem is. I am quite sure it has to do something with the import.sh file as this /usr/bin/env python${PY_VERSION_LIST[0]} ${CARLA_BUILD_TOOLS_FOLDER}/Import.py ${ARGS} is the last codeline in the file

PatrickPromitzer, 2024-08-28:
Hi,
I didn't noticed it at the first time, but your "python" is version 2.7 while your "python3" is version 3.10

My guess is, that the sh scrip may have a problem with python 2.7, but even then it is strange that it works if you call the script directly.

Do you use "python Import.py", "python3 Import.py", or a different command to call the script to make it work for you?

ShahzebTUMBachelor, 2024-09-03:
I use python3.8 Import.py. It does not work with python3.10

PatrickPromitzer, 2024-09-04:
The Import.sh script is using python3

```
line 22
PY_VERSION_LIST=3

line 51
/usr/bin/env python${PY_VERSION_LIST[0]} ${CARLA_BUILD_TOOLS_FOLDER}/Import.py ${ARGS}
```
It I fill out the line 51 without variables, it would look like this
```
/usr/bin/env python3 path/to/carla/Util/BuildTools/Import.py ${ARGS}
```

if you want to fix that issue, you can edit line 22 to use "3.8" instead of "3"