carla-simulator/carla #8069: Correct Way to Modify Building Blueprints

GoodarzMehr, 2024-08-22:
CARLA version: 0.9.15
OS/Platform: Ubuntu 22.04

Hi!

I've been trying to modify the building blueprints used in Town12 so I can make them light up at night like the ones in Town13 and Town15 (see #8057), and I have added the `Static/Building/LargeMap` directory to the list of directories to cook in the project setting. However, when I use `make package` the packaged `LargeMap` folder is empty. Am I missing something?

BTW what I've been trying to do to get the buildings to light up is to add a `StaticMesh` variable inside the event graph of the blueprints, similar to `BP_Building`. Is that how it has to be done?

Thanks as always for your help!

AntonioAerosim, 2024-08-30:
Hi. The important things that need to be done in order for the buildings to light up are:

-Building actor/blueprint has the component "BP_Lights" and its type set to "building" along a intensity value.
-The name of the float parameter value that controls the emissive lighting needs to be called "EmissiveIntensity" in the shader.

Let me know if that helped.

GoodarzMehr, 2024-08-30:
Thanks for the info! I got the lights working by adding "BP_Lights" to the "BP_RandomBuilding" parent class and defining its event graph as shown below. This seems to be working since the interior material instance already has the "EmissiveIntensity" field, though seemingly not active.

[image]
[image]

My understanding is that the color of the building lights cannot be changed through the PythonAPI because the "Emissive Color" in the material instance overrides the light color of "BP_Lights" that's set through the PythonAPI. Is that correct?

AntonioCarla, 2024-08-30:
Yes, that would be correct.