carla-simulator/carla #8073: Make issues

hotmedal, 2024-08-23:
I have previously built Carla successfully on this machine and now I'm doing it again. Not sure what's going wrong, any help is appreciated. 

CARLA version: 0.9.15
Platform/OS: Ubuntu 22.04.4 LTS
Problem you have experienced: Error during make PythonAPI and make launch

```
~/carla$ make PythonAPI

Util/BuildTools/Linux.mk:163: warning: overriding recipe for target 'downloadplugins'
Util/BuildTools/Linux.mk:142: warning: ignoring old recipe for target 'downloadplugins'
Setup.sh: boost-1.80.0-c10 already installed.
Setup.sh: rpclib-v2.2.1_c5-c10 already installed.
Setup.sh: gtest-1.8.1-c10 already installed.
Setup.sh: recast-c10 already installed.
Setup.sh: Libpng already installed.
Setup.sh: Xerces-c already installed.
Setup.sh: Eigen already installed.
Setup.sh: Sqlite already installed.
Setup.sh: PROJ already installed.
Setup.sh: Patchelf already installed.
Setup.sh: CARLA version 0.9.15-230-g1ef3f55c9.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Client.Release" configuration.
[1/1] Linking CXX executable LibCarla/cmake/test/libcarla_test_client_release
FAILED: LibCarla/cmake/test/libcarla_test_client_release 
: && /home/ta/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++ -std=c++14 -pthread -fPIC -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -std=c++14 -pthread -fPIC -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -Wl,-O2 -Wl,--sort-common -Wl,--as-needed -Wl,-z,relro -Wl,-z,now -Wl,--disable-new-dtags -Wl,--gc-sections -Wl,--allow-shlib-undefined -Wl,-rpath,/home/ta/miniconda3/lib -Wl,-rpath-link,/home/ta/miniconda3/lib -L/home/ta/miniconda3/lib LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/carla/profiler/LifetimeProfiled.cpp.o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/carla/profiler/Profiler.cpp.o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/Buffer.cpp.o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/client/OpenDrive.cpp.o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/client/test_image.cpp.o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/client/test_opendrive.cpp.o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/client/test_recurrent_shared_future.cpp.o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/client/test_rpc.cpp.o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/common/test_buffer.cpp.o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/common/test_geom.cpp.o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/common/test_listview.cpp.o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/common/test_miscellaneous.cpp.o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/common/test_msgpack.cpp.o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/common/test_streaming.cpp.o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/common/test_vector3D.cpp.o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/test.cpp.o -o LibCarla/cmake/test/libcarla_test_client_release -L/home/ta/carla/Build/rpclib-v2.2.1_c5-c10-libstdcxx-install/lib   -L/home/ta/carla/Build/gtest-1.8.1-c10-libstdcxx-install/lib -Wl,-rpath,/home/ta/carla/Build/rpclib-v2.2.1_c5-c10-libstdcxx-install/lib:/home/ta/carla/Build/gtest-1.8.1-c10-libstdcxx-install/lib:  -lrpc  -lgtest_main  -lgtest  LibCarla/cmake/client/libcarla_client.a  /home/ta/carla/Build/boost-1.80.0-c10-install/lib/libboost_filesystem.a  /home/ta/carla/Build/recast-c10-install/lib/libRecast.a  /home/ta/carla/Build/recast-c10-install/lib/libDetour.a  /home/ta/carla/Build/recast-c10-install/lib/libDetourCrowd.a && :
/home/ta/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: warning: /usr/lib/gcc/x86_64-linux-gnu/12/../../../x86_64-linux-gnu/crt1.o: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0008002
/home/ta/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: warning: /lib/x86_64-linux-gnu/libm.so.6: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0008002
/home/ta/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: warning: /lib/x86_64-linux-gnu/libmvec.so.1: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0008002
/home/ta/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: warning: /lib/x86_64-linux-gnu/libc.so.6: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0008002
/home/ta/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/client/test_image.cpp.o: in function `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > testing::PrintToString<boost::gil::pixel<unsigned char, boost::gil::layout<boost::mp11::mp_list<boost::gil::red_t, boost::gil::green_t, boost::gil::blue_t, boost::gil::alpha_t>, boost::mp11::mp_list<std::integral_constant<int, 2>, std::integral_constant<int, 1>, std::integral_constant<int, 0>, std::integral_constant<int, 3> > > > >(boost::gil::pixel<unsigned char, boost::gil::layout<boost::mp11::mp_list<boost::gil::red_t, boost::gil::green_t, boost::gil::blue_t, boost::gil::alpha_t>, boost::mp11::mp_list<std::integral_constant<int, 2>, std::integral_constant<int, 1>, std::integral_constant<int, 0>, std::integral_constant<int, 3> > > > const&)':
test_image.cpp:(.text._ZN7testing13PrintToStringIN5boost3gil5pixelIhNS2_6layoutINS1_4mp117mp_listIJNS2_5red_tENS2_7green_tENS2_6blue_tENS2_7alpha_tEEEENS6_IJSt17integral_constantIiLi2EESC_IiLi1EESC_IiLi0EESC_IiLi3EEEEEEEEEEENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_[_ZN7testing13PrintToStringIN5boost3gil5pixelIhNS2_6layoutINS1_4mp117mp_listIJNS2_5red_tENS2_7green_tENS2_6blue_tENS2_7alpha_tEEEENS6_IJSt17integral_constantIiLi2EESC_IiLi1EESC_IiLi0EESC_IiLi3EEEEEEEEEEENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_]+0x2a): undefined reference to `testing::internal2::PrintBytesInObjectTo(unsigned char const*, unsigned long, std::ostream*)'
/home/ta/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/client/test_image.cpp.o: in function `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > testing::PrintToString<boost::gil::pixel<unsigned char, boost::gil::layout<boost::mp11::mp_list<boost::gil::red_t, boost::gil::green_t, boost::gil::blue_t, boost::gil::alpha_t>, boost::mp11::mp_list<std::integral_constant<int, 0>, std::integral_constant<int, 1>, std::integral_constant<int, 2>, std::integral_constant<int, 3> > > > >(boost::gil::pixel<unsigned char, boost::gil::layout<boost::mp11::mp_list<boost::gil::red_t, boost::gil::green_t, boost::gil::blue_t, boost::gil::alpha_t>, boost::mp11::mp_list<std::integral_constant<int, 0>, std::integral_constant<int, 1>, std::integral_constant<int, 2>, std::integral_constant<int, 3> > > > const&)':
test_image.cpp:(.text._ZN7testing13PrintToStringIN5boost3gil5pixelIhNS2_6layoutINS1_4mp117mp_listIJNS2_5red_tENS2_7green_tENS2_6blue_tENS2_7alpha_tEEEENS6_IJSt17integral_constantIiLi0EESC_IiLi1EESC_IiLi2EESC_IiLi3EEEEEEEEEEENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE [... cut ...]
test_msgpack.cpp:(.text._ZN7testing13PrintToStringIN5carla4geom11BoundingBoxEEENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_[_ZN7testing13PrintToStringIN5carla4geom11BoundingBoxEEENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_]+0x2a): undefined reference to `testing::internal2::PrintBytesInObjectTo(unsigned char const*, unsigned long, std::ostream*)'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
make: *** [Util/BuildTools/Linux.mk:129: LibCarla.client.release] Error 1
```

PatrickPromitzer, 2024-08-28:
Hi,
the error doesn't give much information because it is used for many different errors.

Something I noticed, that it helps to set the clang soft link again.
```
sudo update-alternatives --install /usr/bin/clang++ clang++ /usr/lib/llvm-8/bin/clang++ 180 &&
sudo update-alternatives --install /usr/bin/clang clang /usr/lib/llvm-8/bin/clang 180
```

hotmedal, 2024-08-28:
I set those and it didn't help. So I deleted everything and started from scratch and I was getting xerces errors. 
`error: no member named 'int8_t' in the global namespace
using::int8_t;`
So I copied the directories xerces-c-3.2.3-install and xerces-c-3.2.3-install-server from the old install and proceeded. However, now I am running into similar problems in the CMakeFiles part of the make. This obviously has to do with the compiler path or libraries path.
```
src/CMakeFiles/proj.dir/log.cpp.o 
/home/ta/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++ -DMUTEX_pthread -DPROJ_LIB=\"/home/ta/carla/Build/proj-install-server/share/proj\" -I/home/ta/carla/Build/proj-src/src -I/home/ta/carla/Build/proj-src/include -I/home/ta/carla/Build/proj-src/build/src -I/home/ta/carla/Build/sqlite-install/include -std=c++14 -fPIC -stdlib=libc++ -I/home/ta/UnrealEngine_4.26/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1 -Wl,-L/home/ta/UnrealEngine_4.26/Engine/Source/ThirdParty/Linux/LibCxx/lib/Linux/x86_64-unknown-linux-gnu -O3 -DNDEBUG -std=c++11 -fvisibility=hidden -Wall -Wextra -Wswitch -Wshadow -Wunused-parameter -Wmissing-declarations -Wformat -Wformat-security -Wfloat-conversion -MD -MT src/CMakeFiles/proj.dir/log.cpp.o -MF src/CMakeFiles/proj.dir/log.cpp.o.d -o src/CMakeFiles/proj.dir/log.cpp.o -c /home/ta/carla/Build/proj-src/src/log.cpp
clang++: warning: -Wl,-L/home/ta/UnrealEngine_4.26/Engine/Source/ThirdParty/Linux/LibCxx/lib/Linux/x86_64-unknown-linux-gnu: 'linker' input unused [-Wunused-command-line-argument]
In file included from /home/ta/carla/Build/proj-src/src/log.cpp:30:
In file included from /home/ta/UnrealEngine_4.26/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/stdlib.h:100:
/home/ta/UnrealEngine_4.26/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/math.h:773:12: error: no member named 'labs' in the global namespace; did you mean 'abs'?
  return ::labs(__x);
         ~~^
```

Is there a chunk of code I can run that will remove all the packages that can conflict with this and set the correct environment paths? Because I know I have clang, llvm, gcc, libstdc++, etc. And it looks like the process is using all the files from `/UnrealEngine_4.26/Engine/Source/ThirdParty/Linux/LibCxx/include` so I don't see why there should be any problem.

PatrickPromitzer, 2024-08-28:
I tried deleting everything and starting from scratch on an Ubuntu 22.04 system, and I could not get it working.
I don't know if something changed in Ubuntu or Carla, but I copied the Carla/UE4 files from another Ubuntu 20.04 system to the Ubuntu 22.04 System.

I had no time to look into the installation problem, but I know there is be problems with Ubuntu 22.04
I saw some people converting the code to use clang-10 or higher (clang-8 isn't supported on Ubuntu 22.04 anymore), and that makes it tricky to install Carla.
Ubuntu 20.04 should work, but I noticed that someone I help out has a bug installing Carla on a Carla clean system.

I can give some lines of my Carla source Dockerfile for Carla version 0.9.14
Maybe it helps you finding a solution.

Dockerfile lines;
```
RUN sed -i 's/VERSION_LIST=3/VERSION_LIST=3.8/g' /home/carla/carla/Util/BuildTools/Setup.sh

RUN sed -i 's;XERCESC_REPO=https://ftp.cixug.es/apache//xerces/c/3/sources/xerces-c-${XERCESC_VERSION}.tar.gz;XERCESC_REPO=https://archive.apache.org/dist/xerces/c/3/sources/xerces-c-${XERCESC_VERSION}.tar.gz;g' /home/carla/carla/Util/BuildTools/Setup.sh

RUN sed -i 's;wget -c "${FBXSDK_URL}" -P "${CARLA_DOCKER_UTILS_FOLDER}";wget -c "${FBXSDK_URL}" -P "${CARLA_DOCKER_UTILS_FOLDER}" --user-agent="Mozilla";g' /home/carla/carla/Util/BuildTools/BuildUtilsDocker.sh

RUN sed -i 's;wget "https://boostorg.jfrog.io/artifactory/main/release/${BOOST_VERSION}/source/${BOOST_PACKAGE_BASENAME}.tar.gz" || true;wget "https://carla-releases.s3.eu-west-3.amazonaws.com/Backup/${BOOST_PACKAGE_BASENAME}.tar.gz" || true;g' /home/carla/carla/Util/BuildTools/Setup.sh

RUN sed -i 's;FBXSDK_URL=https://www.autodesk.com/content/dam/autodesk/www/adn/fbx/2020-0-1/${LIB_NAME}.tar.gz;FBXSDK_URL=https://damassets.autodesk.net/content/dam/autodesk/www/adn/fbx/2020-0-1/${LIB_NAME}.tar.gz;g' /home/carla/carla/Util/BuildTools/BuildUtilsDocker.sh
```
In a cleaner view
```
OLD_STRING
NEW_STRING_REPLACING_THE_OLD_STRING
FILE

VERSION_LIST=3
VERSION_LIST=3.8
/home/carla/carla/Util/BuildTools/Setup.sh

XERCESC_REPO=https://ftp.cixug.es/apache//xerces/c/3/sources/xerces-c-${XERCESC_VERSION}.tar.gz
XERCESC_REPO=https://archive.apache.org/dist/xerces/c/3/sources/xerces-c-${XERCESC_VERSION}.tar.gz
/home/carla/carla/Util/BuildTools/Setup.sh

wget -c "${FBXSDK_URL}" -P "${CARLA_DOCKER_UTILS_FOLDER}"
wget -c "${FBXSDK_URL}" -P "${CARLA_DOCKER_UTILS_FOLDER}" --user-agent="Mozilla"
/home/carla/carla/Util/BuildTools/BuildUtilsDocker.sh

wget "https://boostorg.jfrog.io/artifactory/main/release/${BOOST_VERSION}/source/${BOOST_PACKAGE_BASENAME}.tar.gz" || true
wget "https://carla-releases.s3.eu-west-3.amazonaws.com/Backup/${BOOST_PACKAGE_BASENAME}.tar.gz" || true
/home/carla/carla/Util/BuildTools/Setup.sh

FBXSDK_URL=https://www.autodesk.com/content/dam/autodesk/www/adn/fbx/2020-0-1/${LIB_NAME}.tar.gz
FBXSDK_URL=https://damassets.autodesk.net/content/dam/autodesk/www/adn/fbx/2020-0-1/${LIB_NAME}.tar.gz
/home/carla/carla/Util/BuildTools/BuildUtilsDocker.sh
```

hotmedal, 2024-08-28:
Something changed in Carla I think, because I have installed and used it on this same Ubuntu before. I still have the old directory, except make launch no longer works. 
Thanks for your help, I'm not sure how to use it directly since I am not using docker.

PatrickPromitzer, 2024-08-29:
I read the error code again and saw it comes from the "proj" installation.
I have a something which could fix it.

pat/to/carla/Util/BuildTools/Setup.sh
```
  mkdir ${PROJ_SRC_DIR}/build
  mkdir ${PROJ_INSTALL_DIR}
  sed -i 's;#include <string>;#include <string>\n#include <stdint.h>\n#include <cstdint>;g' ${PROJ_SRC_DIR}/src/proj_json_streaming_writer.hpp
  sed -i 's;#include <string>;#include <string>\n#include <stdint.h>\n#include <cstdint>;;g' ${PROJ_SRC_DIR}/src/proj_json_streaming_writer.cpp

  pushd ${PROJ_SRC_DIR}/build >/dev/null
```
it should be around line 630

I changed the PROJ version, too, but you can try that if the first solution doesn't work
```
PROJ_VERSION=proj-9.3.0
```
around line 607

jasonlee539, 2024-11-04:
Have u ever solved this problem?I am confused for a long time make we can help each other solve this
.