carla-simulator/carla #8075: Incorrect Vector Rotation Calculation in  PythonAPI agents.tools.misc

ZiyuZ, 2024-08-24:
CARLA version: 0.9.15
Platform/OS: Python Code
Problem you have experienced: Incorrect Vector Rotation Calculation

I found an issue in the `carla/agents/tools/misc.py` file regarding the calculation of rotated vectors. The [relevant code](https://github.com/carla-simulator/carla/blob/1ef3f55c9555de401681cb26ce87f81718943624/PythonAPI/carla/agents/tools/misc.py#L51) is as follows:

```python
rotated_x = math.cos(radians) * x - math.sin(radians) * y
rotated_y = math.sin(radians) * x - math.cos(radians) * y
```

The calculation for `rotated_y` should be `rotated_y = math.sin(radians) * x + math.cos(radians) * y`

ZiyuZ, 2024-08-24:
btw, in [Line 60](https://github.com/carla-simulator/carla/blob/1ef3f55c9555de401681cb26ce87f81718943624/PythonAPI/carla/agents/tools/misc.py#L60)

```python
point = rotate_point(carla.Vector3D(0, 0, area_ext.z), math.radians(base_rot))
```

This seems to do nothing because both x and y are 0, which means the rotation does not affect the point's position.

M-Colley, 2026-10-26:
Draft PR #10001 includes the `ue58-dev` rotated-vector calculation fix described here, with offline geometry regressions. Live agent/simulator validation remains unrun.