carla-simulator/carla #8076: Another way to control vehicle movement

Ikaros520520, 2024-08-25:
Hello, everyone.
I want to set up vehicle tracks with UE. In this trajectory, control the vehicle to accelerate, decelerate and stop.
I was previously using python code to control the vehicle. But when the code controls the vehicle, the speed of the vehicle can't be precisely controlled (because in python the vehicle's throttle is used to control the speed). So I want to abandon python and use UE.
Does anyone know what I should do or have a similar example for my reference?

CARLA version:CarlaUE4.26.2
Platform/OS:Windows

MattRoweEAIF, 2024-08-30:
Hi, it's not entirely clear what you need to do. If you increase throttle, the vehicle will accelerate, if you decrease throttle or apply brake, the vehicle will decelerate and eventually stop, this is all possible through the VehicleControl API.

The throttle is a higher level control of the physics engine, if you wish to bypass this, you will have to start to dive into the vehicle blueprint and the physics engine itself. Alternatively you could use set_target_velocity(), which will accelerate the vehicle until it reaches the target velocity.

You could also turn the physics engine off altogether and teleport the vehicle directly. You can disable physics through the set_simulate_physics() method of the Actor class. Then you can use set_transform() to teleport the vehicle to any location and pose you wish. This will however mean that the vehicle's wheels will not turn, its suspension will not work.

Ikaros520520, 2024-09-02:
> Hi, it's not entirely clear what you need to do. If you increase throttle, the vehicle will accelerate, if you decrease throttle or apply brake, the vehicle will decelerate and eventually stop, this [...]

Thank you for your advice. I'll give it a try.

Ikaros520520, 2024-09-02:
Thank you for your advice. I'll give it a try.

closed 2024-09-13 by MarcelPiNacy-CVC