carla-simulator/carla #8084: Low Client FPS while using logitech steering wheel

Yanchen30247, 2024-08-27:
I am trying to munual control car using Logitech G pro steering wheel, I used joystick to read my steering input and it works, but the client Fps becomes really low(around 10Fps). Even more, the client will randomly not respond. On the opposite everything just works well with keyboard input. If anyone has faced any similar problems?
[image]

Yanchen30247, 2024-08-27:
` 
joystick = pygame.joystick.Joystick(0)
                steer = joystick.get_axis(0)
                gas = joystick.get_axis(3)
                brake = joystick.get_axis(4)
                pygame.time.wait(10)  

def _parse_vehicle_steer(self, steer, gas, brake):
        if not self._ackermann_enabled:
            if gas != 0.0:
                self._control.throttle = max(0, 0.5+gas/2)
            else:
                self._control.throttle = 0

        if not self._ackermann_enabled:
            self._control.brake = max(0, brake)

        if not self._ackermann_enabled:
            self._control.steer = steer
`
I add some code base on official manual control example, code above is my code about steering wheel control

PatrickPromitzer, 2024-09-04:
Hi
the examples folder includes an example for the Logitech G29 steering wheel.
https://github.com/carla-simulator/carla/blob/0.9.15.2/PythonAPI/examples/manual_control_steeringwheel.py

To be able to use this example, you need an "wheel_config.ini file with the following content
```
[G29 Racing Wheel]
steering_wheel = 0
clutch = 1
throttle = 2
brake = 3
handbrake = 4
reverse = 5
```

Maybe it works if you edit the .ini file to your needs.