carla-simulator/carla #8089: Controls Carla from different projects in UE

ADRN1, 2024-08-29:
Hi, 
I try to control Carla's simulation from different projects in UE. To do that I have to include Carla Libs in this project and create C++ objects to connect and control, but.... I have a problem with include.

I probably have to add some names to my file "[project].Build.cs" to "PublicDependencyModuleNames.AddRange", similar to "Boost" file.
Could you send me the name of these carlaLibs in C++ or a different way that I can include these libs?

MarcelPiNacy-CVC, 2024-08-30:
Hi,

in which version of CARLA are you? Is this UE4 CARLA or the UE5 branch?

Marcel

ADRN1, 2024-08-30:
trying on UE5 branch.
version "0.10.0"
Win 11

MarcelPiNacy-CVC, 2024-09-09:
Hi again,

We propagate the dependency libs to UE using some .inc files (Includes, Libraries & Options.inc). These can be found in the unreal build folder, that is Build/Unreal.