carla-simulator/carla #8091: Build a package with CARLA UE5 Error!!

qq118373118, 2024-08-30:
CARLA version:0.9.15
Platform/OS:WIN10
Problem you have experienced:
What you expected to happen:When building a package, an error message is always displayed.
There is a lot of repetitive information

``
LogShaderCompilers: Warning: Last Error 206: Failed to create directory for shader debug info 'E:/UGit/Carla/Unreal/CarlaUnreal/Saved/ShaderDebugInfo/PCD3D_ES3_1/M_Freeway_Asphalt_28b092c24f487deb/Default/FInstancedStaticMeshVertexFactory/TMobileBasePassPSFMobileDirectionalLightCSMAndSHIndirectPolicyHDRLinear64SkylightLOCAL_LIGHTS_DISABLED/0'. Try enabling large file paths or r.DumpShaderDebugShortNames.
LogCook: Display: Cooked packages 8002 Packages Remain 434 Total 8436
LogShaderCompilers: Warning: Last Error 206: Failed to create directory for shader debug info 'E:/UGit/Carla/Unreal/CarlaUnreal/Saved/ShaderDebugInfo/PCD3D_ES3_1/M_Fabric_Master_9b3e027c5b14630a/Default/FInstancedStaticMeshVertexFactory/TMobileBasePassPSFMobileDistanceFieldShadowsLightMapAndCSMLightingPolicyHDRLinear64SkylightLOCAL_LIGHTS_DISABLED/0'. Try enabling large file paths or r.DumpShaderDebugShortNames.
LogShaderCompilers: Warning: Last Error 206: Failed to create directory for shader debug info 'E:/UGit/Carla/Unreal/CarlaUnreal/Saved/ShaderDebugInfo/PCD3D_ES3_1/M_Freeway_Asphalt_28b092c24f487deb/Default/FInstancedStaticMeshVertexFactory/TMobileBasePassPSFMobileDirectionalLightAndSHIndirectPolicyHDRLinear64SkylightLOCAL_LIGHTS_PREPASS_ENABLED/0'. Try enabling large file paths or r.DumpShaderDebugShortNames.
LogShaderCompilers: Warning: Last Error 206: Failed to create directory for shader debug info 'E:/UGit/Carla/Unreal/CarlaUnreal/Saved/ShaderDebugInfo/PCD3D_ES3_1/M_Fabric_Master_9b3e027c5b14630a/Default/FInstancedStaticMeshVertexFactory/TMobileBasePassPSFMobileDistanceFieldShadowsLightMapAndCSMLightingPolicyHDRLinear64SkylightLOCAL_LIGHTS_ENABLED/0'. Try enabling large file paths or r.DumpShaderDebugShortNames.
LogShaderCompilers: Warning: Last Error 206: Failed to create directory for shader debug info 'E:/UGit/Carla/Unreal/CarlaUnreal/Saved/ShaderDebugInfo/PCD3D_ES3_1/M_Fabric_Master_17b68316f6fea10a/Default/FInstancedStaticMeshVertexFactory/TMobileBasePassPSFMobileMovableDirectionalLightCSMWithLightmapPolicyHDRLinear64SkylightLOCAL_LIGHTS_DISABLED/0'. Try enabling large file paths or r.DumpShaderDebugShortNames.
LogShaderCompilers: Warning: Last Error 206: Failed to create directory for shader debug info 'E:/UGit/Carla/Unreal/CarlaUnreal/Saved/ShaderDebugInfo/PCD3D_ES3_1/M_Fabric_Master_9b3e027c5b14630a/Default/FInstancedStaticMeshVertexFactory/TMobileBasePassPSFMobileDistanceFieldShadowsLightMapAndCSMLightingPolicyHDRLinear64LOCAL_LIGHTS_DISABLED/0'. Try enabling large file paths or r.DumpShaderDebugShortNames.
LogShaderCompilers: Warning: Last Error 206: Failed to create directory for shader debug info 'E:/UGit/Carla/Unreal/CarlaUnreal/Saved/ShaderDebugInfo/PCD3D_ES3_1/M_Fabric_Master_9b3e027c5b14630a/Default/FLocalVertexFactory/TMobileBasePassPSFMobileDistanceFieldShadowsLightMapAndCSMLightingPolicyHDRLinear64SkylightLOCAL_LIGHTS_PREPASS_ENABLED/0'. Try enabling large file paths or r.DumpShaderDebugShortNames.
LogShaderCompilers: Warning: Last Error 206: Failed to create directory for shader debug info 'E:/UGit/Carla/Unreal/CarlaUnreal/Saved/ShaderDebugInfo/PCD3D_ES3_1/M_Fabric_Master_17b68316f6fea10a/Default/FInstancedStaticMeshVertexFactory/TMobileBasePassPSFMobileDistanceFieldShadowsLightMapAndCSMLightingPolicyHDRLinear64LOCAL_LIGHTS_DISABLED/0'. Try enabling large file paths or r.DumpShaderDebugShortNames.
LogShaderCompilers: Warning: Last Error 206: Failed to create directory for shader debug info 'E:/UGit/Carla/Unreal/CarlaUnreal/Saved/ShaderDebugInfo/PCD3D_ES3_1/M_Fabric_Master_17b68316f6fea10a/Default/FInstancedStaticMeshVertexFactory/TMobileBasePassPSFMobileMovableDirectionalLightCSMWithLightmapPolicyHDRLinear64SkylightLOCAL_LIGHTS_ENABLED/0'. Try enabling large file paths or r.DumpShaderDebugShortNames.
LogShaderCompilers: Warning: Last Error 206: Failed to create directory for shader debug info 'E:/UGit/Carla/Unreal/CarlaUnreal/Saved/ShaderDebugInfo/PCD3D_ES3_1/M_Fabric_Master_17b68316f6fea10a/Default/FInstancedStaticMeshVertexFactory/TMobileBasePassPSFMobileMovableDirectionalLightCSMWithLightmapPolicyHDRLinear64LOCAL_LIGHTS_PREPASS_ENABLED/0'. Try enabling large file paths or r.DumpShaderDebugShortNames.
LogCook: Display: Cooked packages 8002 Packages Remain 434 Total 8436
LogShaderCompilers: Warning: Last Error 206: Failed to create directory for shader debug info 'E:/UGit/Carla/Unreal/CarlaUnreal/Saved/ShaderDebugInfo/PCD3D_ES3_1/M_Freeway_Asphalt_28b092c24f487deb/Default/FGeometryCollectionVertexFactory/TMobileBasePassPSFMobileDirectionalLightAndSHIndirectPolicyHDRLinear64SkylightLOCAL_LIGHTS_PREPASS_ENABLED/0'. Try enabling large file paths or r.DumpShaderDebugShortNames.
LogShaderCompilers: Warning: Last Error 206: Failed to create directory for shader debug info 'E:/UGit/Carla/Unreal/CarlaUnreal/Saved/ShaderDebugInfo/PCD3D_ES3_1/M_Fabric_Master_9b3e027c5b14630a/Default/FInstancedStaticMeshVertexFactory/TMobileBasePassPSFMobileDistanceFieldShadowsLightMapAndCSMLightingPolicyHDRLinear64LOCAL_LIGHTS_PREPASS_ENABLED/0'. Try enabling large file paths or r.DumpShaderDebugShortNames.
LogShaderCompilers: Warning: Last Error 206: Failed to create directory for shader debug info 'E:/UGit/Carla/Unreal/CarlaUnreal/Saved/ShaderDebugInfo/PCD3D_ES3_1/M_Fabric_Master_17b68316f6fea10a/Default/FInstancedStaticMeshVertexFactory/TMobileBasePassPSFMobileMovableDirectionalLightCSMWithLightmapPolicyHDRLinear64SkylightLOCAL_LIGHTS_PREPASS_ENABLED/0'. Try enabling large file paths or r.DumpShaderDebugShortNames.
LogShaderCompilers: Warning: Last Error 206: Failed to create directory for shader debug info 'E:/UGit/Carla/Unreal/CarlaUnreal/Saved/ShaderDebugInfo/PCD3D_ES3_1/M_Fabric_Master_17b68316f6fea10a/Default/FInstancedStaticMeshVertexFactory/TMobileBasePassPSFMobileDirectionalLightCSMAndLightMapPolicyHDRLinear64SkylightLOCAL_LIGHTS_PREPASS_ENABLED/0'. Try enabling large file paths or r.DumpShaderDebugShortNames.
LogShaderCompilers: Warning: Last Error 206: Failed to create directory for shader debug info 'E:/UGit/Carla/Unreal/CarlaUnreal/Saved/ShaderDebugInfo/PCD3D_ES3_1/M_Fabric_Master_17b68316f6fea10a/Default/FInstancedStaticMeshVertexFactory/TMobileBasePassPSFMobileDistanceFieldShadowsAndLQLightMapPolicyHDRLinear64SkylightLOCAL_LIGHTS_PREPASS_ENABLED/0'. Try enabling large file paths or r.DumpShaderDebugShortNames.
LogShaderCompilers: Warning: Last Error 206: Failed to create directory for shader debug info 'E:/UGit/Carla/Unreal/CarlaUnreal/Saved/ShaderDebugInfo/PCD3D_ES3_1/M_Fabric_Master_17b68316f6fea10a/Default/FInstancedStaticMeshVertexFactory/TMobileBasePassPSFMobileDistanceFieldShadowsLightMapAndCSMLightingPolicyHDRLinear64LOCAL_LIGHTS_PREPASS_ENABLED/0'. Try enabling large file paths or r.DumpShaderDebugShortNames.
LogCook: Display: Cooker has been blocked from saving the current packages for 1201 seconds.
LogCook: Display: 434 packages in the savequeue:
LogCook: Display:     /Engine/EngineMaterials/DefaultTextMaterialOpaque
LogCook: Display:     /Game/Carla/Static/Car/GeneralMaterials/MaterialRedirectors/DummyCar
LogCook: Display:     /Game/Carla/Static/GenericMaterials/Masters/M_BasicMaterial
LogCook: Display:     /Game/Carla/Static/GenericMaterials/Vehicles/M_CarLightsGlass_Master
LogCook: Display:     /Game/Carla/Static/Static/Materials/PlantPot03/M_PlantPot03
LogCook: Display:     /Game/Carla/Static/Particles/Leaves/M_Leaf_falling
LogCook: Display:     /Game/Carla/Static/Dynamic/PedestrianProps/Materials/Phone/M_Mobile
LogCook: Display:     /Game/Carla/Static/Dynamic/Trash/Materials/GarbageAtlas/M_GarbageAtlas
LogCook: Display:     /Game/Carla/Static/Car/4Wheeled/Jeep/Textures_Shadders/JeepWranglerRubicon_White_Mat
LogCook: Display:     /Game/Carla/Static/GenericMaterials/Vehicles/M_CarInteriorAlpha_Master
LogCook: Display:     ...
LogCook: Display: 177 objects that have not yet returned true from IsCachedCookedPlatformDataLoaded:
LogCook: Display:     Material /Engine/EngineMaterials/DefaultTextMaterialOpaque.DefaultTextMaterialOpaque
LogCook: Display:     Material /Game/Carla/Static/Car/GeneralMaterials/MaterialRedirectors/DummyCar.DummyCar
LogCook: Display:     Material /Game/Carla/Static/GenericMaterials/Masters/M_BasicMaterial.M_BasicMaterial
[... cut ...] LIGHTS_DISABLED/0'. Try enabling large file paths or r.DumpShaderDebugShortNames.
LogShaderCompilers: Warning: Last Error 206: Failed to create directory for shader debug info 'E:/UGit/Carla/Unreal/CarlaUnreal/Saved/ShaderDebugInfo/PCD3D_ES3_1/M_Fabric_Master_9b3e027c5b14630a/Default/FInstancedStaticMeshVertexFactory/TMobileBasePassPSFMobileDistanceFieldShadowsLightMapAndCSMLightingPolicyHDRLinear64SkylightLOCAL_LIGHTS_DISABLED/0'. Try enabling large file paths or r.DumpShaderDebugShortNames.
``

jorge-kabuto, 2024-08-30:
Reading the error code, this seems to be a problem with long file paths, exceeding the maximum supported by your system.

lt-wang, 2024-09-04:
hello, I also have the same problem. Did you solve it?

qq1183713118, 2024-09-05:
> 你好，我也有同样的问题。您解决了吗？

> hello, I also have the same problem. Did you solve it?

No, I ignored it because it didn't affect the operation.

qq1183713118, 2024-09-13:
Thanks for your answers. I didn't compile Unreal on WSL, but used Linux instead. Thank you very much for your help.

closed 2024-09-13 by qq1183713118