carla-simulator/carla #8092: make osmrenderer  error in Linux

bg-sumu, 2024-08-30:
Linux 22.04
Carla 0.9.15
make osmrenderer 
Util/BuildTools/Linux.mk:163: warning: overriding recipe for target 'downloadplugins'
Util/BuildTools/Linux.mk:142: warning: ignoring old recipe for target 'downloadplugins'
Cloning libosmscout.
/home/sumukhabg/carla/Build
HEAD is now at e83e4881a Issue #1045 - Update projection
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (MARISA)
  does not match the name of the calling package (Marisa).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/FindMarisa.cmake:40 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  cmake/features.cmake:143 (find_package)
  CMakeLists.txt:51 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (ICONV) does
  not match the name of the calling package (iconv).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/Findiconv.cmake:59 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  cmake/features.cmake:164 (find_package)
  CMakeLists.txt:51 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (LIBLZMA)
  does not match the name of the calling package (LibLZMA).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/FindLibLZMA.cmake:22 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  cmake/features.cmake:165 (find_package)
  CMakeLists.txt:51 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Checking for module 'cairo'
--   No package 'cairo' found
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (LIBAGG)
  does not match the name of the calling package (Agg).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/FindAgg.cmake:38 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  cmake/features.cmake:175 (find_package)
  CMakeLists.txt:51 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find LIBAGG (missing: LIBAGG_INCLUDE_DIRS LIBAGG_LIBRARIES) 
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (Pango) does
  not match the name of the calling package (PANGO).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/FindPANGO.cmake:52 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  cmake/features.cmake:177 (find_package)
  CMakeLists.txt:51 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find Pango (missing: PANGO_INCLUDE_DIRS PANGO_LIBRARIES) 
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (HARFBUZZ)
  does not match the name of the calling package (HarfBuzz).  This can lead
[... 84 lines cut ...]
--  - MATLAB mex function:          OFF
-- doxygen:                         NO
-- documentation:                   OFF
-- webpage:                         OFF
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sumukhabg/carla/Build/libosmscout-build
make[1]: Entering directory '/home/sumukhabg/carla/Build/libosmscout-build'
make[2]: Entering directory '/home/sumukhabg/carla/Build/libosmscout-build'
make[3]: Entering directory '/home/sumukhabg/carla/Build/libosmscout-build'
Consolidate compiler generated dependencies of target OSMScout
make[3]: Leaving directory '/home/sumukhabg/carla/Build/libosmscout-build'
make[3]: Entering directory '/home/sumukhabg/carla/Build/libosmscout-build'
[  0%] Building CXX object libosmscout/CMakeFiles/OSMScout.dir/src/osmscout/util/File.cpp.o
/home/sumukhabg/carla/Build/libosmscout-source/libosmscout/src/osmscout/util/File.cpp:23:10: fatal error: filesystem: No such file or directory
 #include <filesystem>
          ^~~~~~~~~~~~
compilation terminated.
make[3]: *** [libosmscout/CMakeFiles/OSMScout.dir/build.make:202: libosmscout/CMakeFiles/OSMScout.dir/src/osmscout/util/File.cpp.o] Error 1
make[3]: Leaving directory '/home/sumukhabg/carla/Build/libosmscout-build'
make[2]: *** [CMakeFiles/Makefile2:430: libosmscout/CMakeFiles/OSMScout.dir/all] Error 2
make[2]: Leaving directory '/home/sumukhabg/carla/Build/libosmscout-build'
make[1]: *** [Makefile:146: all] Error 2
make[1]: Leaving directory '/home/sumukhabg/carla/Build/libosmscout-build'
make: *** [Util/BuildTools/Linux.mk:160: osmrenderer] Error 2

MarcelPiNacy-CVC, 2024-08-30:
Hi,

Could you share the entire build log, ideally as an attached file?
Also, did you run any commands before `make osmrenderer`, such as `make setup` or `make LibCarla`?

bg-sumu, 2024-08-30:
Hey I Reinstalled Carla from Scratch 
These are the set of commands i used 
1. `git clone carla_url`
2. `cd carla` and `./Update.sh`
3.` make PythonAPI`
4. `make launch`

then i ran `make osmrenderer`

bg-sumu, 2024-09-03:
@MarcelPiNacy-CVC any updates ?

AlexanderRex, 2024-10-01:
@bg-sumu I used this

```
sudo update-alternatives --config g++
There are 2 choices for the alternative g++ (providing /usr/bin/g++).

  Selection    Path             Priority   Status
------------------------------------------------------------
* 0            /usr/bin/g++-7    180       auto mode
  1            /usr/bin/g++-12   100       manual mode
  2            /usr/bin/g++-7    180       manual mode

Press <enter> to keep the current choice[*], or type selection number: 1
update-alternatives: using /usr/bin/g++-12 to provide /usr/bin/g++ (g++) in manual mode
```

Just update g++ version and manually set it to 12