carla-simulator/carla #8094: Error of "make PythonAPI "

SongJiang32, 2024-09-01:
Hello, everyone.I am a student when i install carla I have a problem that is "make python".
    My OS is Ubuntu20.04  and ROS Noetic, I create a python3.7 enviroment named carla by anaconda.When I run "make PythonAPI" after successfully executing "./Update.sh", I got the error as follow:

(carla) jason32@jason32:~/carla$ make PythonAPI
Util/BuildTools/Linux. Mk: 163: warning: cover on target "downloadplugins" formula
Util/BuildTools/Linux. Mk: 142: warning: ignore about the target "downloadplugins" old formula
Setup.sh: boost-1.80.0-c10 already installed.
Setup.sh: Retrieving rpclib.
Cloning to 'rpclib-v2.2.1_c5-c10-source'...
remote: Enumerating objects: 7381, done.
remote: Counting objects: 100% (119/119), done.
remote: Compressing objects: 100% (66/66), done.
remote: Total 7381 (delta 51), reused 82 (delta 39), pack-reused 7262 (from 1)
Among the received objects: 100% (7381/7381), 3.37 MiB | 1.61 MiB/s, complete.
Processing delta: 100% (467/4677), complete.
Note: is to switch to the '8430 cddd7b86a19e19823fac3d72d899794f7256'.

You are in a separate head pointer state. You can view, make experimental changes and commit, and you can switch between
When returning to a branch, the commit made in this state is discarded without affecting the branch.

If you want to keep commits made in this state by creating a branch, you can do so by using the switch command
Add the -c parameter to do this (now or later). For example:

git switch -c < new branch name >

Or undo this action:

git switch -

Turn this advice off by setting the configuration variable advise.detachedhead to false

Elapsed Time downloading rpclib: 8 seconds
Setup.sh: Building rpclib with libc++.
CMake Error at/usr/share/CMake - 3.16 / Modules/CMakeDetermineCCompiler. CMake: 49 (the message) :
Could not find compiler set in environment variable CC:

/ Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux Linux_x64 / v17_clang - 10.0.1 - centos7 / x86_64 - unknown - Linux - gnu/bin/clang.
Call Stack (most recent call first):
CMakeLists.txt:2 (project)

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
See also "/ home/jason32 / Carla/Build/rpclib - v2.2.1 _c5 - c10 - libcxx - Build/CMakeFiles/CMakeOutput log".
Make: * * * [Util/BuildTools/Linux. Mk: 142: setup] 1

PS:I also run "make .PythonAPI ARGS="--python-version=3.8" but the same error occurred. I will appreciate it if you could  give me some help.

PatrickPromitzer, 2024-09-04:
Hi,
try using glang-8
```
sudo update-alternatives --install /usr/bin/clang++ clang++ /usr/lib/llvm-8/bin/clang++ 180 &&
sudo update-alternatives --install /usr/bin/clang clang /usr/lib/llvm-8/bin/clang 180
```

to install clang-8
```
sudo apt-get installclang-8
```

qaz490365155, 2024-09-05:
> 嗨，试试使用 glang-8
> 
> ```
> sudo update-alternatives --install /usr/bin/clang++ clang++ /usr/lib/llvm-8/bin/clang++ 180 &&
> sudo update-alternatives --install /usr/bin/clang clang /usr/lib/llvm-8/bin/clang 180
> ```
> 
> 安装 clang-8
> 
> ```
> sudo apt-get installclang-8
> ```
sudo apt-get install clang-8

ShahzebTUMBachelor, 2024-09-05:
Have the same error but this did not help

ShahzebTUMBachelor, 2024-09-05:
@PatrickPromitzer 
I get this error after switching to clang-8
/home/shahzeb/carla/Build/boost-1.80.0-c10-source/tools/build/src/tools/common.jam:1140: in common.find-compiler from module common
error: toolset clang-linux initialization:
error: version '10.0' requested but 'clang++-10.0' not found and version '8.0.1' of default 'clang++' does not match
error: initialized from
/home/shahzeb/carla/Build/boost-1.80.0-c10-source/tools/build/src/tools/clang-linux.jam:52: in clang-linux.init from module clang-linux
/home/shahzeb/carla/Build/boost-1.80.0-c10-source/tools/build/src/build/toolset.jam:44: in toolset.using from module toolset
/home/shahzeb/carla/Build/boost-1.80.0-c10-source/tools/build/src/tools/clang.jam:33: in clang.init from module clang
/home/shahzeb/carla/Build/boost-1.80.0-c10-source/tools/build/src/build/toolset.jam:44: in toolset.using from module toolset
/home/shahzeb/carla/Build/boost-1.80.0-c10-source/tools/build/src/build-system.jam:543: in process-explicit-toolset-requests from module build-system
/home/shahzeb/carla/Build/boost-1.80.0-c10-source/tools/build/src/build-system.jam:610: in load from module build-system
/home/shahzeb/carla/Build/boost-1.80.0-c10-source/tools/build/src/kernel/modules.jam:294: in import from module modules
/home/shahzeb/carla/Build/boost-1.80.0-c10-source/tools/build/src/kernel/bootstrap.jam:135: in module scope from module

make: *** [Util/BuildTools/Linux.mk:142: setup] Error 1

SongJiang32, 2024-09-06:
> @PatrickPromitzer I get this error after switching to clang-8 /home/shahzeb/carla/Build/boost-1.80.0-c10-source/tools/build/src/tools/common.jam:1140: in common.find-compiler from module common [...]

Dude,so do I.

ShahzebTUMBachelor, 2024-09-06:
@SongJiang32 I found the error. In my case the UE4_ROOT variable was not set correctly

andrewbartels1, 2024-09-08:
I was having this same issue using the stable version of the documentation. Using the updated Ubuntu 22.04 Carla docs on `latest`  solved all my issues (link below). Mainly the Noble release doesn't have a clang-10 and lld-10 unless it's pointed to the right main universe apt repo from what I found anyway. I also had libvulkan1 package issues (likely not related) but had to uninstall and reinstall all related packages.

These directions worked for me to successfully run `make PythonAPI` https://carla.readthedocs.io/en/latest/build_linux/

It's worth noting I had a much better time just downloading the pre-built source and got led down another rabbit hole since the carla apt repo server is broken giving no response. This was resolved per the the download directions from [here](https://github.com/carla-simulator/carla/issues/7017).

I hope this helps others, it's also worth mentioning after installing the right setuptools version at the bottom of the [Software requirements section](https://carla.readthedocs.io/en/latest/build_linux/) I ended up downgrading my python to 3.8 rather than 3.12  and the seg faults on `python -c 'import carla'` resolved itself.

The setuptools version specified is older and not compatible with 3.8> for the pre-built source. For my use of integrating with [kedro](https://docs.kedro.org/en/stable/) this still works just fine!

SongJiang32, 2024-09-09:
> I was having this same issue using the stable version of the documentation. **Using the updated Ubuntu 22.04 Carla docs on `latest` generated a workaround to all my issues (link below).** Mainly the Noble release doesn't have a c++11 capable compiler, and it couldn't find clang-10 and llvd-10 requirements unless it's pointed to the right main universe apt repo from what I found anyway.
> 
> I also had libvulkan1 package issues (likely not related) but had to uninstall and reinstall all related packages.
> 
> These directions link below worked for me to to setup the dependencies on Ubuntu before manually installing the pre-built carla
> 
> https://carla.readthedocs.io/en/latest/build_linux/ (I know it's a workaround, but it got the job done)
> ## Workaround
> 
> It's worth noting I had a much better time just downloading the pre-built source and got led down another rabbit hole since the carla apt repo server is broken giving no response. This was resolved per the the download directions from [here](https://github.com/carla-simulator/carla/issues/7017). I understand this isn't fully satisfactory for some folks, but it gets you up and running.
> 
> I hope this helps others, it's also worth mentioning after installing the right setuptools version at the bottom of the [Software requirements section](https://carla.readthedocs.io/en/latest/build_linux/) I ended up downgrading my python to 3.8 rather than 3.12 and the seg faults on `python -c 'import carla'` resolved itself.
> 
> The setuptools version specified is older and not compatible with 3.8> for the pre-built source. For my use of integrating with [kedro](https://docs.kedro.org/en/stable/) this still works just fine!

Thank you for your help.And now I choose precompiled version to use carla:
    # Create a new python enviroment
    $ conda create -n <name> python=3.7 -y
    # Install UnrealEngine
    $ git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git ~/UnrealEngine_4.26
    $ cd ~/UnrealEngine_4.26
    $ ./Setup.sh && ./GenerateProjectFiles.sh && make
    $ sudo gedit ~/.bashrc
       UE4_ROOT=~/UnrealEngine_4.26
    # Install Carla0.9.13
    $ sudo apt-get install aria2
    # Download  CARLA_0.9.13.tar.gz & AdditionalMaps_0.9.13.tar.gz
       https://github.com/carla-simulator/carla/releases/tag/0.9.13/
    $ tar -xzvf CARLA_0.9.13.tar.gz
    # Put "AdditionalMaps_0.9.13.tar.gz" under "Import" in CARLA_0.9.13 
    $ cd CARLA_0.9.13/
    $ ./ImportAssets.sh

kanbarashen, 2025-09-16:
> Same problem. "/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang" is in UE4 so the problem is Setup.sh cannot find the right UE4_ROOT
> I set UE4_ROOT in the terminal but it not works so I set it in Setup.sh
> try add
> `export UE4_ROOT= the/path/to/your/UE4`
> after
> `source $(dirname "$0")/Environment.sh `
> before
> `export CC="$UE4_ROOT/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang"`
> which works for me 

 _Originally posted by @kanbarashen in [#7516](https://github.com/carla-simulator/carla/issues/7516#issuecomment-3295239847)_