carla-simulator/carla #8096: build carla UE5 on win11(PythonAPI ERROR)

lt-wang, 2024-09-01:
CARLA version: 0.9.15
Platform/OS: win11
Problem you have experienced:
"cmake --build Build --target carla-python-api-install" error

What you expected to happen:
Steps to reproduce:
[image]
[image]

Other information (documentation you consulted, workarounds you tried):

MarcelPiNacy-CVC, 2024-09-02:
Hi, have you run `python -m pip install -r requirements.txt`?

lt-wang, 2024-09-03:
> Hi, have you run `python -m pip install -r requirements.txt`?

I run this command and the problem is solved. Thanks for your reply.

closed 2024-09-03 by lt-wang