carla-simulator/carla #8103: Access error when make launch

Sephiroth1118, 2024-09-03:
CARLA version: 0.9.15
Platform/OS: Ubuntu 20.04
Problem you have experienced: 
This is the log.
[image]
It should be a network error. But the strange thing is this link is 404 Page not found now. Amd yesterday when I run make launch it does not have this problem. I check the Plugins folder in carla and StreetMap does exist. What might be the problem? Does carla need to access this link everytime running make launch?
[image]
What you expected to happen:
Tell me what's happening.
Steps to reproduce:
No idea. Just run make launch.

Sephiroth1118, 2024-09-03:
Another strange thing. Although I can not access this link in web browser, I can access it by git clone.

Blyron, 2024-09-03:
Repository is still there https://github.com/carla-simulator/StreetMap
Try again

Sephiroth1118, 2024-09-04:
> Repository is still there https://github.com/carla-simulator/StreetMap Try again

Thank you very much for your reply! I delete my carla folder and remake another one, it works again. But I'm confused that each time I run make launch, it will try to access ths Repo. If I do not connect to internet, make launch won't work. Is this normal? I'm new to Carla, it is kind of wierd that I can not run UE editor if I do not have internet connection.

Blyron, 2024-09-04:
Our build system is not the best one. We are open to contributios while dealing with other stuff. Sorry for inconveniences

closed 2024-09-04 by Blyron