carla-simulator/carla #8106: make PythonAPI failed

ShahzebTUMBachelor, 2024-09-03:
CARLA version: 0.9.15
Platform/OS: Ubuntu 22.04
Python3-Version: 3.10.12

Carla was actually running normally but I had to install some plugins for RoadRunner. Unfortunately my PC crashed and after that I could not launch CarlaUE4 anymore. 

This was the error log before I deleted everything:
The following untracked working tree files would be overwritten by checkout:
	.appveyor.yml
	.gitattributes
	.github/issue_template.md
	.github/workflows/build-linux.yml
	.github/workflows/build-macos.yml
	.github/workflows/build-windows.yml
	.github/workflows/docker.yml
	.github/workflows/documentation.yml
	.github/workflows/wheel-linux.yml
	.github/workflows/wheel-macos.yml
	.github/workflows/wheel-windows.yml
	.gitignore
	.gitmodules
	AUTHORS
	CITATION.cff
	CMakeLists.txt
	CMakeSettings.json
	CONTRIBUTING.md
	ChangeLog
	Jenkinsfile
	LICENSE
	NOTICE.md
	README.md
	bin/start-command-line.bat
	cmake-variants.yaml
	data/3D/car-microcargo-citrus.mtl
	data/3D/car-minibus-citrus.mtl
	data/3D/car-normal-citrus.mtl
	data/3D/humanResting.mtl
	data/3D/tl.obj.mtl
	data/3D/tram.mtl
	data/emissions/MMPEVEM/BMW_i3.xml
	data/emissions/MMPEVEM/SUV.xml
	data/emissions/MMPEVEM/VW_ID3.xml
	data/emissions/MMPEVEM/VW_ID4.xml
	data/emissions/MMPEVEM/VW_eUp.xml
	data/emissions/PHEMlight/PC_D_EU4.PHEMLight.veh
	data/emissions/PHEMlight/PC_D_EU4.csv
	data/emissions/PHEMlight/PC_D_EU4_FC.csv
	data/emissions/PHEMlight/PC_G_EU4.PHEMLight.veh
	data/emissions/PHEMlight/PC_G_EU4.csv
	data/emissions/PHEMlight/PC_G_EU4_FC.csv
	data/emissions/PHEMlight5/Deterioration.det
	data/emissions/PHEMlight5/Mileage.vma
	data/emissions/PHEMlight5/NOxCor.tno
	data/emissions/PHEMlight5/PC_EU4_D_MW.PHEMLight.veh
	data/emissions/PHEMlight5/PC_EU4_D_MW.csv
	data/emissions/PHEMlight5/PC_EU4_D_MW_FC.csv
	data/emissions/PHEMlight5/PC_EU4_G.PHEMLight.veh
	data/emissions/PHEMlight5/PC_EU4_G.csv
	data/emissions/PHEMlight5/PC_EU4_G_FC.csv
	data/font/Roboto-Medium.ttf
	data/lang/visumEN.txt
	data/lang/visumFR.txt
	data/lang/visumIT.txt
	data/logo/netedit-application-icon.ico
	data/logo/sumo-128x138.png
	data/logo/sumo-145x50.png
	data/logo/sumo-application-icon.ico
	data/po/de_gui.po
	data/po/de_sumo.po
	data/po/es_gui.po
	data/po/es_sumo.po
	data/po/fr_gui.po
	data/po/fr_sumo.po
	data/po/hu_gui.po
	data/po/hu_sumo.po
	data/po/tr_gui.po
	data/po/tr_sumo.po
	data/po/zh-Hant_gui.po
	data/po/zh-Hant_sumo.po
	data/po/zh_gui.po
	data/po/zh_sumo.po
	data/typemap/navteqPolyconvert.typ.xml
	data/typemap/opendriveNetconvert.typ.xml
	data/typemap/opendriveNetconvertBicycle.typ.xml
	data/typemap/opendriveNetconvertPedestrians.typ.xml
	data/typemap/osmNetconvert.typ.xml
	data/typemap/osmNetconvertAerialway.typ.xml
	data/typemap/osmNetconvertAirport.typ.xml
	data/typemap/osmNetconvertBicycle.typ.xml
	data/typemap/osmNetconvertBidiRail.typ.xml
	data/typemap/osmNetconvertExtraRail.typ.xml
	data/typemap/osmNetconvertPedestrians.typ.xml
	data/typemap/osmNetconvertPedestriansNES.typ.xml
	data/typemap/osmNetconvertRailUsage.typ.xml
	data/typemap/osmNetconvertShips.typ.xml
	data/typemap/osmNetconvertUrbanDe.typ.xml
	data/typemap/osmPolyconvert.typ.xml
	data/typemap/osmPolyconvertRail.typ.xml
	data/typemap/visumPolyconvert.typ.xml
	data/xsd/activitygenConfiguration.xsd
	data/xsd/additional_file.xsd
	data/xsd/amitran/linkdata.xsd
	data/xsd/amitran/network.xsd
	data/xsd/amitran/od.xsd
	data/xsd/amitran/routes.xsd
	data/xsd/amitran/trajectories.xsd
	data/xsd/baseTypes.xsd
	data/xsd/battery_file.xsd
	data/xsd/calibratorstats_file.xsd
	data/xsd/collision_file.xsd
	data/xsd/connections_file.xsd
	data/xsd/datamode_file.xsd
	data/xsd/det_e1_file.xsd
	data/xsd/det_e1meso_file.xsd
	data/xsd/det_e2_file.xsd
	data/xsd/det_e3_file.xsd
	data/xsd/detectors_file.xsd
	data/xsd/dfrouterConfiguration.xsd
	data/xsd/duarouterConfiguration.xsd
	data/xsd/edgediff_file.xsd
	data/xsd/edges_file.xsd
	data/xsd/emission_file.xsd
	data/xsd/fcd_file.xsd
	data/xsd/full_file.xsd
	data/xsd/genericparameter_file.xsd
	data/xsd/instant_e1_file.xsd
	data/xsd/jtrrouterConfiguration.xsd
	data/xsd/marouterConfiguration.xsd
	data/xsd/meandataTypes.xsd
	data/xsd/meandata_file.xsd
	data/xsd/net_file.xsd
	data/xsd/netconvertConfiguration.xsd
	data/xsd/netgenerateConfiguration.xsd
	data/xsd/netstate_file.xsd
	data/xsd/nodes_file.xsd
	data/xsd/od2tripsConfiguration.xsd
	data/xsd/person_summary_file.xsd
	data/xsd/polyconvertConfigurat
Aborting
make: *** [Util/BuildTools/Linux.mk:157: osm2odr] Error 1

I deleted Carla but not Unreal and now I am installing it again but getting this error when doing make PythonAPI

This is the main error log:
27 warnings generated.
...skipped <pbin.v2/libs/python/build/clang-linux-10.0/release/python-3.10/threading-multi/visibility-hidden>libboost_numpy310.so.1.80.0 for lack of <pbin.v2/libs/python/build/clang-linux-10.0/release/python-3.10/threading-multi/visibility-hidden>numpy/dtype.o...
...skipped <p/media/shahzeb/Volume/carla/Build/boost-1.80.0-c10-source/stage/lib>libboost_numpy310.so.1.80.0 for lack of <pbin.v2/libs/python/build/clang-linux-10.0/release/python-3.10/threading-multi/visibility-hidden>libboost_numpy310.so.1.80.0...
...skipped <p/media/shahzeb/Volume/carla/Build/boost-1.80.0-c10-source/stage/lib>libboost_numpy310.so for lack of <p/media/shahzeb/Volume/carla/Build/boost-1.80.0-c10-source/stage/lib>libboost_numpy310.so.1.80.0...
...skipped <pbin.v2/libs/python/build/clang-linux-10.0/release/python-3.10/threading-multi/visibility-hidden>libboost_numpy-variant-shared-py3.10.cmake for lack of <pbin.v2/libs/python/build/clang-linux-10.0/release/python-3.10/threading-multi/visibility-hidden>libboost_numpy310.so.1.80.0...
...skipped <p/media/shahzeb/Volume/carla/Build/boost-1.80.0-c10-source/stage/lib/cmake/boost_numpy-1.80.0>libboost_numpy-variant-shared-py3.10.cmake for lack of <pbin.v2/libs/python/build/clang-linux-10.0/release/python-3.10/threading-multi/visibility-hidden>libboost_numpy-variant-shared-py3.10.cmake...
...failed updating 2 targets...
...skipped 10 targets...
...updated 266 targets...
make: *** [Util/BuildTools/Linux.mk:142: setup] Error 1

Although I am also getting many warnings before. 

What am I doing wrong? Please help! I am a student.

PatrickPromitzer, 2024-09-04:
Hi,
if you work with Carla, use python3.8 or you have to edit some versions and files to make it work.

On Ubuntu 22.04, I would not change the python3 soft link.

You can try giving the make command arguments (I never tried it before but should work)
```
make PythonAPI ARGS=" --python-version 3.8"
```

I normally edit the files for a permanent switch to python3.8 

path/to/carla/Util/BuildTools/*
```
search for
PY_VERSION_LIST=3

replace with
PY_VERSION_LIST=3.8
```

The files I know about (without looking through all of them) are
BuildPythonAPI.sh
Setup.sh
Import.sh

ShahzebTUMBachelor, 2024-09-04:
@PatrickPromitzer this does not help. I still get the same error. I even deleted the partition and reinstalled Ubuntu and still I am having this problem. Can you tell me what Ubuntu Version you are using?

ShahzebTUMBachelor, 2024-09-04:
...failed updating 2 targets...
...skipped 10 targets...
...updated 266 targets...
make: *** [Util/BuildTools/Linux.mk:142: setup] Error 1

PatrickPromitzer, 2024-09-05:
The best Ubuntu version for Carla 0.9.15 is Ubuntu 20.04
I got it working with Ubuntu 22.04 but one package isn't supported for Ubuntu 22.04 anymore

If you have a new Ubuntu 22.04 system, it would take a lot more time to get it working.
You can try Ubuntu 20.04 and look how far you can get with the installation

ShahzebTUMBachelor, 2024-09-05:
@PatrickPromitzer I solved this issue. But now I get this error: 
CMake Error at /usr/share/cmake-3.22/Modules/CMakeDetermineCCompiler.cmake:49 (message):
  Could not find compiler set in environment variable CC:

  
  /Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
See also "/home/shahzeb/carla/Build/rpclib-v2.2.1_c5-c10-libcxx-build/CMakeFiles/CMakeOutput.log".
make: *** [Util/BuildTools/Linux.mk:142: setup] Error 1
You have any idea?

ShahzebTUMBachelor, 2024-09-05:
@PatrickPromitzer I switched to clang-8 but still I am getting the same error

ShahzebTUMBachelor, 2024-09-05:
I have

closed 2024-09-05 by ShahzebTUMBachelor

WilliamLiu108, 2025-04-11:
I got the same error with you. Can you share the solution how to solve the first issue? Thanks!

PatrickPromitzer, 2025-04-11:
@WilliamLiu108 The solution was shared one message before yours.