carla-simulator/carla #8108: Load opdrive without fbx in UEeditor

Sephiroth1118, 2024-09-04:
Hi! 
I'm now working on a project that needs to create a map in Carla which should contains buildings. The standalone mode doesn't meet the requirement and I do not have the license of Roadrunner. What I'm trying to do is load the xodr file inUEeditor as a bare map like in standalone mode and then add some buildings manually. But I found make import do not work for xodr file only. Is there anyway to store the bare map in standalone mode as umap so that I can edit it in UEeditor?
Thank you so much!

PatrickPromitzer, 2024-09-04:
Hi,
I found this tool which could work in your favor.

https://github.com/ishaan95/blenderOpenDRIVEImporter

Sephiroth1118, 2024-09-04:
> Hi, I found this tool which could work in your favor.
> 
> https://github.com/ishaan95/blenderOpenDRIVEImporter

Oh! Thank you so much! This problem is really troubling me for a long time. You provide a new perspective to solve the problem. I will try this. Thank you again!

Sephiroth1118, 2024-09-05:
In carla/LibCarla/source/carla/geom/Mesh.cpp, there is a function called Mesh::GenerateOBJ(). It can export the mesh in Obj file and then be used in blender. But I'm not sure how to use it. I didn't find much information about C++ API. How can I get the mesh information.

PatrickPromitzer, 2024-09-06:
The function you found is probably for the Digital Twin tool, which should also be able to load opdenDRIVE files.
The documentation is incomplete, and many people (me included) don't know how to get it running.

Sephiroth1118, 2024-09-06:
> The function you found is probably for the Digital Twin tool, which should also be able to load opdenDRIVE files. The documentation is incomplete, and many people (me included) don't know how to get it running.

Hi! Thank you for your reply! Actually I found this function in the issue #3478 https://github.com/carla-simulator/carla/issues/3478. The author said just call this function and it will get the OBJ string. But I'm not sure how to call this function in the code.