carla-simulator/carla #8114: Question about how to create a real word map for carla simulation

JarvisLee0423, 2024-09-05:


PatrickPromitzer, 2024-09-06:
Hi,
about importing a .xodr and .fbx file, you could post the error text to be able to help you fix the issue.

Carla has a digital twin tool to load real-world road data in the simulation, but the documentation is not complete and many people don't know how to use it.

JarvisLee0423, 2024-09-07:
> Hi, about importing a .xodr and .fbx file, you could post the error text to be able to help you fix the issue.
> 
> Carla has a digital twin tool to load real-world road data in the simulation, but the documentation is not complete and many people don't know how to use it.

Thanks for your attention. Here are my error texts. I got the map from the Hong Kong environment mesh web (https://3d.map.gov.hk/catalogue), and my mesh is show in following picture.

PatrickPromitzer, 2024-09-09:
The array error could come from anywhere. You can copy the whole text and post it between two ```(without space) lines
```
like this
```

The last part is a warning, and should not make a problem. If it does, it would be something that need to be fixed.