carla-simulator/carla #8115: Building CarlaUnrealEditor failed under WSL

qq1183713118, 2024-09-06:
I tried to build Carla's dev-ue5 branch under WSL but it failed.
I followed the UE5_BUILD_README file but it failed in Building CarlaUnrealEditor 

[image]
[image]
[image]

PatrickPromitzer, 2024-09-06:
Hi,
this is from the CarlaUE5 doc.
[image]

I can see you use "/mnt/e/", which is the default path of an external drive.

merryHunter, 2024-09-08:
I somehow missed this doc, but I faced similar issue trying to run Carla from WSL... and docker on windows didn't go well either :(

qq1183713118, 2024-09-09:
> Hi, this is from the CarlaUE5 doc. [...]

Thanks, I'll try again

PatrickPromitzer, 2024-09-09:
Don't use sudo.
The output is telling you "Refusing to run with the root privileges."
If you have a permission problem, you set them to 777 to make sure it works (755 would be better)
```
chmod -R 777 path/to/carla

or

sudo chmod -R 777 path/to/carla
```

I used this commands, because I had a problem with the ros installation

``` 
git clone -b ue5-dev https://github.com/carla-simulator/carla.git CarlaUE5

cd CarlaUE5

env GIT_LOCAL_CREDENTIALS=your_git_user@your_git_token bash -x CarlaSetup.sh

cmake -G Ninja -S . -B Build --toolchain=$PWD/CMake/LinuxToolchain.cmake -DLAUNCH_ARGS="-prefernvidia" -DCMAKE_BUILD_TYPE=Release -DBUILD_CARLA_UNREAL=ON -DCARLA_UNREAL_ENGINE_PATH=$CARLA_UNREAL_ENGINE_PATH

cmake --build Build --target launch
```

qq1183713118, 2024-09-10:
> 不要使用 sudo。输出告诉您 “Refusing to run with the root privileges”。如果你有权限问题，你可以将它们设置为 777 以确保它有效（755 会更好）
> 
> ```
> chmod -R 777 path/to/carla
> 
> or
> 
> sudo chmod -R 777 path/to/carla
> ```
> 
> 我使用了这个命令，因为我在 ros 安装时遇到了问题
> 
> ```
> git clone -b ue5-dev https://github.com/carla-simulator/carla.git CarlaUE5
> 
> cd CarlaUE5
> 
> env GIT_LOCAL_CREDENTIALS=your_git_user@your_git_token bash -x CarlaSetup.sh
> 
> cmake -G Ninja -S . -B Build --toolchain=$PWD/CMake/LinuxToolchain.cmake -DLAUNCH_ARGS="-prefernvidia" -DCMAKE_BUILD_TYPE=Release -DBUILD_CARLA_UNREAL=ON -DCARLA_UNREAL_ENGINE_PATH=$CARLA_UNREAL_ENGINE_PATH
> 
> cmake --build Build --target launch
> ```

I tried this command and it seems to be useful. But there was an error when running UE5Editor. And I ensure that the GPU driver and Vulkan have been installed, and information can be queried using nvidia-smi and vulkaninfo
[image]

MarcelPiNacy-CVC, 2024-09-12:
Hi,

What Patrick said should indeed work. Could you share the output of both nvidia-smi and vulkaninfo? Btw, please attach logs using text files instead of images if possible.

Marcel

PatrickPromitzer, 2024-09-12:
Do you have an CPU with graphics card?

Maybe your computer is using the CPU instead of the Nvidia graphic card (a guess)
```sudo nvidia-settings```
[image]

closed 2024-09-13 by qq1183713118