carla-simulator/carla #8116: Standard output functionality in libcarla is not operating as expected

AOOOOOA, 2024-09-08:
CARLA version: 0.9.14
Platform/OS: ubuntu18.04 
Problem you have experienced: 
I attempted to generate intermediate output in Libcarla using std::cout. However, after testing in multiple cpp files and executing 'make PythonAPI' and 'make launch', and running a Python script to control a vehicle in the Carla environment, the expected output did not appear.

I'm puzzled by this behavior and would appreciate your assistance in understanding why the output is not being displayed as anticipated. Could you please help me troubleshoot this issue?

PatrickPromitzer, 2024-09-09:
Hi,
I don't work with the C++ part of Carla, but I am not sure if you can use the C++ code to collect data.
With the python API you can control the vehicles, add sensors, and collect data from the sensors and the simulation.

You could ask in the Carla Discord if someone can help you or at least answer some questions about C++

AOOOOOA, 2024-09-10:
> Hi, I don't work with the C++ part of Carla, but I am not sure if you can use the C++ code to collect data. With the python API you can control the vehicles, add sensors, and collect data from the [...]

I have fixed it already. Thanks for your kindly reply!

closed 2024-09-10 by AOOOOOA