carla-simulator/carla #8120: The packaged version of carla cannot call carsim(VehicleSim Dynamics plugin)

kuangzh123, 2024-09-10:
CARLA version: 0.9.14
Platform/OS: Windows 10/Ubuntu18.04
Problem I have experienced:
According to the official documentation, both source-compiled and  packaged version of carla should be able to use the carsim (VehicleSim Dynamics plugin). However, after testing, it is found that only the source-compiled version of carla can call carsim normally, and the  packaged version of carla cannot use carsim.
[image]
Has anyone experienced this problem?

-->

yufei0726, 2024-11-04:
dude how do you get the plugin

EvilQQQ, 2024-11-06:
Only Carla0.9.11 can use carsim on windows.

yufei0726, 2024-11-06:
> Only Carla0.9.11 can use carsim on windows.

Do you have a tutorial bro?

EvilQQQ, 2024-11-06:
[image]
you can find Carsim file in CARLA_0.9.11.
[image]
And then you need to create .sim file by carsim. As shown in the below figure,
[image]
Generate Files for this run. 
And then use carla API like this enable_carsim("uesimfile.sim")

9900ff, 2025-01-10:
How do I get the VehicleSim Dynamics Plugin? As a personal researcher I don't have a carsim license at the moment.

EvilQQQ, 2026-05-09:
I’m not sure what exactly you need. I’ve already forgotten the specific steps I took at that time.
However, as far as I remember, the precompiled version of CARLA 0.9.11 could directly work with CarSim, and the setup process was not complicated. There are also plenty of related resources available online.

I haven’t tried other versions.

In addition, CarSim itself is commercial software, so you can either use some “tricks” to access it or purchase a legitimate license.