carla-simulator/carla #8124: How to generate CARLA  random traffic generation for left hand drive (LHD) in custom map?

JIANGZhengm, 2024-09-11:
CARLA version: 0.9.15 source build
Platform/OS: windows 11
Problem you have experienced: I am using CARLA 0.9.15 compiled from source on windows, could anybody tell me how to achieve left traffic flow simulation? Is it possible to just select left traffic rule when exporting to opendrive through roadrunner?
The effect I expect to achieve is like a youtube link: https://www.youtube.com/watch?v=zc8Gx6zszW0

Dave-agility3, 2024-10-17:
@JIANGZhengm - Did you find a solution?
We have the same issue. Selecting the left side traffic rule in RoadRunner results in an OpenDrive map that is correct when viewed in other tools, but it doesn't generate traffic correctly in Carla.

Abhilashsabhi, 2025-02-18:
Did anyone find the solution?

closed 2025-08-25 by glopezdiest