carla-simulator/carla #8127: Tag UnrealEngine alongside CARLA tags

qhaas, 2024-09-11:
CARLA version: ue5-dev
Platform/OS: Ubuntu 22.04
Problem you have experienced: Currently, [`CarlaSetup.sh`](https://github.com/carla-simulator/carla/blob/239b73e849de3635e07f612c2c9e6638276c90a9/CarlaSetup.sh#L117) clones the head of [CarlaUnreal `ue5-dev-carla` branch](https://github.com/CarlaUnreal/UnrealEngine/tree/ue5-dev-carla).  This is also recommended [in the documentation](https://github.com/carla-simulator/carla/blob/ue5-dev/UE5_BUILD_README.md#set-up-the-environment)
What you expected to happen: Given UE is a major dependency, every time a tag / release for CARLA is generated, the CARLA fork of UE should have a tag created to ensure reproducibility.  The build script and documentation should leverage said tag.

Blyron, 2024-11-06:
We haven't generated tags as we haven't done the release this. Once the version is released the tags will be generated

closed 2024-11-06 by Blyron

qhaas, 2024-12-24:
Looks like the CARLA UE devs have tagged it, thanks!  https://github.com/CarlaUnreal/UnrealEngine/releases/tag/0.10.0