carla-simulator/carla #8134: make import issue using roadrunner map

Ahnhojin1223, 2024-09-13:
CARLA version: 0.9.5
Platform/OS: Ubuntu 20.04
Problem you have experienced: 

Previously, I put the map I created through roadrunner into the import folder and made it available on carla by using 
'''
make import ARGS="--package=name"
'''

I did the same thing about 2~3 weeks later, but the custom map does not open on the carla.

The error occurd at <./CarlaUE4.sh> terminal is:
'''
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
LowLevelFatalError [File:Unknown] [Line: 149] 
A FRenderResource was deleted without being released first!
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
'''

And there is some strange part while doing <make import ARGS="--package=name">:
end of that terminal...
'''
[2024.09.13-11.56.10:025][  0]LogModuleManager: Shutting down and abandoning module MeshPaintMode (502)
[2024.09.13-11.56.10:025][  0]LogModuleManager: Shutting down and abandoning module SceneDepthPickerMode (500)
[2024.09.13-11.56.10:025][  0]LogModuleManager: Shutting down and abandoning module ActorPickerMode (498)
[2024.09.13-11.56.10:025][  0]LogModuleManager: Shutting down and abandoning module PlacementMode (496)
[2024.09.13-11.56.10:025][  0]LogModuleManager: Shutting down and abandoning module LevelSequence (494)
[2024.09.13-11.56.10:025][  0]LogModuleManager: Shutting down and abandoning module MovieSceneTracks (492)
[2024.09.13-11.56.10:025][  0]LogModuleManager: Shutting down and abandoning module MovieScene (490)
[2024.09.13-11.56.10:025][  0]LogModuleManager: Shutting down and abandoning module StreamingPauseRendering (488)
[2024.09.13-11.56.10:025][  0]LogModuleManager: Shutting down and abandoning module ImageWriteQueue (486)
[2024.09.13-11.56.10:025][  0]LogModuleManager: Shutting down and abandoning module AIModule (484)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module GameplayDebugger (483)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module AITestSuite (481)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module NavigationSystem (479)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module AudioSynesthesiaEditor (476)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module CarlaExporter (474)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module CarlaTools (472)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module ContentBrowserAssetDataSource (470)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module ContentBrowserClassDataSource (468)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module ToolMenus (467)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module ContentBrowserFileDataSource (464)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module CryptoKeysOpenSSL (462)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module CryptoKeys (460)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module MaterialAnalyzer (458)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module SpeedTreeImporter (456)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module EditorScriptingUtilities (454)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module CurveEditorTools (452)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module PluginBrowser (450)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module DatasmithContentEditor (448)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module VariantManagerContentEditor (446)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module GooglePAD (444)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module CableComponent (442)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module NiagaraEditorWidgets (440)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module ScreenshotTools (438)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module AssetTags (436)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module EditableMesh (434)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module CustomMeshComponent (432)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module ProceduralMeshComponentEditor (430)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module ProceduralMeshComponent (428)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module AlembicHairTranslatorModule (426)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module HairStrandsEditor (425)
[2024.09.13-11.56.10:026][  0]LogModuleManager: Shutting down and abandoning module LightPropagationVolumeEditor (422)
[... 185 lines cut ...]
[2024.09.13-11.56.10:580][  0]LogModuleManager: Shutting down and abandoning module PlatformCryptoOpenSSL (42)
[2024.09.13-11.56.10:580][  0]LogModuleManager: Shutting down and abandoning module PlatformCryptoTypes (40)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module PlatformCrypto (38)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module AnimationModifiers (36)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module AudioEditor (34)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module PropertyEditor (33)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module TextureCompressor (30)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module RenderCore (28)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module Landscape (26)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module SlateRHIRenderer (24)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module OpenGLDrv (22)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module AnimGraphRuntime (20)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module Renderer (18)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module Engine (16)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module CoreUObject (14)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module NetworkFile (12)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module CookedIterativeFile (10)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module StreamingFile (8)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module SandboxFile (6)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module PakFile (4)
[2024.09.13-11.56.10:581][  0]LogModuleManager: Shutting down and abandoning module RSA (3)
[2024.09.13-11.56.10:845][  0]LogContentStreaming: Display: There are 1 unreleased StreamingManagers
[2024.09.13-11.56.10:845][  0]LogContentStreaming: Display: There are 1 unreleased StreamingManagers
[2024.09.13-11.56.10:848][  0]LogExit: Exiting.
'''

liaopuyun, 2024-09-20:
Got the same issues. Please let me know if there is a solution.

Ahnhojin1223, 2024-09-25:
Is anyone to solve this problem????