carla-simulator/carla #8135: carla's python api set_transform () has a position error

array-svg, 2024-09-13:


MarcelPiNacy-CVC, 2024-09-13:
Hi,

Can you elaborate?

array-svg, 2024-09-14:
> Hi,
> 
> Can you elaborate?

Thank you for replying and helping me immediately. i re-studied the log i printed and found that the transorm position set in the i-th frame will not arrive until the (i+1)-th frame. So is this the mechanism of carla set_transform?
here is the log:
`frame1: 
    ego_transform.x: -497.6048278808594 ego_transform.y: 3672.13525390625
    ego_trans.x: -497.603515625 ego_trans.y: 3672.40234375

frame2: 
    ego_transform.x: -497.60638427734375 ego_transform.y: 3671.818603515625
    ego_trans.x: -497.6047668457031 ego_trans.y: 3672.13671875

frame3: 
    ego_transform.x: -497.6082458496094 ego_transform.y: 3671.43994140625
    ego_trans.x: -497.6063232421875 ego_trans.y: 3671.819091796875

frame4: 
    ego_transform.x: -497.6102294921875 ego_transform.y: 3671.011474609375
    ego_trans.x: -497.6081848144531 ego_trans.y: 3671.439208984375

frame5: 
    ego_transform.x: -497.6125793457031 ego_transform.y: 3670.533935546875
    ego_trans.x: -497.6101379394531 ego_trans.y: 3671.00927734375

frame6: 
    ego_transform.x: -497.6148986816406 ego_transform.y: 3670.049072265625
    ego_trans.x: -497.61248779296875 ego_trans.y: 3670.53076171875`