carla-simulator/carla #8139: Failed to find XercesC (missing: XercesC_VERSION)

SirMichcik, 2024-09-15:
CARLA version: 0.9.13
Platform/OS: Ubuntu 20.04
Problem you have experienced: Failed to find XercesC (missing: XercesC_VERSION)
What you expected to happen: building PythonAPI
Steps to reproduce: make PythonAPI
Other information (documentation you consulted, workarounds you tried): i have Xerces 3.2.5 installed instead of 3.2.3 and changed Setup.sh and BuildOSM2ODR.bat accordingly.

Error message that i got:

CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Failed to find XercesC (missing: XercesC_VERSION)
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.16/Modules/FindXercesC.cmake:99 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:71 (find_package)

-- Configuring incomplete, errors occurred!
See also "/media/michal/7eeeb763-63f0-425a-88b3-041b93352d7a/carla/Build/libosm2dr-build/CMakeFiles/CMakeOutput.log".
make: *** [Util/BuildTools/Linux.mk:152: osm2odr] Error 1

PatrickPromitzer, 2024-09-18:
Hi,
In your situation, I would try using Carla 0.9.15.2
There are many different bug fixes since version 0.9.13, and one of them is a changed download link.

If you can't change to Carla 0.9.15.2, I could look at the problem at a later time.

SirMichcik, 2024-09-19:
Hi, thanks for responding. 
I can't use 0.9.15 version, as i need to use a ROS Bridge after that, and this is configured on 0.9.13 version. Unless there is a way of making it work on 0.9.15 that i am not aware of?

PatrickPromitzer, 2024-09-19:
Hi,
open the following file
```path/to/carla/Utils/BuildTools/Setup.sh```

and search for this line 
```
line 432
XERCESC_REPO=https://ftp.cixug.es/apache//xerces/c/3/sources/xerces-c-${XERCESC_VERSION}.tar.gz
```
and change it to
```
XERCESC_REPO=https://archive.apache.org/dist/xerces/c/3/sources/xerces-c-${XERCESC_VERSION}.tar.gz
```

The download link has changed, and I hope this is the only problem.