carla-simulator/carla #8150: Link to CarSim.Build.cs broken

maubrunn, 2024-09-17:
CARLA version: any
Platform/OS: Linux
Problem you have experienced: I am trying to use CarSim with Carla. The documentation of Carla instructs to change the CarSim.Build.cs file with another one that can be downloaded. Unfortunately this link is broken. Is there a new location for this file?

liuqiang12040913, 2024-10-17:
@maubrunn Did you find the link already? I am working on this CARLA+CarSim, but cannot find the link to CarSim.Build.cs.

maubrunn, 2024-10-21:
No unfortunately not, however I rewrote it myself so that it builds:

using System.IO;
using System;

namespace UnrealBuildTool.Rules
{
  public class CarSim : ModuleRules
  {
    private string BaseDir
    {
      get { return Path.GetFullPath( Path.Combine(ModuleDirectory, "../..") ); }
    }

    private string ThirdPartyDir
    {
      get { return Path.Combine( BaseDir, "ThirdParty" ); }
    }

    private bool  IsDebug( ReadOnlyTargetRules Target )
    {
      return      Target.Configuration == UnrealTargetConfiguration.Debug
              ||  Target.Configuration == UnrealTargetConfiguration.DebugGame;
    }

    private bool  IsWindows( ReadOnlyTargetRules Target )
    {
      return      Target.Platform == UnrealTargetPlatform.Win64
              ||  Target.Platform == UnrealTargetPlatform.Win32;
    }

    public CarSim( ReadOnlyTargetRules Target )
    : base(Target)
    {
      PCHUsage = ModuleRules.PCHUsageMode.UseExplicitOrSharedPCHs;
      PrivatePCHHeaderFile = "Public/CarSimPCH.h";

      string WrapperBaseDirectory = Path.Combine(ThirdPartyDir, "vs_vehicle");
      string WrapperBuildDirectory = Path.Combine(WrapperBaseDirectory, "build");
      string WrapperPlatform;
      string WrapperConfig;
      string WrapperDllName;
      string SolversDirectory;

      if (Target.Platform == UnrealTargetPlatform.Win32)
      {
        WrapperPlatform = "Win32";

        // CarSim 32-bit solvers:
        SolversDirectory = Path.Combine(ThirdPartyDir, "VehicleSim/Prog");
      }
      else if (Target.Platform == UnrealTargetPlatform.Win64)
      {
        WrapperPlatform = "x64";

        // CarSim 64-bit solvers:
        SolversDirectory = Path.Combine(ThirdPartyDir, "VehicleSim/Prog");
      }
      else
      {
        WrapperPlatform = "Linux";
        SolversDirectory = Path.Combine(ThirdPartyDir, "VehicleSim/Prog");
      }

      WrapperConfig = "Release";
      WrapperDllName = "libvs_vehicle.so.2021.0";

      string VsVehicleWrapperPath = Path.Combine(WrapperBuildDirectory, WrapperPlatform, WrapperConfig);
      string VsVehicleWrapperDll = Path.Combine(VsVehicleWrapperPath, WrapperDllName);

      PrivateIncludePaths.AddRange(
          new string[] {  "CarSim/Private", WrapperBaseDirectory
          }
      );

      // Names of modules that this module's public interface depends on.
      PublicDependencyModuleNames.AddRange(
          new string[]
          {
                      "Core"
                    , "CoreUObject"
                    , "Engine"
          }
      );

      if ( IsWindows(Target) )
      {
        // Windows-only dependencies:
        PublicDependencyModuleNames.AddRange(
            new string[]
            {
                        "DisplayCluster"
            }
        );
      }

      // Names of modules that this module's private implementation depends on.
      PrivateDependencyModuleNames.AddRange(
          new string[] {
                    "Projects"
          }
      );

      // What I [bdp] think this does: Tells build system that this file is
      // required at runtime, so the build system will package it when the
      // shipping "game" is built ("packaged").
      string VsConnectDll = Path.Combine(ThirdPartyDir, "vs_connect/libvs_connect.so.2021.0");

      RuntimeDependencies.Add(VsConnectDll);
      RuntimeDependencies.Add(VsVehicleWrapperDll);

      /// \todo Iterate over all DLL's in CarSimSolversDirectory and add them all to RuntimeDependencies() instead of specifying hard-coded filename(s):
      RuntimeDependencies.Add(Path.Combine(SolversDirectory, "libcarsim.so.2021.0"));
      RuntimeDependencies.Add(Path.Combine(SolversDirectory, "libtrucksim.so.2021.0"));

      // CarSim Parsfiles:
      string RunsDirectory = Path.Combine(ThirdPartyDir, "VehicleSim/Data");
      /// \todo Iterate over all parsfiles in CarSimRunsDirectory and add them all to RuntimeDependencies().
      RuntimeDependencies.Add(Path.Combine(RunsDirectory, "carsim_all.par"));
      RuntimeDependencies.Add(Path.Combine(RunsDirectory, "trucksim_all.par"));

      // List of delay load DLLs - typically used for External (third party) modules
      PublicDelayLoadDLLs.Add(VsVehicleWrapperDll);

      // VS Connect:
      PublicDelayLoadDLLs.Add(VsConnectDll);
    }
  }
}

liuqiang12040913, 2024-10-25:
Thanks for your script. @maubrunn 
Did you make it work? 
W have some issues when using carla 0.9.15, trucksim 2021, and carsim 2021 plugin. 
When we use carla build from source, then there is an error when compiling the carsim 2021 plugin (i.e., make launch ARGS="--carsim").
If we use carla packaged version, we do not know if it actually connects via the plugin.

Do you mind sharing some of your settings or versions?

maubrunn, 2024-10-29:
I am using carla 0.9.15 together with the 2020.0.3 carsim plugin, but I also had some problems setting it up. Maybe if you post your problems I can help you with it

liuqiang12040913, 2024-11-01:
We have successfully installed the plugin and run the co-simulation with vehicle.enable_carsim(). 
Our setups are: Ubuntu 22.04, Carla 0.9.15, python 3.12.6, TruckSim 2021.0, CarSim Plugin 2021.0

But when we run the manual_control_carsim.py, after pressing Ctrl+K to enable carsim simfile, we can’t control the ego vehicle and the vehicle moves straight forward without any brakes even curse into building.

It seems because simfile issues, but not sure. Do you have any insights about this? What examplary .py file you are using?

leomilk-eng, 2025-01-02:
@liuqiang12040913 I have the same problem as you, if you solve it, can you please contact me?

leomilk-eng, 2025-01-02:
@liuqiang12040913 Where did you get the plugin from?

liuqiang12040913, 2025-01-02:
We downloaded the plugin from its official website.

alfredgu001324, 2025-01-19:
Hello guys, I am working on the same problem of connecting carla with carsim, however I have not started yet. I am wondering since my carsim is on windows, and carla is on linux, how can I connect the two together? Do I just need to follow the instructions of the website?