carla-simulator/carla #8171: Failed to process dependencies for carla (0.9.10.1)

ChinryZhu, 2024-09-19:
Strictly followed the steps for installation then this happened. (I'm pretty sure the issue occurred when "Processing dependencies for carla==0.9.10". If everything were fine,  it should be followed with something like "finished processing dependencies for carla==0.9.10")
[image]
And I think that leads to this
[image]
(%1 is not a valid Win32 application.)

I'm pretty sure the problem isn't with the package itself because my classmate copied the exact same files to her pc then successfully run  it within ten minutes.(So at first I thought maybe something is wrong with the disk folder but that means this might not be the issue) And the same issue already happened to both my pc and another backup computer.

NEED HELP
NEED HELP

PatrickPromitzer, 2024-09-19:
Hi,
try using the .whl file to install the CarlaAPI in python.

```
path/to/python -m pip install path/to/carla.whl
```

ChinryZhu, 2024-09-20:
> Hi, try using the .whl file to install the CarlaAPI in python.
> 
> ```
> path/to/python -m pip install path/to/carla.whl
> ```

Thanks for reply! But I can't find the whl file for 0.9.10. Is this a specific issue with this specfic version?

PatrickPromitzer, 2024-09-23:
I can't answer that, because I worked with Carla version >= 0.9.13 until now.

You can try installing the Carla Python package without the .whl file, but I am not sure if everything works if you do.
```
path/to/python -m pip install carla==0.9.12
```

PabloVD, 2024-11-08:
We currently do not officially provide support to CARLA versions below 0.9.12. Please upgrade to that version or higher and let us know if the problem persists.

closed 2025-02-19 by Blyron