carla-simulator/carla #8175: Is importing of .xodr map files (already) possible in CARLA-UE5?

XLR8-VISION, 2024-09-20:
Hello everyone,

**CARLA version:** CARLA-UE5 0.10.0
**Platform/OS:** Ubuntu 22.04
**Problem you have experienced:** 'make import' for importing .fbx and .xodr maps is not working (yet). The makefile says this function is disabled.
**What you expected to happen:** Import of .osm maps (or .fbx and .xodr files) into CARLA 
**Steps to reproduce:** See: https://carla.readthedocs.io/en/latest/tuto_content_authoring_maps/#importing-your-road-network-into-carla
**Other information (documentation you consulted, workarounds you tried):** Ingesting .osm., .fbx and .xodr in CARLA like described in following offical doc is not working as well in CARLA-UE5 (See: https://github.com/carla-simulator/carla/blob/master/Docs/tuto_G_openstreetmap.md#ingest-into-carla )

Is there another possibility to get .xodr files as map into CARLA / Unreal Editor 5 ?

Blyron, 2024-09-20:
No, is not possible atm

closed 2024-09-20 by Blyron

Blyron, 2024-09-23:
Not right now.

bututou777, 2025-08-19:
> 现在不行。
Is there a way now?