carla-simulator/carla #8179: Ported to the sea surface

CoderXuans, 2024-09-21:
Hello, everyone:
I want to implement a driving simulation on the sea surface. What methods can I use to convert the ground into a sea level and the vehicle into an unmanned surface vehicle?

MattRoweEAIF, 2024-09-25:
Hi. If you want to create a water surface, we would recommend to take a look at the water features in Town 12 and 13, the lakes. They have a good water material that you could reuse. You might be better asking this question in the Discord forums where other users might have some advice to give, the GitHub issues are more for bugs or documentation fixes. Thanks

closed 2024-09-25 by MattRoweEAIF

CoderXuans, 2024-09-25:
Thank you for your response. Before you answered, I didn't know what the Discord forum was, so I tried asking on GitHub. I only know about Towns 1 to 7 and Town 10 in Carla, but I will go back and check Towns 12 and 13.

CoderXuans, 2024-09-25:
@MattRoweEAIF 
Thank you for your response. Before you answered, I didn't know what the Discord forum was, so I tried asking on GitHub. I only know about Towns 1 to 7 and Town 10 in Carla, but I will go back and check Towns 12 and 13.

CoderXuans, 2024-09-26:
@MattRoweEAIF 
Hello, I registered on Discord, but I can't find a forum about Carla. Can you tell me how to join?

CoderXuans, 2024-09-26:
@MattRoweEAIF 
Hello, I registered on Discord, but I can't find a forum about Carla when searching for available servers. Can you tell me how to join?