carla-simulator/carla #8180: Grd map

johnren-code, 2024-09-23:
How to get the grid map of carla map?

PatrickPromitzer, 2024-09-26:
Hi,
if you want the Road data of a Map, you can get it with 
```python
client = carla.Client("127.0.0.1",  2000) 
world = client.get_world()
map = world.get_map()
opendrive_string = map.to_opendrive()

```

If you want something else, I would need more information of the kind of Grid you want.

johnren-code, 2024-09-26:
It is the raster map used for path planning that contains the map possessed or not possessed

PatrickPromitzer, 2024-09-26:
I didn't find something fitting, but with the Semantic LIDAR sensor you should be able to crate one yourself.
https://carla.readthedocs.io/en/latest/ref_sensors/#semantic-lidar-sensor

Set the Semantic Lidar to different places of the map, and use the "tag" to find roads, ground, buildings, ....

```
raw_data (bytes) 	

- Array containing the point cloud with instance and semantic information. 
  For each point, four 32-bits floats are stored. XYZ coordinates.
- cosine of the incident angle.
- Unsigned int containing the index of the object hit.
- Unsigned int containing the semantic tag of the object it.
```

Blyron, 2025-04-01:
Are we okay to close this?