carla-simulator/carla #8191: Digital Twin Tool: 'make osmrenderer' gives error ‘class lunasvg::Bitmap’ has no member named ‘valid’

XLR8-VISION, 2024-09-25:
**Hello everyone,**

when building the Digital Twin Tool / OSM renderer with 'make osmrenderer' I get this error message: 
‘class lunasvg::Bitmap’ has no member named ‘valid’

**CARLA version:** 
0.9.15

**Platform/OS:**
Ubuntu 20.04

**Problem you have experienced:** 
Error message ‘class lunasvg::Bitmap’ has no member named ‘valid’ is interrupting the build process.

**What you expected to happen:**
Digital Twin Tool / OSM renderer is built 100% successfully.

**Steps to reproduce:** 
Building the OSM renderer according to this documentation https://carla.readthedocs.io/en/0.9.15/adv_digital_twin/

**Other information (documentation you consulted, workarounds you tried):**
Updating CMake from version 3.16 to 3.30 also doesn't help.

**Error message as screenshot:**
[image]

**Error message as text:**
Install the project...
-- Install configuration: ""
-- Up-to-date: /home/ubuntu/carla/osm-world-renderer/ThirdParties/include/lunasvg/lunasvg.h
-- Up-to-date: /home/ubuntu/carla/osm-world-renderer/ThirdParties/lib/liblunasvg.a
-- Up-to-date: /home/ubuntu/carla/osm-world-renderer/ThirdParties/lib/cmake/lunasvg/lunasvgTargets.cmake
-- Up-to-date: /home/ubuntu/carla/osm-world-renderer/ThirdParties/lib/cmake/lunasvg/lunasvgTargets-noconfig.cmake
-- Up-to-date: /home/ubuntu/carla/osm-world-renderer/ThirdParties/lib/cmake/lunasvg/lunasvgConfig.cmake
-- Up-to-date: /home/ubuntu/carla/osm-world-renderer/ThirdParties/lib/cmake/lunasvg/lunasvgConfigVersion.cmake
make[1]: Leaving directory '/home/ubuntu/carla/Build/lunasvg-build'
Building osm-map-renderer
-- Configuring done (0.0s)
-- Generating done (0.0s)
-- Build files have been written to: /home/ubuntu/carla/Build/osm-world-renderer-build
make[1]: Entering directory '/home/ubuntu/carla/Build/osm-world-renderer-build'
make[2]: Entering directory '/home/ubuntu/carla/Build/osm-world-renderer-build'
make[3]: Entering directory '/home/ubuntu/carla/Build/osm-world-renderer-build'
make[3]: Leaving directory '/home/ubuntu/carla/Build/osm-world-renderer-build'
make[3]: Entering directory '/home/ubuntu/carla/Build/osm-world-renderer-build'
[ 16%] Building CXX object CMakeFiles/OsmRenderer.dir/OsmRenderer/src/MapRasterizer.cpp.o
/home/ubuntu/carla/osm-world-renderer/OsmRenderer/src/MapRasterizer.cpp: In member function ‘void MapRasterizer::RasterizeSVG(uint8_t*, std::string, int)’:
/home/ubuntu/carla/osm-world-renderer/OsmRenderer/src/MapRasterizer.cpp:26:26: error: ‘class lunasvg::Bitmap’ has no member named ‘valid’
   26 |     if(!RasterizedBitmap.valid())
      |                          ^~~~~
make[3]: *** [CMakeFiles/OsmRenderer.dir/build.make:104: CMakeFiles/OsmRenderer.dir/OsmRenderer/src/MapRasterizer.cpp.o] Error 1
make[3]: Leaving directory '/home/ubuntu/carla/Build/osm-world-renderer-build'
make[2]: *** [CMakeFiles/Makefile2:85: CMakeFiles/OsmRenderer.dir/all] Error 2
make[2]: Leaving directory '/home/ubuntu/carla/Build/osm-world-renderer-build'
make[1]: *** [Makefile:91: all] Error 2
make[1]: Leaving directory '/home/ubuntu/carla/Build/osm-world-renderer-build'
make: *** [Util/BuildTools/Linux.mk:160: osmrenderer] Error 2

sammycage, 2024-09-28:
@XLR8-VISION This issue has been resolved in the latest commit of lunasvg.

LightBreakfast, 2024-10-26:
I checked out the project today, and am also having this error.

sammycage, 2024-10-26:
@LightBreakfast The `lunasvg::Bitmap::valid` function is definitely in the source code now (you can see it [here](https://github.com/sammycage/lunasvg/blob/2246b6a171bb4e64a20d793861de8f44862fed53/include/lunasvg.h#L248)). If you’re still having issues, try cleaning the build directory and rebuilding. Beyond that, I’m not sure what else it could be.