carla-simulator/carla #8192: Custom map not open in carla

Ahnhojin1223, 2024-09-25:
CARLA version: 0.9.5
Platform/OS: Ubuntu 20.04
Problem you have experienced:

I make my custom map using roadrunner.
And I put fbx, geojson, xml, xodr file in Import folder.
Then when I use this command
```
make import ARGS="--package=name"
```

Then map folder was created in Unreal/CarlaUE4/Content folder. However, there was something strange when I imported using the command above.
```
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc44100] LinuxAArch64NoEditor
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc42e00] LinuxClient
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc42d00] LinuxAArch64Client
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc42c00] LinuxServer
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc42b00] LinuxAArch64Server
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc42a00] Android
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc42900] Android_Low
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc42800] Android_Mid
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc42700] Android_High
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc42600] Android_Default
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc42500] Android_Adreno4xx
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc42400] Android_Adreno5xx_Low
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc42300] Android_Adreno5xx
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc42200] Android_Adreno6xx
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc46300] Android_Adreno6xx_Vulkan
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc46200] Android_Mali_T6xx
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc46100] Android_Mali_T7xx
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc46000] Android_Mali_T8xx
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc45f00] Android_Mali_G71
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc45e00] Android_Mali_G72
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc4c400] Android_Mali_G72_Vulkan
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc4c500] Android_Mali_G76
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc4c600] Android_Mali_G76_Vulkan
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc48500] Android_Mali_G77
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc48600] Android_Mali_G77_Vulkan
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc48700] Android_Vulkan_SM5
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc48800] Android_PowerVR_G6xxx
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc48900] Android_PowerVR_GT7xxx
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc48a00] Android_PowerVR_GE8xxx
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc48b00] Android_PowerVR_GM9xxx
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc48c00] Android_PowerVR_GM9xxx_Vulkan
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc48d00] Android_TegraK1
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc48e00] Android_Unknown_Vulkan
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc48f00] Lumin
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc49100] Lumin_Desktop
[2024.09.25-14.15.33:150][  0]LogDeviceProfile: Device profile begin destroy: [0x7f852cc49200] HoloLens
[2024.09.25-14.15.33:175][  0]LogDemo: Cleaned up 0 splitscreen connections with owner deletion
[2024.09.25-14.15.33:180][  0]LogWorld: UWorld::CleanupWorld for None, bSessionEnded=true, bCleanupResources=true
[2024.09.25-14.15.33:180][  0]LogExit: Editor shut down
[2024.09.25-14.15.33:230][  0]LogExit: Object subsystem successfully closed.
[2024.09.25-14.15.33:239][  0]LogShaderCompilers: Display: Shaders left to compile 0
[2024.09.25-14.15.33:239][  0]LogShaderCompilers: Display: Shaders left to compile 0
[2024.09.25-14.15.33:243][  0]LogModuleManager: Shutting down and abandoning module TextureAlignMode (568)
[2024.09.25-14.15.33:243][  0]LogModuleManager: Shutting down and abandoning module BspMode (566)
[2024.09.25-14.15.33:243][  0]LogModuleManager: Shutting down and abandoning module GeometryMode (564)
[2024.09.25-14.15.33:243][  0]LogModuleManager: Shutting down and abandoning module AvfMediaFactory (562)
[2024.09.25-14.15.33:243][  0]LogModuleManager: Shutting down and abandoning module AvfMediaEditor (560)
[2024.09.25-14.15.33:243][  0]LogModuleManager: Shutting down and abandoning module WmfMediaFactory (558)
[2024.09.25-14.15.33:243][  0]LogModuleManager: Shutting down and abandoning module WmfMediaEditor (556)
[2024.09.25-14.15.33:244][  0]LogModuleManager: Shutting down and abandoning module WebMMediaFactory (554)
[2024.09.25-14.15.33:244][  0]LogModuleManager: Shutting down and abandoning module WebMMediaEditor (552)
[2024.09.25-14.15.33:244][  0]LogModuleManager: Shutting down and abandoning module MediaPlayerEditor (550)
[2024.09.25-14.15.33:244][  0]LogModuleManager: Shutting down and abandoning module ImgMediaFactory (548)
[2024.09.25-14.15.33:244][  0]LogModuleManager: Shutting down and abandoning module ImgMediaEditor (546)
[2024.09.25-14.15.33:244][  0]LogModuleManager: Shutting down and abandoning module MediaCompositingEditor (544)
[2024.09.25-14.15.33:244][  0]LogModuleManager: Shutting down and abandoning module SequenceRecorder (543)
[2024.09.25-14.15.33:244][  0]LogModuleManager: Shutting down and abandoning module Persona (542)
[... 269 lines cut ...]

If I then open the custom map using the following command, the map will not open.
```
./CarlaUE4.sh
python3 config.py -m map_name
```

The following error occurred in the terminal window where you entered ./CarlaUE4.sh .
```
./CarlaUE4.sh 

4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
LowLevelFatalError [File:Unknown] [Line: 149] 
A FRenderResource was deleted without being released first!
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

I would appreciate it if you could tell me how I can solve this problem and open the custom map I made with roadrunner in carla without errors.

PatrickPromitzer, 2024-09-26:
Hi,
I guess you made a Carla package with your custom Carla map.
You can check if the map is included in you package with getting a map list.

```
python3 config.py --list
```

If your map is not includ
[image]
ed, you need to add it manually to the package option.

Ahnhojin1223, 2024-09-27:
> Hi, I guess you made a Carla package with your custom Carla map. You can check if the map is included in you package with getting a map list.
> 
> ```
> python3 config.py --list
> ```
> 
> If your map is not included, you need to add it manually to the package option. ![carla_inclute_map_to_package](https://private-user-images.githubusercontent.com/152880896/371163685-868c9d2c-8afd-46bb-bfe9-da258f200fd5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc0MTAxMTMsIm5iZiI6MTcyNzQwOTgxMywicGF0aCI6Ii8xNTI4ODA4OTYvMzcxMTYzNjg1LTg2OGM5ZDJjLThhZmQtNDZiYi1iZmU5LWRhMjU4ZjIwMGZkNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyN1QwNDAzMzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lM2M1NDBhMmFhMzgwNTE5YmEyYTQyZmU5NmMwZTg5NGU2MjI3ODFmNDlmYmFkZDNkNWEyOWE0YTdjMjZmNTZhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.adP3VOrhODfNMYrCKM3wi4oOe2TZhWiUbVadmEzZidU)

Thank you for your response.

```
python3 config.py --list
```

If you enter the above command, my custom map appears in the list, but if you import it, the above error appears.

```
python3 config.py --map map_name
```

PatrickPromitzer, 2024-09-27:
Try opening your map in the unreal editor and press play.
My next guess is, that an object/actor has an problem, and this makes Carla crash.

Unreal Engine should give you an detailed error instead of the default "Segmentation fault (core dumped)" one.

Ahnhojin1223, 2024-09-27:
> Try opening your map in the unreal editor and press play. My next guess is, that an object/actor has an problem, and this makes Carla crash.
> 
> Unreal Engine should give you an detailed error instead of the default "Segmentation fault (core dumped)" one.

Then when I open my custom map, do I have to run on the unreal engine? So far, I've opened it with the command "./CarlaUE4.sh".

PatrickPromitzer, 2024-09-30:
If you made a package, you don't need the Unreal Engine Editor, but for finding the problem with your custom map, it you get more information with the Unreal Engine Editor.

Did you make a package with "make package", or did you just copy the files to the package you can download from GitHub?
To use the map, you have to make a package with the map included or it will not work. Using the files "make import" creates will not work with the CarlaUE4.sh script.

chuanye-Wang, 2024-11-16:
> Then when I open my custom map, do I have to run on the unreal engine? So far, I've opened it with the command "./CarlaUE4.sh".

I exported the custom map, and import into a release version of carla0.9.15, sometime it accure "segmentation fault", and sometime it wont. But, it wors fine in carla UE4engine, why's that?

PatrickPromitzer, 2024-11-18:
How did you export it?
With "make package" the map will get cooked (converted into a special file format) to be able to be used in the package version of Carla.
If the map is not cooked, it will only work with the source version of Carla, because it can handle uncooked files.

PatrickPromitzer, 2024-11-26:
Hi,
I can't see a export command in the make file.

Do you mean the Unreal Engine export?
If you get out a .fbx file, it is just the map as an object.

gfresd, 2025-01-30:
Hi,
same or very similar problem here.
@chuanye-Wang
Did you solve the issue?

@PatrickPromitzer
Here are more details

Carla version: 0.9.14
OS: Ubuntu 20.04

### Brief description
- Built a custom map `sim01` with RoadRunner version R2024b, and exported the files `sim01.oxdr` and `sim01.fbx`.
- Ingested them into CARLA via `make import ARGS="--package=sim01"`. Carla was built from sources, by following [https://carla.readthedocs.io/en/0.9.14/tuto_M_add_map_source/](https://carla.readthedocs.io/en/0.9.14/tuto_M_add_map_source/)
The directory tree looks like this
>carla/Import
>|_sim01.oxdr
>|_sim01.fbx
- Map loads and works fine in the UE4 editor
- Created distribution package with `make package ARGS="--packages=sim01"`, by following [https://carla.readthedocs.io/en/0.9.15/tuto_A_create_standalone/](https://carla.readthedocs.io/en/0.9.15/tuto_A_create_standalone/)
- Imported the tar.gz package into a CARLA binary distribution of the same version (0.9.14) via `ImporAssets.sh`, following [https://carla.readthedocs.io/en/0.9.15/tuto_A_create_standalone/#import-assets-into-a-carla-package](https://carla.readthedocs.io/en/0.9.15/tuto_A_create_standalone/#import-assets-into-a-carla-package)
- Launched the binary CARLA version with  `CarlaUE4.sh`.  The python API call `get_available_maps()` shows the custom map in the list of available maps as `/Game/Carla/Maps/sim01/sim01`
- When loading the map with `load_world("sim01/sim01")`, the map is not found.
- When loading the map with `load_world("sim01")`, CARLA crashes.
A dialog box opens with the following title  _"Serialization Error: Action Needed"_ and following text _"Corrupt data found, please verify your installation"_
The terminal output from `CarlaUE4.sh` is 
`LowLevelFatalError [File:Unknown] [Line: 149] 
A FRenderResource was deleted without being released first!
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
`

PatrickPromitzer, 2025-01-30:
If the the list of available maps shows "/Game/Carla/Maps/sim01/sim01", you should use the whole string
```
client = carla.Client("127.0.0.1", 2000)
client.load_world("/Game/Carla/Maps/sim01/sim01")
```

You can try making a package with "make package" and test if it works with the newly made package.
Maybe the ARGS break something, or the package is not compatible for unknown reasons.