carla-simulator/carla #8197: The following blueprintshave unresolvedcompiler errors. BP_TLOpenDrive

OUSHAOHUA, 2024-09-26:
[image]
[image]

OUSHAOHUA, 2024-09-26:
How should I fix this problem

OUSHAOHUA, 2024-09-26:
[image]

PatrickPromitzer, 2024-09-26:
Hi,
from the name "/Temp/UEDPIE_0_Untitled_0.Untitled" it looks like a custom map.

To make a Carla map, you have the option to use the "make Import" command or the empty openDRIVE map.
Creating a new map with the Unreal Engine Editor option gives errors.

You can try going in a default Carla Map (example: the openDRIVE map) and look for actors included int he map.
Your custom map have to contain these actors.

OUSHAOHUA, 2024-09-26:
> Hi, from the name "/Temp/UEDPIE_0_Untitled_0.Untitled" it looks like a custom map.
>
> To make a Carla map, you have the option to use the "make Import" command or the empty openDRIVE map. Creating a [...]

thanks for your answer，How should I find the corresponding map?

PatrickPromitzer, 2024-09-26:
The openDRIVE map is in
```
path/to/Carla/Unreal/CarlaUE4/Content/Carla/Maps/OpenDriveMap
```
or
```
Content/Carla/Maps/OpenDriveMap
```

I can't tell where your custom map is located.