carla-simulator/carla #8240: [Strange Performance] Unusual Fluctuations in Local Trajectories while Running Apollo in Carla

AOOOOOA, 2024-10-04:
Hi,

I'm running Apollo(0.9.8) in Carla(0.9.14) using the official bridge provided by guardstrikelab ([GitHub link](https://github.com/guardstrikelab/carla_apollo_bridge)). I attempted to simulate a realistic driving scenario by adding multiple vehicles in Carla alongside Apollo, using the generate_traffic.py script from the PythonAPI/examples folder. Specifically, I added 20 vehicles and 20 walkers, setting them to safe mode to avoid collisions. Carla was configured in synchronous mode with a frame time of 0.05.

However, after adding the additional vehicles and walkers, I noticed unusual fluctuations in Apollo's local trajectory. This issue rarely occurs when running Apollo alone in Carla. Most of the time, Apollo manages to correct itself and return to the proper path, but occasionally it gets stuck.

Is this behavior expected?

I've attached a sample video below for reference.

https://github.com/user-attachments/assets/80540900-b938-4f17-91d0-29f6c5124fba

PatrickPromitzer, 2024-10-08:
Hi,
I can't say how Apollo works with Carla, but try 0.1 for the frame time.

With more Actors, sensors, and objects (in the map), the simulation is getting slower.
My guess is, Apollo is getting confused if a frame is missing or to late.

PatrickPromitzer, 2024-10-08:
The common way is using the Synchronous mode.
In this mode Carla is waiting at each step until it gets the permission to run the next step.
For example:
1.) Carla is waiting
2.) code is running
3.) you send a command to give Carla the permission to tick
4.) Carla is running for 0.1 seconds
5.) repeat 1-4

There are examples for this process, and even if you can't get it to real time, you have your component synchronized.
With this mode, it doesn't care if the sensor take 0.001 seconds or 10.0 seconds, because you make Carla wait until you did everything you need it t do.
https://carla.readthedocs.io/en/0.9.15/adv_traffic_manager/#synchronous-mode
https://github.com/carla-simulator/carla/blob/ue4-dev/PythonAPI/examples/sensor_synchronization.py

I didn't work with the ROS part of Carla, but maybe there is a synchronous-mode for the ROS side, too.
It looks less like a bug, and more like very simple solution to make it work. I can't say if it is a good or bad solution because I didn't try it.
If the ROS part doesn't support the synchronous-mode (It would be strange if it isn't supported, because it should be possible), it would be better for you to have an python code as an interface between the Apollo and Carla instance.

AOOOOOA, 2024-10-15:
I've already set Carla to run in synchronous mode from the start, with a default step time of 0.05 seconds. Apollo and Carla communicate via CyberRT, which is a ROS-like framework. In this setup, Apollo publishes control commands, and Carla is supposed to receive and execute them. 

Since Apollo's control module updates at a high frequency (50 Hz in this case), it continuously sends control commands to Carla. In the Carla-Apollo-Bridge, there's a callback function that gets triggered every time Apollo publishes a control command. This function should then call vehicle.apply_control() with the updated command. According to the Carla documentation, the simulator will execute the last control command it receives before the next simulation tick.

The issue arises when multiple NPC actors are generated in Carla's world. When this happens, Carla becomes too busy to handle the callback function in a timely manner. CyberRT, which uses shared pointers to access the control command data, can get overloaded if Carla doesn't process the commands quickly enough. This leads to the data buffer in the CyberRT topic filling up, causing older control commands to be dropped. Over time, this accumulation of missed commands results in the Apollo vehicle behaving incorrectly.

I've noticed that there doesn't seem to be a timestamp checking mechanism in Carla, which makes it difficult for Carla to determine whether the control data it receives is the most up-to-date. I wonder if this could be contributing to the issue.

I'm also curious whether such a mechanism also exists in the traffic manager module, or if there are other important pieces of information that might also suffer from timeliness issues. If that's the case, it could lead to various performance problems, particularly when the system is handling a high workload.

AOOOOOA, 2024-10-15:
> The common way is using the Synchronous mode. In this mode Carla is waiting at each step until it gets the permission to run the next step. For example: 1.) Carla is waiting 2.) code is running 3.) [...]

I've already set Carla to run in synchronous mode from the start, with a default step time of 0.05 seconds. Apollo and Carla communicate via CyberRT, which is a ROS-like framework. In this setup, Apollo publishes control commands, and Carla is supposed to receive and execute them. 

Since Apollo's control module updates at a high frequency (50 Hz in this case), it continuously sends control commands to Carla. In the Carla-Apollo-Bridge, there's a callback function that gets triggered every time Apollo publishes a control command. This function should then call vehicle.apply_control() with the updated command. According to the Carla documentation, the simulator will execute the last control command it receives before the next simulation tick.

The issue arises when multiple NPC actors are generated in Carla's world. When this happens, Carla becomes too busy to handle the callback function in a timely manner. CyberRT, which uses shared pointers to access the control command data, can get overloaded if Carla doesn't process the commands quickly enough. This leads to the data buffer in the CyberRT topic filling up, causing older control commands to be dropped. Over time, this accumulation of missed commands results in the Apollo vehicle behaving incorrectly.

I've noticed that there doesn't seem to be a timestamp checking mechanism in Carla, which makes it difficult for Carla to determine whether the control data it receives is the most up-to-date. I wonder if this could be contributing to the issue.

I'm also curious whether such a mechanism also exists in the traffic manager module, or if there are other important pieces of information that might also suffer from timeliness issues. If that's the case, it could lead to various performance problems, particularly when the system is handling a high workload.

PatrickPromitzer, 2024-10-15:
In a normal case this should now be a problem.

With world.tick() Carla is updating the simulation, and if you want to get Actor data, like the position of every vehicle, it should be up to date after calling the world.tick() function.

There a 2 types of sensors.
- sending data each step
- sending data after an event

I only found 2 sensors which doesn't send data each step.
- sensor.other.obstacle - sending data if an Actor was detected
- sensor.other.lane_invasion - sending data if a lane was crossed

Every other sensor will send data to the python API (probably ROS, too) each step. With this information, you can use a Queue and wait until the data arrived.

```python
from queue import Queue

sensor_transform = carla.Transform(
    carla.Location(
        x=0.0,
        y=0.0,
        z=3.0),
    carla.Rotation(
        pitch=0.0,
        yaw=0.0,
        roll=0.0))

# connect to server
client.set_timeout(2.0)](client: carla.Client = carla.Client("127.0.0.1", 2000))

# set Carla to sync mode
settings = world.get_settings()
settings.synchronous_mode = True
settings.fixed_delta_seconds = 0.1
world.apply_settings(settings)

# get camaera sensor
world: carla.World = client.get_world()
bp_lib = world.get_blueprint_library()
cam_sensor_bp = bp_lib.find("sensor.camera.rgb")

#spawn camera sensor
sensor_cam = world.try_spawn_actor(cam_sensor_bp, sensor_transform)

#add camera listener
sensor_queue = Queue()
sensor_cam.listen( lambda item: sensor_queue.put(item)) # putting the data in the queue if Carla send one

while True:
    # tell Carla to run for a step
    world.tick()
    
    # waiting for the date to arrive
    sensor_data =  sensor_queue.get() 
    
    # this point is only reached if the data of the sensor has arrived 
``` 
With this setup you have 2 things.
- The script is telling Carla when it should run "world.tick()"
- The script waits for the sensor data of Carla and doesn't continue if no sensor data reached the script "sensor_queue.get()"

The only problem which could happen is, if more than one Client is using the "world.tick()" function.
If more than one client is using the "world.tick()" function, you lose the synchronous aspect, because both clients think they control the timing of Carla, and two ticks happen while only one should.

I have 2 guesses in your situation.
1.) one Client tries to keep the simulation real time, and because of that reason it skip frames
2.) Two clients try to use "world.tick()" and because of that reason Carla is running as fast as possible, which make it lose the synchronous aspect

In both cases it can get to the issue you have, but it could be something else I didn't thought about, too.

AOOOOOA, 2024-10-15:
> Quote reply
> Reference in new

1. I’m carefully managing the synchronization process. I generate multiple NPCs using generate_traffic.py through a separate client, but I rely solely on the Carla_Apollo_bridge client to control synchronization via world.tick(). In generate_traffic.py, I modified the command to world.wait_for_tick() to prevent the scenario where two clients attempt to tick the world simultaneously. Therefore, I don’t believe this is the cause of the issue.
2. There seems to be some misunderstanding. In the example code you provided, the world appears to be ticked only when sensor data arrives. However, in my case, the issue stems from a large volume of control data continuously coming from Apollo in the whole simulation process, which Carla needs to process. The update frequency of this control data is higher than Carla's simulation frequency. While it's expected that Carla may not handle every single piece of control data, as the ADS control module sends data at a very high frequency, missing too much control data is undesirable.
Even when I use a queue after world.tick(), the issue persists because Carla becomes overwhelmed by other tasks and doesn’t have "enough" time to process all the incoming control data. As a result, older control data continues to be dropped from the queue.

I tried increasing Carla's simulation frequency, but this caused the system to behave erratically. Conversely, when I decreased the simulation frequency, it led to a new problem where control commands were over-executed, an issue that has been reported by other users as well.

It seems that this issue could be resolved if Carla had an extra message-checking mechanism for handling large-scale actors and high-frequency client communication. At this point, I’m inclined to consider this a potential bug. Would you need a more specific description or code to reproduce it?

PatrickPromitzer, 2024-10-15:
I would need to set everything up with Apollo to test it myself (currently I have too less time for that).
I am only using Carla and not with the development team, but still try to find a solution to some problems.

I noticed with the python API, if you set the position to an Actor, it will get an timeout error if the server doesn't answer.
I believe it would be the same for updating the control data of the Actors.

I never tested a synchronization loop without sensor data, but if the ```world.tick()``` doesn't wait until the Carla server is done with updating the requests and simulation step, it sounds like bug. I can make a short script to test it, but I can't say when I have time for it.

I don't know how the Apollo bridge works (probably ROS like), but if you can make the bridge wait for a tick, as a quick fix I suggest adding a 3th client and let this one tick very slow. It is not a "solution", but you may get at least some data (slowly).

AOOOOOA, 2024-10-15:
> I would need to set everything up with Apollo to test it myself (currently I have too less time for that). I am only using Carla and not with the development team, but still try to find a solution [...]

Thank you for your advice! I initially thought you were part of the development team due to your active presence in the Carla community. I'll explore some potential solutions to address the issue in the meantime, and we can also wait for the development team to confirm whether it's a bug. Once again, thank you so much for your help!

Blyron, 2024-10-18:
We are trying to answer as much issues as possible, Co-Simulation related ones takes much time and we are on other stuff so we cannot have enough time to research this now days. We are gonna try to look at this. Try to ask at Apollo's support

PatrickPromitzer, 2024-10-18:
I wrote a small script to test if Carla has a problem with the world.tick() update and timing.
I used the python Carla package.

```python
def test_carla_steping(server_address):

    start_position = {
        "x": 100.0,
        "y": -40.0,
        "z": 0.5
    }
    cars_amount = 25
    step_length = 28
    x_offset = 10.0
    y_offset = 0.0

    x_move = 0.0
    y_move = -10.0
    max_distance = 1.0

    client = carla.Client(server_address, 2000)
    world = client.get_world()
    original_world_settings = world.get_settings()
    original_world_settings.synchronous_mode = False

    world_settings = world.get_settings()
    world_settings.fixed_delta_seconds = 0.1
    world_settings.synchronous_mode = True
    world.apply_settings(world_settings)
    world.tick()
    blueprint_lib = world.get_blueprint_library()
    vhicle_bb = blueprint_lib.filter("vehicle.*")[0]

    vehicle_list = []
    for i in range(cars_amount):
        carla_loctaion = carla.Location(
            x=start_position["x"] + (i * x_offset),
            y=start_position["y"] + (i * y_offset),
            z=start_position["z"]
        )
        vehicle_actor = world.spawn_actor(vhicle_bb, carla.Transform(carla_loctaion))
        vehicle_list.append(vehicle_actor)
    error_list = []
    for i in range(step_length):
        world.tick()
        for index, vehicle_actor in enumerate(vehicle_list):
            location = vehicle_actor.get_location()
            position_is_ok = True
            distance_x = location.x - (start_position["x"] + (index * x_offset) + (i * x_move))
            distance_y = location.y - (start_position["y"] + (index * y_offset) + (i * y_move))
            if distance_x > max_distance or distance_x < -max_distance:
                position_is_ok = False
            if distance_y > max_distance or distance_y < -max_distance:
                position_is_ok = False
            if position_is_ok:
                error_list.append(0)
            else:
                error_list.append(1)
            location.x = start_position["x"] + (index * x_offset) + ((i + 1) * x_move)
            location.y = start_position["y"] + (index * y_offset) + ((i + 1)  * y_move)
            vehicle_actor.set_location(location)

    print(sum(error_list))

    for vehicle_actor in vehicle_list:
        vehicle_actor.destroy()
    world.apply_settings(original_world_settings)
```
If you want to run this script yourself, you need an empty place on a Map.

I got 0 errors if I keep the max_distance at 1.0, but if I get lower than 0.01, I start getting offsets. They probably come from the Vehicles rolling away.
I tested it with ```world_settings.fixed_delta_seconds = 0.001``` and still have no errors.

AOOOOOA, 2024-10-25:
> We are trying to answer as much issues as possible, Co-Simulation related ones takes much time and we are on other stuff so we cannot have enough time to research this now days. We are gonna try to look at this. Try to ask at Apollo's support

Sure, I will report the bug to Apollo and Carla-Apollo-Bridge. However, based on our findings, the issue seems to stem from Carla's lack of a precise timestamp determination mechanism. I would really appreciate it if you could take a closer look at this, as it appears to be a critical aspect affecting the integration.

Blyron, 2024-10-25:
You can use simulation time as time stamp as our recorder/replayer does. So we have a way to determine the time stamps. I cannot help you more there

AOOOOOA, 2024-10-25:
> You can use simulation time as time stamp as our recorder/replayer does. So we have a way to determine the time stamps. I cannot help you more there

What I’m trying to convey is that Carla currently lacks a built-in mechanism to ensure that the control data it receives is always the most up-to-date. As a result, it processes whatever data arrives, which can occasionally cause unexpected behavior in the Autonomous Driving System (ADS), particularly when Carla is under significant computational load. While I could implement a mechanism to filter outdated data myself, this issue tends to become a hidden bug when scaling up to a large number of vehicles within the Carla environment. I suspect that many users have already experienced this issue without realizing it, and while it may seem subtle, it could lead to performance inconsistencies that would benefit from further consideration.

Blyron, 2024-10-25:
We will review this in the future. Right now we are dealing with UE5 migration. Cannot estimate an ETA for this.

AOOOOOA, 2024-11-21:
> We will review this in the future. Right now we are dealing with UE5 migration. Cannot estimate an ETA for this. If someone implements this; We are open to contributions.

I understand that the team is currently focused on the UE5 migration, and I really appreciate all the hard work going into that. That being said, I believe this joint simulation issue is quite important and worth addressing. Simulating multiple vehicles is a common use case for both Carla and Apollo, and this issue directly impacts the ability to scale simulations for realistic scenarios—something that’s essential for many users, including myself.

I’ve also noticed that a few others in the community have reported similar problems, so it seems this might not be an isolated case. If the issue continues, it could limit the usability of Carla-Apollo joint simulations for large-scale testing, which might affect ongoing projects or even discourage users from relying on this workflow.

I understand it’s not always easy to tell right away whether something is a bug or a configuration issue. If possible, could you suggest any specific diagnostics or settings I could try to help narrow down the root cause? I’m more than happy to provide additional logs, run further tests, or help out in any way I can to move this forward.

AOOOOOA, 2024-11-21:
> I will pass this into the correct people to see if they are able to help you

Thank you so much for all your effort. I truly appreciate it.

AOOOOOA, 2024-11-25:
Do you mean Synkrotron.ai, the developer of the Carla-Apollo-Bridge project (https://github.com/guardstrikelab/carla_apollo_bridge)?

I’ve already tried reaching out to them, but unfortunately haven’t received a reply yet. Have you had a chance to test this issue? Or could you possibly point me to someone I can directly contact? I suspect there might be some settings that still need to be configured.

AOOOOOA, 2024-11-25:
> No, we have contacted with the breach developers as in CARLA everything seems fine

Do you mean Synkrotron.ai, the developer of the Carla-Apollo-Bridge project (https://github.com/guardstrikelab/carla_apollo_bridge)?

I’ve already tried reaching out to them, but unfortunately haven’t received a reply yet. Have you had a chance to test this issue? Or could you possibly point me to someone I can directly contact? I suspect there might be some settings that still need to be configured.

Blyron, 2024-11-25:
Our developers tested, and see the issue, but in CARLA end seems everything to be ok.

Blyron, 2024-11-25:
And Yes, I mean Synkrotron, they are our partners so I have contacted them. I cannot point anyone to contact just wait issues answers on github

AOOOOOA, 2024-11-25:
> Our developers tested, and see the issue, but in CARLA end seems everything to be ok.

Yes, on the Carla side, everything seems fine. However, after running Apollo for a while with multiple vehicles in the Carla environment, some misbehavior occurs in Apollo. Upon further investigation, I found that the issue originates from Carla being too busy handling the environment vehicles, which prevents it from processing control data from the Apollo control topic in a timely manner.

As a result, there is a control data overflow in the Apollo control topic (not a traditional memory buffer overflow). The Apollo control topic drops the corresponding control data, and when Carla retrieves the control data, it only takes the latest data rather than the most appropriate ones. Since Carla lacks a timestamp checking mechanism, this discrepancy accumulates over time, leading to the observed misbehavior in Apollo.

Are you suggesting waiting for Synkrotron to respond on this issue, or should I report it directly on the Carla Apollo Bridge repository?

Blyron, 2024-11-25:
Report directly in apollo bridge.

AOOOOOA, 2024-11-25:
> Report directly in apollo bridge.

I reported this issue to the bridge repository a month ago but haven’t received any response yet. It seems that the repository hasn’t been actively maintained recently.

XiaoFei9704, 2024-11-26:
@AOOOOOA You can try to modify the frequency of the planning module:
modules/planning/planning_base/gflags/planning_gflags.cc -> planning_loop_rate
We can let CARLA limit the frequency of Apollo's control output by sending ticks to Apollo through CARLA Server, but this requires modifying Apollo. In order to adapt to the native Apollo, we later abandoned this solution.

AOOOOOA, 2024-11-26:
> @AOOOOOA You can try to modify the frequency of the planning module: modules/planning/planning_base/gflags/planning_gflags.cc -> planning_loop_rate We can let CARLA limit the frequency of Apollo's [...]

Thank you for your detailed response and for suggesting the workaround regarding modifying the frequency of the planning module. I will try this solution and see if it helps resolve the issue.

That being said, this behavior seems to cause unexpected control issues in Apollo when working with CARLA, especially under high-load scenarios with multiple vehicles in the environment. It creates noticeable performance degradation and affects the accuracy of control commands, which has a significant impact on the overall system's stability.

Would it be reasonable to classify this as a bug, or is it more of a design limitation due to the adaptation to native Apollo? I'm asking because this issue appears to deviate from expected behavior and might require further adjustments to ensure reliable performance in similar scenarios.

Once again, thank you for your support. I'll share any feedback after testing the workaround.

yuxip, 2024-11-27:
I would consider this as a design limitation of the native Apollo, i.e., it does not allow external triggering for synchronization purpose. Take for example, when Carla is only capable of updating sensor data at 10Hz, it does not make any sense for Apollo to run planning at 50Hz (generates 4 useless plans in between sensor updates). Slowing down Apollo would be the way to go here.

I am curious as to why Apollo is able to run so fast in your experiment. How many sensors/cameras did you configure in Carla? Does your Apollo instance include the perception module or is it only running in Planning and Control mode?

AOOOOOA, 2024-11-27:
> I would consider this as a design limitation of the native Apollo, i.e., it does not allow external triggering for synchronization purposes. Take for example, when Carla is only capable of updating sensor data at 10Hz, it does not make any sense for Apollo to run planning at 50Hz (generates 4 useless plans in between sensor updates). Slowing down Apollo would be the way to go here.
> 
> I am curious as to why Apollo is able to run so fast in your experiment. How many sensors/cameras did you configure in Carla? Does your Apollo instance include the perception module or is it only running in Planning and Control mode?

I just use the default setting of Apollo 8.0, as advised in the Carla-Apollo-Bridge documentation, the simulation includes not only the Planning and Control (PnC) modules but also the Perception modules. This setup is used for co-simulation with Carla, where the vehicle state is updated by Carla.

**Why the Control Module Runs Faster**
From my understanding , the Control Module in an ADS typically runs at a much higher frequency than other modules (e.g., Perception or Planning). This is because:
1. Vehicle dynamics (speed, direction, posture, etc.) change rapidly, requiring frequent adjustments to throttle, braking, and steering.
2. High-frequency updates allow precise trajectory tracking, stability, and safety.

To achieve this high frequency, while the other modules update at a much slower rate, multiple methods are used:

1. Trajectory Interpolation: The control module interpolates the planned trajectory to generate a denser set of waypoints, ensuring smooth and continuous updates.
2. Predicted Vehicle State: When the actual vehicle state (e.g., position, speed) updates slowly, a model-predicted state is calculated to provide a "fake" real-time input for the control module to compute control commands.
3. High-Frequency Vehicle State Updates: In real-world vehicles, the state (e.g., speed, acceleration, and position) is updated at a high frequency, which allows the control module to respond quickly to changes and adjust precisely.

**Limitations in Carla-Apollo Co-Simulation**
In the Carla-Apollo co-simulation, the vehicle state is updated by Carla. Since Carla’s default frequency is 20Hz, the vehicle state updates are much slower than in a real-world vehicle. This creates a challenge for Apollo because the Control Module typically requires much higher update frequencies.

Apollo’s Control Module relies heavily on predicted vehicle states (calculated using a vehicle dynamics model) to generate new control commands. This dependency on predictions can introduce errors, especially in dynamic situations, as the control is not based on real-time updates. Another significant source of error arises from the issue described here: when multiple vehicles exist in the environment, Carla becomes busy handling them and fails to process control commands from Apollo in a timely manner. This delay causes the vehicle to execute outdated or unsuitable control commands once Carla resumes processing them.

**Design Limitation**
I believe this issue reflects a limitation of the Carla-Apollo-Bridge, and more fundamentally, a limitation of Carla itself. Ideally,  A driving simulator like Carla should serve the autonomous diving systems by matching the needs of them

Slowing down Apollo to match Carla’s frequency would introduce new errors and reduce the accuracy of the control module. Therefore, the problems above remain a bottleneck in co-simulation.

yuxip, 2024-11-28:
I agree control module usually runs faster than perception and planning. But Carla's actual frame rate is simply limited by your machine's compute power. E.g., with a single machine it simply cannot keep up with the 50Hz sensor update in realtime if you have many sensor/camera installed.

However, I think you misunderstood the definition of Carla's default frequency. 20Hz, or 0.05s simulation step is in "**simulation time**", the actual frame rate / frequency is determined by your hardware capability.  You can set simulation step to 0.02s but with a less powerful machine it can take much more than 0.02s to calculate a frame, thus the actual frame rate could be less than 50Hz which will not match Apollo's planning frequency. This is not a limitation of Carla but a limitation of compute hardware.

yuxip, 2024-11-28:
In a Hardware-in-the-loop (HiL) testing scenario, we usually deploy Carla across multiple machines to achieve ~30Hz frame rate in real-time via distributed rendering. The vehicle dynamics model (PhysX or CarSim) will be deployed in a real-time machine to achieve 100-1000Hz compute frequency to provide more dense inputs to the ADS controller (e.g. could be running Apollo). It is only with this type of setup can we expect to work with Apollo's realtime requirement.

In a Software-in-the-loop (SiL) scenario Carla usually runs in a single machine, one cannot expect that with multi-sensor configuration Carla could match the realtime requirement of any ADS (it's simply outside the scope of SiL testing). In this scenario a proper synchronization between the simulator and the ADS is all we can ask for. Carla can be synchronized via ticks, but native Apollo (without modification) does not accept such mechanism.

AOOOOOA, 2024-11-29:
> I agree control module usually runs faster than perception and planning. But Carla's actual frame rate is simply limited by your machine's compute power. E.g., with a single machine it simply cannot [...]

Does this mean that the simulation time in Carla is more of a theoretical value rather than an exact, guaranteed one? For example, if I set the time-step to 0.02s in synchronous mode, would it be correct to say that on a less powerful machine, the actual time-step might become larger than 0.02s due to computational limitations? If so, am I understanding correctly that the synchronous mode may not always maintain a perfectly consistent step length in such cases? Could you clarify how this is handled?

yuxip, 2024-12-02:
You can read about simulation time step [here](https://carla.readthedocs.io/en/latest/adv_synchrony_timestep/). The synchronous mode only guarantees that a constant time (e.g., \delta_t_1 = 0.02s) is stepped through in the **simulated world**，and Carla server takes the next step only when being ticked by a client. As for how much actual time (\delta_t_2) it takes between adjacent frames, it really depends the compute power of your machine. This is why we talked about super-real-time (when \delta_t_1 < \delta_t_2), real-time (\delta_t_1 == \delta_t_2) and non-real-time (\delta_t_1 > \delta_t_2)

AOOOOOA, 2024-12-11:
> You can read about simulation time step [here](https://carla.readthedocs.io/en/latest/adv_synchrony_timestep/). The synchronous mode only guarantees that a constant time (e.g., \delta_t_1 = 0.02s) [...]

Sorry for the late response. 

I understand your point that simulation time and real computation time are not equivalent. However, this discrepancy introduces significant challenges when running an autonomous driving system (ADS), such as Apollo, in the simulator.

The first and most straightforward impact is that the update frequency of critical modules like localization, vehicle state, and control commands cannot meet the frequency requirements of the ADS. While this is not ideal, it is somewhat acceptable as long as the ADS can still function normally.

The bigger issue arises from the computation time limitation, which prevents Carla from processing and applying the corresponding control command in a timely manner. Since the control module in the ADS continuously computes the latest control command based on the predicted vehicle location and state (a fundamental principle of ADS control), the control command that Carla ends up executing in the simulation is not the intended one but the most recently computed command. This results in Carla updating the ADS vehicle in the simulated world using an outdated vehicle state and a mismatched control command. The longer the real computation time, the greater the discrepancy between the outdated vehicle state and the executed control command.
Over time, especially in scenarios with complex road conditions, this mismatch accumulates and amplifies, leading to erroneous driving behaviors, as I previously reported.

While I agree that using more powerful hardware can alleviate this issue to some extent, I don’t believe it should be considered the ultimate solution. Instead, I strongly suggest implementing additional mechanisms, such as timestamp validation in the Carla-Apollo Bridge or within the simulator itself, to address this problem.

One of the key purposes of a simulator is to serve the ADS, whether for algorithm development, testing, or other use cases. If this issue persists, it will severely hinder the simulator’s usability and its effectiveness for ADS development. Therefore, I believe it is critical to treat this as a bug rather than simply a limitation to be resolved by increasing computational resources.

AOOOOOA, 2024-12-12:
> I see a flag in the generate_traffic.py script, where it will use world.wait_for_tick() or world.tick()

I set the generate_traffic.py as in async mode. Thus, it uses the world.wait_for_tick(), it is not the one controlling the simulation.

PatrickPromitzer, 2024-12-13:
If Apollo is the only client who sends a world.tick(), the Apollo endpoint doesn't call the world.tick() the right way.
It could be that the Carla ROS bridge has an error, but under normal condition, the bug should be in the code of the Carla-Apollo-bridge and  not in Carla.

Apollo should wait until the simulation is updated, but for some reasons it doesn't.

The ROS bridge even has an status topic
https://carla.readthedocs.io/projects/ros-bridge/en/latest/run_ros/#ros-api
/carla/status 	[carla_msgs/CarlaStatus](https://carla.readthedocs.io/projects/ros-bridge/en/latest/ros_msgs/#carlastatusmsg) 	Read the current status of CARLA

Saying it is a Apollo limitation doesn't sound right.
If the Apollo bridge only work if Carla_work_time<Apollo_work_time, it is not implemented right or just a test bridge.

In the real car with a Lidar sensor, you could have log times like this one.
log time Lidar message with a frequency of 10Hz: 
```
runtime (offset from optimal 0.1s time between steps)
0.100s (+-0.000s)
0.205s (+0.005s)
0.390s (+0.090s)
0.410s (+0.010s)
0.550s (+0.050s)
0.602s (+0.002s)
```
 
There is no exact frequency, and a delay close to one step could happen.
In that case, Apollo should have a way to handle this situation.

Can we come to the same conclusion and say it is a Apollo problem and not a Carla one?

AOOOOOA, 2024-12-16:
> If Apollo is the only client who sends a world.tick(), the Apollo endpoint doesn't call the world.tick() the right way. It could be that the Carla ROS bridge has an error, but under normal [...]

The topic /carla_msgs/CarlaStatus only provides the status of the current frame (e.g., frame ID and sync settings), but it doesn’t indicate whether the current simulation frame has finished or not. Do you mean we can use the information from this topic to determine if the simulation has moved to the next frame?

There are also some key differences between real-world conditions and the simulation environment. In the real world, I agree with your point that Apollo should be capable of handling delays in data, as these delays are unavoidable. However, in a simulation environment, the simulator or the bridge has the opportunity to actively manage synchronization issues. 

Furthermore, since the simulator is designed as a universal tool to work with different autonomous driving agents (not just Apollo), it should take responsibility for managing synchronization to ensure the accuracy of the simulation. Relying on the autonomous driving systems to adapt to specific design limitations of the simulator may not be ideal. By taking on this responsibility, Carla or the bridge could improve its robustness and better serve other autonomous driving agents as well.

So from a general perspective, I don’t think this should be considered an issue for autonomous driving systems. Instead, it’s something that Carla or the bridge should address or optimize.

AOOOOOA, 2024-12-16:
> You can read about simulation time step [here](https://carla.readthedocs.io/en/latest/adv_synchrony_timestep/). The synchronous mode only guarantees that a constant time (e.g., \delta_t_1 = 0.02s) [...]

Here raises another question. If the synchronized time step is set to 0.05s, but due to computational power limitations, the actual computation time between frames is 0.1s, what would the computed vehicle state in the simulator represent? Would it be the state after 0.05s (as per the configured simulation time step) or after 0.1s (reflecting the actual computation delay)?

PatrickPromitzer, 2024-12-16:
> The topic /carla_msgs/CarlaStatus only provides the status of the current frame (e.g., frame ID and sync settings), but it doesn’t indicate whether the current simulation frame has finished or not. [...]

The status of the current frame is what should tell you if the frame is finished. The information If the sensor data finished transferring is not included. You should get the sensor data each frame (if you set it up that way), and the client should know that.
If the client ignores all of that information and still keeps running, you will always have a problem if the simulation is not faster than the client. 
 
```
/carla/status
CarlaStatus.msg
frame  	Current frame number
```
If you get the status message with the new frame number, and the simulation is not updated, we can talk say it is a Carla bug.
If you get the old frame number and ignore it, it is a client fault.

> Furthermore, since the simulator is designed as a universal tool to work with different autonomous driving agents (not just Apollo), it should take responsibility for managing synchronization to [...]

That would not make much sense.
For example:
If a Client tells Carla to step 3 times, and doesn't want Carla to step 3 time, you have conflict from the Client part.
The simulation can only make sure that the Simulation is doing the synchronization right. If you want the client to work with the synchronization, it has to do it the right way.

An real life example would be the following:
You want to order 1 apple each day, and you will get them the day after.
You order 1 apple in the morning, 1 apple at noon, and one apple at the evening.
On the day after, you wonder why you have 3 apples because you only wanted 1 apple, and give the supplier the fault.
The supplier don't know if you want 1 or 3 apples, and has no option to know if you order them wrong.
You could make sure you ordered an apple and wait until it arrived, but if you don't do that and just order more, it should be no surprise that you don't get what you wanted.

AOOOOOA, 2024-12-17:
> The status of the current frame is what should tell you if the frame is finished. The information If the sensor data finished transferring is not included. You should get the sensor data each frame [...]

1. **The root cause of this issue is not the misalignment of sensor input data but rather the mismatch between the update frequencies of Carla and Apollo.** Specifically, Carla updates too slowly while Apollo's control module updates much faster. The control module continuously sends control commands to Carla, but Carla only executes the **last received command** at each simulation frame update, ignoring all previous commands. With a long computation time between frames, this results in a significant mismatch between the executed control command and the actual vehicle state in Carla, leading to noticeable discrepancies in the simulation.

> An real life example would be the following: You want to order 1 apple each day, and you will get them the day after. You order 1 apple in the morning, 1 apple at noon, and one apple at the evening. [...]

2. **In real-world autonomous driving systems (ADS), it is common for different modules to operate at different frequencies, but there are mechanisms in place to address these discrepancies**. For example:

- Modules synchronize using timestamps to ensure time alignment.
- Control command queues and delay-tolerance mechanisms are used to discard outdated commands.
- Interpolation and prediction techniques are applied to smooth out discrepancies.

The frequency mismatch between Carla and Apollo is analogous to situations that occur in real-world systems. However, real vehicles handle these discrepancies effectively:

- In real vehicles, the hardware and control modules share a common time reference.
- The hardware often includes built-in buffering and interpolation mechanisms, ensuring that when commands are received too quickly, they are not simply overridden by the latest command.
- The hardware is designed to respond to control commands as quickly as possible, with very low execution latency, reducing the kind of delay and mismatch we observe between Carla and Apollo.

3. **Apollo’s mechanisms are fundamentally designed for real-world vehicles, with the assumption that the execution environment provides low-latency and high real-time feedback.** As a general-purpose simulation tool, Carla’s responsibility is to simulate the physical behavior of the vehicle and the execution environment. However, when coupled with ADS like Apollo, we can see that Carla lacks several critical synchronization mechanisms:

- **Time-stamp management:** Carla directly executes the latest control command without considering whether it aligns with the current vehicle state.
- **Buffering and interpolation:** Carla does not buffer or interpolate between received control commands, which leads to abrupt state mismatches and command discontinuities.
- **Time alignment:** Carla does not offer proper time alignment for control commands, resulting in significant discrepancies between the control commands and the vehicle state.
Additionally, Carla’s simulation frequency and time step do not match Apollo’s control module frequency, which further exacerbates the problem. These issues should be addressed by Carla, not Apollo.

4. **This is not just about Apollo—it’s about other ADS and ADAS clients as well.** Carla is a general-purpose simulation tool designed to work with a wide range of clients. You cannot expect every client to modify their logic to adapt to Carla's limitations. Instead, it is Carla's responsibility to improve its internal mechanisms to ensure compatibility with diverse clients. By implementing timestamp management, buffering, and synchronization mechanisms, Carla could better align with the requirements of various ADS and ADAS systems, including but not limited to Apollo.

AOOOOOA, 2024-12-25:
> > ```
> > 1. **The root cause of this issue is not the misalignment of sensor input data but rather the mismatch between the update 
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > frequencies of Carla and Apollo.** Specifically, Carla updates too slowly while Apollo's control module updates much faster. The
> > control module continuously sends control commands to Carla, but Carla only executes the **last received command** at each
> > simulation frame update, ignoring all previous commands. With a long computation time between frames, this results in a
> > significant mismatch between the executed control command and the actual vehicle state in Carla, leading to noticeable
> > discrepancies in the simulation.
> 
> Only executing the **last received command** is how it should be. Why would you want to execute an old command that is outdated? If Apollo is sending more control update data than the simulation can handle, it would do the same in a real car. You can say that you have all kinds of tools in the car that handles problematic behavior like time mismatch or delayed control updates, but this are tools someone created for the car. The main different of Carla and a real car is, that you can send and receive at the same time with a delay.
> 
> > ```
> > 2. **In real-world autonomous driving systems (ADS), it is common for different modules to operate at different frequencies, but there are mechanisms in place to address these discrepancies**. For example:
> > 
> > 
> > * Modules synchronize using timestamps to ensure time alignment.
> > 
> > * Control command queues and delay-tolerance mechanisms are used to discard outdated commands.
> > 
> > * Interpolation and prediction techniques are applied to smooth out discrepancies.
> > ```
> 
> * you have timestamps in Carla for each sensor and for the current world information
> * software on a hardware
> * software
> 
> > The frequency mismatch between Carla and Apollo is analogous to situations that occur in real-world systems. However, real vehicles handle these discrepancies effectively:
> > ```
> > * In real vehicles, the hardware and control modules share a common time reference.
> > 
> > * The hardware often includes built-in buffering and interpolation mechanisms, ensuring that when commands are received too quickly, they are not simply overridden by the latest command.
> > 
> > * The hardware is designed to respond to control commands as quickly as possible, with very low execution latency, reducing the kind of delay and mismatch we observe between Carla and Apollo.
> > ```
> 
> * and for the simulation you should use simulation time, because there is no real time
> * special software
> * special hardware
> 
> Everything you listed is a software that manages problematic behavior or special hardware. Add every part of the software you listed into a Carla bride, you you have what you want. A software on a hardware is the same as a bridge between Carla and Apollo. You get raw data, convert/preprocess the data, and send it to the next part. The same with controlling the car.
> 
> To get the "very low execution latency" you need better hardware. Use a Nokia 3210 hardware to convert Lidar data, and it would take too long if it would even be possible. Carla is not a low power usage software, and if you want a faster calculation time, you would need a bigger hardware.
> 
> > ```
> > 3. **Apollo’s mechanisms are fundamentally designed for real-world vehicles, with the assumption that the execution environment provides low-latency and high real-time feedback.** As a general-purpose simulation tool, Carla’s responsibility is to simulate the physical behavior of the vehicle and the execution environment. However, when coupled with ADS like Apollo, we can see that Carla lacks several critical synchronization mechanisms:
> > 
> > 
> > * **Time-stamp management:** Carla directly executes the latest control command without considering whether it aligns with the current vehicle state.
> > 
> > * **Buffering and interpolation:** Carla does not buffer or interpolate between received control commands, which leads to abrupt state mismatches and command discontinuities.
> > 
> > * **Time alignment:** Carla does not offer proper time alignment for control commands, resulting in significant discrepancies between the control commands and the vehicle state.
> >   Additionally, Carla’s simulation frequency and time step do not match Apollo’s control module frequency, which further exacerbates the problem. These issues should be addressed by Carla, not Apollo.
> > ```
> 
> And that is not a Carla problem.
> 
> * **Time-stamp management** and **Buffering and interpolation:**: Like I said before, that is special software.
> * **Time alignment**: Carla does, and that is the simulation step. Carla should not match Apollo's control module frequency, Apollo's control module frequency has to match Carla. If Carla is faster than Apollo, Carla has to wait for Apollo. If Apollo is faster than Carla, Apollo has to wait for Carla. If Apollo has not the ability to wait for Carla, is is not Simulation compatible. In a Simulation you want to run as fast as possible, to be able to run 10 seconds simulation time in 1 second real time.
[... 2 lines cut ...]
[... cut ...]
Actually, the situation between Apollo and Carla is not the same as in a real car. In reality, the actuator is generally faster than the control module, which contradicts the description in this issue.

As Carla is an open-source simulator for autonomous driving research, it should meet the actual requirements of an autonomous driving system. This functionality should be considered a fundamental feature.

Since the bridge, developed and maintained by Carla's official partner, is an integral part of the Carla platform, Carla should take responsibility for addressing and resolving this issue, regardless of whether the problem lies within Carla itself or the bridge.

AOOOOOA, 2025-03-14:
I encountered another issue related to Carla-Apollo integration. When running Apollo in Carla, I recorded the control commands and vehicle states at each frame during the first run. Then, in a second run (without Apollo), I replayed the same control commands and spawned the vehicle at the same location. However, the vehicle's states and trajectory in the second run showed significant deviations compared to the first run.

To reduce the deviation, I experimented with setting physics substeps in both runs, which did help mitigate the issue. Upon comparing the vehicle dynamics in the bridge (inherited from the Apollo project) with those in Carla (using the same vehicle type, Lincoln MKZ 2014), I noticed differences in **steering response** and **tire behavior**, which affect turning and acceleration. While these dynamic differences are an issue that needs to be addressed, they don't fully explain why the same control commands produce such different performances between the Apollo-controlled vehicle and the replayed vehicle.

I also observed a similar issue when using the Traffic Manager (TM) instead of Apollo in the first run #8742. However, the deviations were less severe in this case. I do believe this is a problem that should be addressed. If you need any further info, I'm very happy to provide additional information to help pinpoint the source of the issue.

Below is a comparison of the results from two runs, which, while not the most severe case, still shows significant trajectory deviation and large differences in vehicle states.

[image]

@yuxip @PatrickPromitzer @XiaoFei9704 Hi, I believe the trajectory deviation is an interesting finding that could point to underlying issues. I would appreciate it if you could take a closer look.