carla-simulator/carla #8250: Pedestrian Navigation Generation Issue in Custom Map

ronyshaji, 2024-10-07:
Hello everyone,

I'm encountering a problem with pedestrian spawning in my custom map. 
I’ve imported a new .xodr and .fbx file into CARLA and followed the tutorial (https://carla.readthedocs.io/en/latest/tuto_M_generate_pedestrian_navigation/) to create the pedestrian navigation file. 
However, when I run `generate_traffic.py,` no pedestrians are spawning.

I have a question about the placement of my .bin file. My custom map is located at **/home/going/carla/Unreal/CarlaUE4/Content/THI/Maps**. Since there wasn’t a Nav folder there, I created one and placed the .bin file inside. 

Additionally, I also put the .bin file in **/home/going/carla/Unreal/CarlaUE4/Content/Carla/Maps/Nav**, where the default CARLA town .bin files are stored.

Does anyone have any suggestions on what I should do next?

CARLA Version: 0.9.14
OS: Ubuntu 20.04

Thanks,
Rony

PatrickPromitzer, 2024-10-08:
Hi,
```generate_traffic.py``` is using spawn points to spawn the actors.
If the spawn point is slightly to low, the walker will fall through the floor.
Make sure your floor is at z=0.0 or even better a little below zero (z=-0.01) 

About file sorting, I made a folder for each kind of content type.

```
Content/Carla
Content/my_map
Content/my_vehicle
...
```

You can place everything you need for you map in the folder your created. As long as you add your map to the package setting, it should work.
I sorted it that way have make it easier to switch Carla versions. You still need to add your own actors to the Carla blueprints if you have a new Carla installation.

ronyshaji, 2024-10-08:
Hi @PatrickPromitzer 

In my case, I can spawn pedestrian in my custom map as i tired to spawn and it spawns. But the issue for me is that it is taking the spawn points (as far as I understood) from the .bin file from the Nav folder.

In the `generate_traffic.py` I think this line https://github.com/carla-simulator/carla/blob/0ce355916231d37caaf77a7328707277e7fd23ac/PythonAPI/examples/generate_traffic.py#L272 calls the location to spawn. Unfortunately it returns None and nothing is spawned.

Thats why I got doubt with the .bin file of the custom map.

PatrickPromitzer, 2024-10-08:
If you open the Unreal engine Editor, you can see red lines above the roads (default Maps).
Each red dot above the road is a spawn point.

I don't know if the .bin file does something, because for me it looks like It is a Map setting.
(I can be wrong)

ronyshaji, 2024-10-08:
@PatrickPromitzer 
I need to spawn the pedestrians instead of the vehicle. Normally for spawning the vehicle, there are spawn points available. But based on the tutorial --> https://carla.readthedocs.io/en/latest/tuto_M_generate_pedestrian_navigation/ 

We need a navigation file for the pedestrian. I guess there is issue with that file and that's why its not spawning.

PatrickPromitzer, 2024-10-08:
I didn't know that, my bad.

The.bin file should be in the same folder as your Map file.
If your map is in 
```
/home/going/carla/Unreal/CarlaUE4/Content/THI/Maps
```
the .bin file should be in
```
/home/going/carla/Unreal/CarlaUE4/Content/THI/Maps/Nav
```

ronyshaji, 2024-10-09:
@PatrickPromitzer 
Yes, I put the Nav file inside my map file. But somehow its not detecting or access it.

PatrickPromitzer, 2024-10-09:
You could try using a .bin file of a default map, and check if it can be loaded.
You could rule out the file having a problem.

Blyron, 2024-10-18:
Hello! Are you sure your bin file is valid? @ronyshaji It seems your bin file is not correct, Maybe the obj you are using to generate the bin file is not valid. 

TIP: Remember to delete the carlaCache. The cache can be found at "HOME\carlaCache" or "USERPROFILE\carlaCache", depending on OS.

ronyshaji, 2024-10-18:
@Blyron Is there any way to check the .bin file? I already cleared the carlaCache, but still can't able to use the .bin file.

Blyron, 2024-10-18:
https://youtu.be/mHiUUZ4xC9o?si=RQTcxYVURndlu3Y5&t=662 Look at this. Maybe is helpful

dhogueAltF4, 2024-11-08:
Hi @ronyshaji. I had a similar issue, but my pedestrians would spawn and then be destroyed due to collision errors. In my case, the `generate_traffic.py` was the cause. I found if I remove the `all_actors[i].go_to_location(world.get_random_location_from_navigation())` call on line 335, pedestrians could spawn and randomly walk around the custom map.