carla-simulator/carla #8254: Replace the vehicle model

CoderXuans, 2024-10-08:
My colleague provided me with an FBX file for the exterior. Can I replace only the exterior of the existing vehicle in CARLA without changing the original vehicle dynamics model?

PatrickPromitzer, 2024-10-08:
Hi,
you may want to read this guid.
https://carla.readthedocs.io/en/0.9.15/tuto_A_add_vehicle/

While I looked for the link, I found this guid.
https://carla.readthedocs.io/en/0.9.15/tuto_content_authoring_vehicles/

CoderXuans, 2024-10-08:
@PatrickPromitzer 
The tutorial method involves creating and exporting to obtain the mesh, the skeleton, and the physics asset. However, when I used a ready-made FBX file, I found that there was only one content. I converted the model drawn by my colleague in SolidWorks to FBX format using 3D Max.

CoderXuans, 2024-10-09:
Hi，My model is an unmanned ship. The tutorial includes setting up four wheels; should I set the wheels or not ?

CoderXuans, 2024-10-09:
Hi,Sorry, I forgot to mention that my model is an unmanned ship. The tutorial includes setting up four wheels; should I set the wheels or not?

PatrickPromitzer, 2024-10-09:
If you want it to drive on the ground, yes.
I am not sure if it is currently possible to have an vehicle without wheels which can drive.
The wheels are the reason the vehicles drive around.

There is an option to deactivate gravity if you want it to float. If you use this option, you can try adding it without wheels and push it around with some code.
It would be difficult with collision, but an interesting  concept.

CoderXuans, 2024-10-09:
Okay, thank you for your suggestion.

Rodrigo-Villarias, 2024-10-18:
Hello! did you manage to solve the problem, or do you still need help? greetings

CoderXuans, 2024-10-18:
Hello, I have successfully implemented the unmanned ship with four small wheels. Now I am facing a new problem. When I run the leaderboard test, I want to replace both the ego vehicle and other road participants with my recently implemented model, which is named ship_small_whe. I searched the entire project for get_world().get_actors().filter('vehicle.*') and replaced it with get_world().get_actors().filter('vehicle.myship.ship_small_whe'). However, it has not been replaced with my desired ship_small_whe. Can you help with this issue? Thank you.

Rodrigo-Villarias, 2024-10-18:
The fastest option that I would recommend is to go to the blueprint Vehicle_Factory and delete all vehicles, and add a class with only your vehicle, so in the scene it will spawn only that one, and regarding your vehicle when you go to do the manual control use “--filter (the name of your vehicle)”

CoderXuans, 2024-10-18:
[image]
My vehicle is the 29th one. Do I just need to delete all the previous 29 vehicles?

Rodrigo-Villarias, 2024-10-18:
Yes, this will ensure that only yours can spawn

CoderXuans, 2024-10-18:
It worked, thank you.

CoderXuans, 2024-10-19:
If I want to adjust the number of background vehicles, do you have any suggestions?

CoderXuans, 2024-10-19:
@Rodrigo-Villarias 
Sorry to bother you, but I have another question about the traffic lights. Even though I moved them underground and ran the evaluation code for the leaderboard, the traffic lights still appear. I moved the nearby low houses underground, and they disappeared, but when I tried to back up the map data and delete the traffic lights, clicking play caused Unreal Engine to crash.

closed 2024-11-22 by Blyron