carla-simulator/carla #8259: Trouble with trajectory generation (when world.tick() is delayed)

minhaj6, 2024-10-08:
CARLA version: 0.915
Platform/OS: Ubuntu 24.04
Problem you have experienced: Abnormal trajectory generation
What you expected to happen: Simulation to stop when world.tick() is not given
Steps to reproduce: delay between two world.tick() event
Other information (documentation you consulted, workarounds you tried):

Hi, I am trying to generate log of vehicle trajectories at every 100ms interval. What I am doing for this is I am recording the vehicle trajectories in simulation, giving a world.tick() (i have set up fixed delta time = 0.1). So I expect simulation to progress 1 step or 100ms and then I record the trajectories again. As well as visualize. 

Time settings, 

```py
        settings = world.get_settings()

        traffic_manager.set_synchronous_mode(True)
        if not settings.synchronous_mode:
            synchronous_master = True
            settings.synchronous_mode = True
            settings.fixed_delta_seconds = 0.1
        else:
            synchronous_master = False
```

Visualize and logging to csv. 
```py
    def update(frame):
        vehicle_tracker.update()    # writing to csv the current trajectories
        all_trajectories = vehicle_tracker.get_trajectories() # getting the same trajectories for visualization
        world.tick()        # simulation tick
        
        for vid, trajectory in all_trajectories.items():
            if vid not in trajectories:
                trajectories[vid], = ax.plot([], [], '-', linewidth=2, alpha=0.7)
                vehicle_points[vid], = ax.plot([], [], 'o', markersize=5)
            if len(trajectory) > 0:
                trajectories[vid].set_data(trajectory[:, 0], trajectory[:, 1])
                vehicle_points[vid].set_data(trajectory[-1, 0], trajectory[-1, 1])
        
        # Remove trajectories for vehicles that no longer exist
        for vid in list(trajectories.keys()):
            if vid not in all_trajectories:
                trajectories[vid].remove()
                vehicle_points[vid].remove()
                del trajectories[vid]
                del vehicle_points[vid]
        
        if vehicle_tracker.frame_id >= 40:
            plt.close(fig)

        return list(trajectories.values()) + list(vehicle_points.values())

    ani = FuncAnimation(fig, update, frames=None, init_func=init, blit=True, interval=100)
    plt.show()
```

But this approach doesn't work. The vehicles do not freeze in place between world.tick() -  contrary to my expectation. To verify this behavior, I intentionally put `time.sleep(2)` in between the world.tick(). That results in a quite funny behavior - 
https://youtu.be/ccYNasKs1cs?si=IbFo9b2Y2fO3AiJk

[image]
[image]
[image]

Anyway, it is clear that I am doing it wrong. How should I log vehicle trajectory at 100ms interval? Thank you.

PatrickPromitzer, 2024-10-15:
Hi,
you didn't apply the settings.

```
world_settings = world.get_settings()

world_settings.synchronous_mode = True
world_settings.fixed_delta_seconds = 0.1

world.apply_settings(world_settings)
```

minhaj6, 2024-10-15:
Ohho... Thats silly of me. 
Thank you @PatrickPromitzer for pointing that out!

closed 2024-10-15 by minhaj6