carla-simulator/carla #8263: transform not update?

daohu527, 2024-10-10:
CARLA version: ue4_dev
Platform/OS: ubuntu 20.04

I spawn a prop and use `set_transform` method to set the location, and I found I set the transform and read it immediately and found that the two were inconsistent. What is the problem?
```
        prop.set_transform(transform)
        print(f'---> 1 {transform.location}')
        print(f'---> 2 {prop.get_transform().location}')
```

PabloVD, 2024-11-08:
You need to perform a tick of the simulation after `set_transform`:
`world.tick()`
Then when you run the `get_transform()` you should get the updated transform.

PabloVD, 2025-01-15:
Thanks for sharing your issue @lijieamd. I have checked that and I reproduce the error, the transform does not get updated after the wait for tick. This is likely due to the server ticking before the setter has finished. The server may not be receiving the setter on time (due to using the AsyncCall method under the hood, rather than CallAndWait). Note that this is not always happening: in the snippet below, the first `set_transform` does not update on time the value, but the second setter does. We will discuss whether we should enforce for waiting in these setters. 

```python
import carla

client = carla.Client()
world = client.get_world()
world.wait_for_tick()
spectator = world.get_spectator()
print(spectator.get_transform())
spectator.set_transform(carla.Transform(carla.Location(x=0)))
world.wait_for_tick()
print(spectator.get_transform())
spectator.set_transform(carla.Transform(carla.Location(x=1)))
world.wait_for_tick()
print(spectator.get_transform())
```
The output:
```
Transform(Location(x=-172.196579, y=183.859116, z=27.638052), Rotation(pitch=0.000000, yaw=-45.592148, roll=0.000000))
Transform(Location(x=-172.196579, y=183.859116, z=27.638052), Rotation(pitch=0.000000, yaw=-45.592148, roll=0.000000))
Transform(Location(x=1.000000, y=0.000000, z=0.000000), Rotation(pitch=0.000000, yaw=0.000000, roll=0.000000))
```

lijieamd, 2025-01-17:
@PabloVD Thanks for your detailed explanation!
Currently I'm using double `wait_for_tick()` before `get_transform()` to solve this.

PabloVD, 2025-01-17:
Yes, I also tried that and it worked for updating the transform. Probably if there are more client calls before the `wait_for_tick()` and not just the `set_transform()`, one `wait_for_tick()` should be enough.

PabloVD, 2025-01-17:
Closing the issue for now since there are ways to overcome the issue, reopen if needed.

closed 2025-01-17 by PabloVD