carla-simulator/carla #8265: Engine crashed when making package

Sephiroth1118, 2024-10-10:
CARLA version: carla-dev build from source
Platform/OS: Ubuntu 22.04
Problem you have experienced: Engine crashed when making package
I used the digital twin tool to generate a new map and wanted to make a package version of the new map. But when I run make package, the engine crashed. I have tried to make the Town as package and that succeed. But all the map that I created by digital twin tool can not be packaged and will cause this problem. I'm new to UE engine. This is the log. Is it because my swap area is not big enough?

  LogCook: Display: Cooked packages 5864 Packages Remain 178 Total 6042
  LogCook: Display: Cooked packages 5902 Packages Remain 140 Total 6042
  LogCook: Display: Cooked packages 5902 Packages Remain 140 Total 6042
  LogCook: Display: Cooked packages 5938 Packages Remain 104 Total 6042
  LogCook: Display: Cooked packages 5938 Packages Remain 104 Total 6042
  LogCook: Display: Cooked packages 5938 Packages Remain 104 Total 6042
  LogCook: Display: Cooked packages 5938 Packages Remain 104 Total 6042
  LogCook: Display: Cook Diagnostics: OpenFileHandles=0, VirtualMemory=22478MiB
  LogCook: Display: Cook Diagnostics: OpenFileHandles=0, VirtualMemory=22478MiB
  LogCook: Display: Cooked packages 5938 Packages Remain 104 Total 6042
  LogCook: Display: Cooked packages 5938 Packages Remain 104 Total 6042
  LogCook: Display: Cooked packages 5938 Packages Remain 104 Total 6042
  LogCook: Display: Cooked packages 5938 Packages Remain 104 Total 6042
  LogCook: Display: Cooked packages 5938 Packages Remain 104 Total 6042
  LogCook: Display: Cooked packages 5938 Packages Remain 104 Total 6042
  LogCook: Display: Cooked packages 5938 Packages Remain 104 Total 6042
  LogCook: Display: Cooked packages 5938 Packages Remain 104 Total 6042
  LogCook: Display: Cooked packages 5938 Packages Remain 104 Total 6042
  LogCook: Display: Cooked packages 5938 Packages Remain 104 Total 6042
  Signal 11 caught.
  CommonUnixCrashHandler: Signal=11
  Malloc Size=65538 LargeMemoryPoolOffset=65554
  LogPlatformFile: Not using cached read wrapper
  LogInit: NumberOfWorkerThreadsToSpawn:
  LogInit:  - Number of physical cores available for the process: 12
  LogInit:  - Number of logical cores available for the process: 20
  LogInit:  - Worker number by default: 11 (you can change this number with the command line parameter '-workersthreadpool X', for a max of 26 threads)
  LogTaskGraph: Started task graph with 5 named threads and 16 total threads with 1 sets of task threads.
  LogStats: Stats thread started at 0.210626
  LogInit: NumberOfWorkerThreadsToSpawn:
  LogInit:  - Number of physical cores available for the process: 12
  LogInit:  - Number of logical cores available for the process: 20
  LogInit:  - Worker number by default: 11 (you can change this number with the command line parameter '-workersthreadpool X', for a max of 26 threads)
  LogICUInternationalization: ICU TimeZone Detection - Raw Offset: +8:00, Platform Override: ''
  LogInit: Build: ++UE4+Release-4.26-CL-0
  LogInit: Engine Version: 4.26.2-0+++UE4+Release-4.26
  LogInit: Compatible Engine Version: 4.26.0-0+++UE4+Release-4.26
  LogInit: Net CL: 0
  LogInit: OS: GenericOSVersionLabel (GenericOSSubVersionLabel), CPU: 12th Gen Intel(R) Core(TM) i7-12700, GPU: GenericGPUBrand
  LogInit: Compiled (64-bit): Sep 18 2024 16:10:37
  LogInit: Compiled with Clang: 10.0.1
  LogInit: Build Configuration: Shipping
  LogInit: Branch Name: ++UE4+Release-4.26
  LogInit: Command Line:  -Abslog="/home/UnrealEngine_4.26/Engine/Programs/AutomationTool/Saved/Cook-2024.10.10-14.50.32-CRC.txt" -Unattended "/home/Workplace/carla/Unreal/CarlaUE4/Saved/Crashes/crashinfo-CarlaUE4-pid-5506-58A221986D56491AB6B6720E854DA532/"
  LogInit: Base Directory: /home/qiujing/UnrealEngine_4.26/Engine/Binaries/Linux/
  LogInit: Allocator: binned2
  LogInit: Installed Engine Build: 1
  LogInit: Presizing for max 100000 objects, including 0 objects not considered by GC, pre-allocating 0 bytes for permanent pool.
  LogConfig: Applying CVar settings from Section [/Script/Engine.StreamingSettings] File [/home/qiujing/.config/Epic/CrashReportClient/Saved/Config/Linux/Engine.ini]
  LogConfig: Setting CVar [[s.MinBulkDataSizeForAsyncLoading:131072]]
  LogConfig: Setting CVar [[s.AsyncLoadingThreadEnabled:0]]
  LogConfig: Setting CVar [[s.EventDrivenLoaderEnabled:1]]
  LogConfig: Setting CVar [[s.WarnIfTimeLimitExceeded:0]]
  LogConfig: Setting CVar [[s.TimeLimitExceededMultiplier:1.5]]
  LogConfig: Setting CVar [[s.TimeLimitExceededMinTime:0.005]]
  LogConfig: Setting CVar [[s.FlushStreamingOnExit:1]]
  LogInit: Object subsystem initialized
  LogConfig: Applying CVar settings from Section [ConsoleVariables] File [/home/qiujing/.config/Epic/CrashReportClient/Saved/Config/Linux/Engine.ini]
  [2024.10.10-06.57.50:086][  0]LogInit: Unix hardware info:
  [2024.10.10-06.57.50:086][  0]LogInit:  - we are the first instance of this executable
  [2024.10.10-06.57.50:086][  0]LogInit:  - this process' id (pid) is 5928, parent process' id (ppid) is 5506
  [2024.10.10-06.57.50:086][  0]LogInit:  - we are not running under debugger
  [2024.10.10-06.57.50:086][  0]LogInit:  - machine network name is 'qiujing'
  [2024.10.10-06.57.50:093][  0]LogInit:  - user name is 'qiujing' (qiujing)
  [2024.10.10-06.57.50:093][  0]LogInit:  - we're logged in locally
[... 70 lines cut ...]
  [2024.10.10-06.57.51:325][  0]CrashReportCoreLog: Sending HTTP request: https://datarouter.ol.epicgames.com/datarouter/api/v1/public/data?AppID=CrashReporter&AppVersion=4.26.2-0%2B%2B%2BUE4%2BRelease-4.26&AppEnvironment=Release&UploadType=crashreports&UserID=b21652a9827644048b1d4b88aa233e49-000003e8%7C%7Cb21652a9827644048b1d4b88aa233e49
  [2024.10.10-06.57.53:432][  0]CrashReportCoreLog: OnProcessRequestComplete(), State=SendingFiles bSucceeded=1
  [2024.10.10-06.57.53:432][  0]CrashReportCoreLog: State change from SendingFiles to SendingFiles
  [2024.10.10-06.57.53:432][  0]CrashReportCoreLog: All uploads done
  [2024.10.10-06.57.53:432][  0]CrashReportCoreLog: State change from SendingFiles to Finished
  [2024.10.10-06.57.54:268][  0]CrashReportCoreLog: Final state (Receiver) = Finished
  [2024.10.10-06.57.54:268][  0]CrashReportCoreLog: Final state (Receiver) = Unknown UploadState value
  [2024.10.10-06.57.54:268][  0]LogCore: Engine exit requested (reason: CrashReportClientApp RequestExit)
  [2024.10.10-06.57.54:268][  0]LogExit: Preparing to exit.
  [2024.10.10-06.57.54:269][  0]LogExit: Object subsystem successfully closed.
  [2024.10.10-06.57.54:284][  0]LogModuleManager: Shutting down and abandoning module HTTP (12)
  [2024.10.10-06.57.54:284][  0]LogHttp: Display: cleaning up 0 outstanding Http requests.
  [2024.10.10-06.57.54:313][  0]LogModuleManager: Shutting down and abandoning module SSL (11)
  [2024.10.10-06.57.54:314][  0]LogModuleManager: Shutting down and abandoning module CrashDebugHelper (8)
  [2024.10.10-06.57.54:314][  0]LogModuleManager: Shutting down and abandoning module CoreUObject (6)
  [2024.10.10-06.57.54:314][  0]LogModuleManager: Shutting down and abandoning module PakFile (4)
  [2024.10.10-06.57.54:314][  0]LogModuleManager: Shutting down and abandoning module RSA (3)
  [2024.10.10-06.57.54:315][  0]LogExit: Exiting.
  Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Took 507.315446s to run UE4Editor, ExitCode=139
ERROR: Cook failed.
       (see /home/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt for full exception trace)
AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)
RunUAT ERROR: AutomationTool was unable to run successfully.
make: *** [Util/BuildTools/Linux.mk:17: package] Error 25

Sephiroth1118, 2024-10-11:
And when I co-sim with sumo, the UE will also crashed when calling GetCarlaMapTile() in LargeMapManager.cpp. It seems the engine expected some missing data. And this error only happened with map generated by digital twin tool.
LoginId:b21652a9827644048b1d4b88aa233e49-000003e8
EpicAccountId:

Assertion failed: Tile [File:/home/Workplace/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/MapGen/LargeMapManager.cpp] [Line: 619]

Sephiroth1118, 2024-10-11:
LoginId:b21652a9827644048b1d4b88aa233e49-000003e8
EpicAccountId:

Assertion failed: Tile [File:/home/Workplace/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/MapGen/LargeMapManager.cpp] [Line: 619]

libUE4Editor-Core.so!FGenericPlatformMisc::RaiseException(unsigned int) [/home/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/GenericPlatform/GenericPlatformMisc.cpp:472]
libUE4Editor-Core.so!FOutputDevice::LogfImpl(char16_t const*, ...) [/home/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/Misc/OutputDevice.cpp:61]
libUE4Editor-Core.so!AssertFailedImplV(char const*, char const*, int, char16_t const*, __va_list_tag*) [/home/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/Misc/AssertionMacros.cpp:104]
libUE4Editor-Core.so!FDebug::CheckVerifyFailedImpl(char const*, char const*, int, char16_t const*, ...) [/home/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/Misc/AssertionMacros.cpp:458]
libUE4Editor-Carla.so!ALargeMapManager::GetCarlaMapTile(ULevel*) [/home/Workplace/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/MapGen/LargeMapManager.cpp:619]
libUE4Editor-Carla.so!ALargeMapManager::OnLevelRemovedFromWorld(ULevel*, UWorld*) [/home/Workplace/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/MapGen/LargeMapManager.cpp:150]
libUE4Editor-Carla.so!TBaseUObjectMethodDelegateInstance<false, ALargeMapManager, void (ULevel*, UWorld*), FDefaultDelegateUserPolicy>::ExecuteIfSafe(ULevel*, UWorld*) const [/home/UnrealEngine_4.26/Engine/Source/Runtime/Core/Public/Delegates/DelegateInstancesImpl.h:609]
libUE4Editor-Engine.so!TMulticastDelegate<void (ULevel*, UWorld*), FDefaultDelegateUserPolicy>::Broadcast(ULevel*, UWorld*) const [/home/UnrealEngine_4.26/Engine/Source/Runtime/Core/Public/Delegates/DelegateSignatureImpl.inl:955]
libUE4Editor-Engine.so!UEngine::LoadMap(FWorldContext&, FURL, UPendingNetGame*, FString&) [/home/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/UnrealEngine.cpp:12731]
libUE4Editor-Engine.so!UEngine::Browse(FWorldContext&, FURL, FString&) [/home/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/UnrealEngine.cpp:12338]
libUE4Editor-Engine.so!UEngine::TickWorldTravel(FWorldContext&, float) [/home/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/UnrealEngine.cpp:12536]
libUE4Editor-UnrealEd.so!UEditorEngine::Tick(float, bool) [/home/UnrealEngine_4.26/Engine/Source/Editor/UnrealEd/Private/EditorEngine.cpp:1689]
libUE4Editor-UnrealEd.so!UUnrealEdEngine::Tick(float, bool) [/home/UnrealEngine_4.26/Engine/Source/Editor/UnrealEd/Private/UnrealEdEngine.cpp:423]
UE4Editor!FEngineLoop::Tick() [/home/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:4830]
UE4Editor!GuardedMain(char16_t const*) [/home/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/Launch.cpp:171]
libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/UnrealEngine_4.26/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
libc.so.6!UnknownFunction(0x29d8f)
libc.so.6!__libc_start_main(+0x7f)
UE4Editor!_start()

IrfanAli91170, 2025-02-21:
Yes, i have also the same issue here.

IrfanAli91170, 2025-02-28:
Since then, have you managed to resolve this problem?
Because I'm having exactly the same problem as you. 

Because of this, I'm stuck in my work.

IrfanAli91170, 2025-03-03:
Thank you for your answer. I'm actually stuck in the step "make package" when the digital twin tool generate a custom map, and i need this very urgently for my company project, and i have tried all multiple and different solutions.

Finally, it remains the same......

IrfanAli91170, 2025-03-03:
https://github.com/carla-simulator/carla/issues/8696

Sephiroth1118, 2025-03-04:
I'm sorry that I have not figured it out either. A possible way to bypass it is do not use the map generated by digital twin tool. You can create a fbx file using xodr file with LibOpenDrive and assimp, and then import it into UE editor. The map will only have road mesh and you need to decorate it manually with some trees or buildings. I think you can copy these assets from the digital twin map. It may take some time but this map created by hand should be alright with "make package".

IrfanAli91170, 2025-03-04:
Thank you for your answer. It was the first step in my project work, but all my team members want a realistic and exact environment to simulate (a real city for example). Even, i have already tried the manual work, but they want that i should use the digital twin tool to make it happen.

Sephiroth1118, 2025-03-05:
That's a pity. Maybe you can find someone and pay for the map.

IrfanAli91170, 2025-03-05:
Yeah, I know. I think i will use the ue4 editor to play Python scripts directly in the generated map, if i don't have the choice.

Blyron, 2025-09-22:
This has been fixed recently

closed 2025-09-22 by Blyron