carla-simulator/carla #8274: Linking error with libproj.a during make launch in Carla 0.9.15 on Ubuntu 20.04

SteveLi-0, 2024-10-13:
Carla Version: 0.9.15
OS: Ubuntu 20.04
Compiler: Clang 10.0.1
Issue: Linking error with libproj.a during the make launch step

I am building Carla 0.9.15 on Ubuntu 20.04. While make PythonAPI succeeds without errors, I encounter several undefined symbol errors during the make launch step, particularly related to osgeo::proj::CPLJSonStreamingWriter. Here are the errors:

```
ld.lld: error: undefined symbol: osgeo::proj::CPLJSonStreamingWriter::AddObjKey(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)
referenced by singleoperation.cpp singleoperation.cpp.o:(osgeo::proj::operation::OperationMethod::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /path/to/libproj.a
```

and many other similar undefined symbol errors.

Steps to Reproduce:
Build Carla 0.9.15 on Ubuntu 20.04.
Use Clang 10.0.1 for compilation.
Run make PythonAPI successfully (PROJ_VERSION=proj-7.2.1). 
Run make launch and encounter the linking errors.

```
[23/29] Compile CarlaUE4.cpp
4 warnings generated.
8 warnings generated.
4 warnings generated.
4 warnings generated.
4 warnings generated.
4 warnings generated.
4 warnings generated.
4 warnings generated.
6 warnings generated.
3 warnings generated.
10 warnings generated.
15 warnings generated.
26 warnings generated.
14 warnings generated.
32 warnings generated.
[24/29] Link (lld) libUE4Editor-Carla.so
[25/29] Link (lld) libUE4Editor-StreetMapRuntime.so
[26/29] Link (lld) libUE4Editor-CarlaUE4.so
[27/29] Link (lld) libUE4Editor-StreetMapImporting.so
[28/29] Link (lld) libUE4Editor-CarlaTools.so
ld.lld: error: undefined symbol: osgeo::proj::CPLJSonStreamingWriter::AddObjKey(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)
>>> referenced by singleoperation.cpp
>>>               singleoperation.cpp.o:(osgeo::proj::operation::OperationMethod::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by singleoperation.cpp
>>>               singleoperation.cpp.o:(osgeo::proj::operation::OperationParameterValue::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by singleoperation.cpp
>>>               singleoperation.cpp.o:(osgeo::proj::operation::OperationParameterValue::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by singleoperation.cpp
>>>               singleoperation.cpp.o:(osgeo::proj::operation::OperationParameterValue::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by singleoperation.cpp
>>>               singleoperation.cpp.o:(osgeo::proj::operation::OperationParameterValue::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced 114 more times

ld.lld: error: undefined symbol: osgeo::proj::CPLJSonStreamingWriter::Add(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)
>>> referenced by singleoperation.cpp
>>>               singleoperation.cpp.o:(osgeo::proj::operation::OperationMethod::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by singleoperation.cpp
>>>               singleoperation.cpp.o:(osgeo::proj::operation::OperationParameterValue::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
[... 132 lines cut ...]
>>> referenced by datum.cpp
>>>               datum.cpp.o:(osgeo::proj::datum::DatumEnsemble::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by common.cpp
>>>               common.cpp.o:(osgeo::proj::common::ObjectDomain::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by common.cpp
>>>               common.cpp.o:(osgeo::proj::common::ObjectDomain::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by common.cpp
>>>               common.cpp.o:(osgeo::proj::common::ObjectUsage::baseExportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by common.cpp
>>>               common.cpp.o:(osgeo::proj::common::ObjectUsage::baseExportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a

ld.lld: error: undefined symbol: osgeo::proj::CPLJSonStreamingWriter::CPLJSonStreamingWriter(void (*)(char const*, void*), void*)
>>> referenced by io.cpp
>>>               io.cpp.o:(osgeo::proj::io::JSONFormatter::Private::Private()) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a

ld.lld: error: undefined symbol: osgeo::proj::CPLJSonStreamingWriter::Add(long)
>>> referenced by common.cpp
>>>               common.cpp.o:(osgeo::proj::common::UnitOfMeasure::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by metadata.cpp
>>>               metadata.cpp.o:(osgeo::proj::metadata::Identifier::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/lxx/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [Makefile:315: CarlaUE4Editor] Error 6
make[1]: Leaving directory '/home/lxx/carla/carla/Unreal/CarlaUE4'
make: *** [Util/BuildTools/Linux.mk:8: launch] Error 2
```

SteveLi-0, 2024-10-15:
I have resolved the issue. The problem was similar to [#8006](https://github.com/carla-simulator/carla/issues/8006#issuecomment-2262227466) and [#6474](https://github.com/carla-simulator/carla/issues/6474) so I followed the information provided there and modified setup.sh, but it resulted in an error when running make launch.

The correct solution was to change the `PROJ_VERSION`  to `9.3.0 `on my system, without any additional modifications.

closed 2024-10-15 by SteveLi-0

preesent, 2025-05-27:
I meet the same quesion. I refer to #8006 abd set PROJ_VERSION  to 9.3.0, but don't solve it.