carla-simulator/carla #8279: black window after run ./CarlaUE4.sh

jxc616, 2024-10-14:
CARLA version: 0.9.15
Platform/OS: Ubuntu 20.04.06 LTS

Problem you have experienced:
Hi everyone, 
i have recently downloaded carla from package installation, After following the tutorial https://carla.readthedocs.io/en/0.9.15/start_quickstart/#before-you-begin, when i try to run ./CarlaUE4.sh, the launch window comes out but with nothing, just black.
[image]
Thank a lot for your help in advance.

PatrickPromitzer, 2024-10-15:
Hi,
what happens if you use ```./CarlaUE4.sh``` without the parameters?

jxc616, 2024-10-15:
Hi, thanks for your reply, i have tried without the parameters, but still doesn't work.
btw, wenn i start the bash, the nividia-smi shows the GPU is working, with a memory usage of aroung 399MB, and then after a while, it shows like Carla-shipping is not responding and let me choice ethier force quit or wait.
I am not sure if it is due to i have installed it in external hard drive.

PatrickPromitzer, 2024-10-16:
Hi,
I am not sure if it makes a different if it is installed on an external drive, but it could be, that it takes to long to load.

The black screen itself is normal, but after some time (normally under 20 seconds), it should get an picture.
If nothing else gives an error, it looks like it takes too long to load.

jxc616, 2024-10-16:
Hi, thanks for your advise, I think some errors occurred, because after serveal seconds, the window shows "Program not responding", but unfortunately there are no more hints pointing to where the error occurred :(

PatrickPromitzer, 2024-10-16:
Hi,
It doesn't have to be an error because of the "Program not responding" message.
The message appears if the GUI doesn't update, and this can happen for many reasons.

One reason could be, that your hardware is not powerful enough and takes longer.
Another reason could be, that your external drive is too slow.

You can try starting Carla and keep it open for a few minutes. Maybe it works if you wait long enough.

jxc616, 2024-10-16:
Hi Patrick, 
thanks a lot for your patient guidance, after I clicked the wait button several times, finally the scenario showed up!
Best wishes!

closed 2024-10-16 by jxc616