carla-simulator/carla #8283: Problem using osm world renderer to generate the map

JarvisLee0423, 2024-10-15:
Hi, I encountered a problem using the OSM world renderer to generate the map from the OSM.

My Carla version is 0.9.15
My system version is Unbuntu 22.04

I follow the tutorial in [https://carla.readthedocs.io/en/latest/adv_digital_twin/](url), and download the OSM zip for Colorado, and then using the ./build.sh in libosmscout-source to create the map folder and assets, after successfully executed make osmrenderer.
[image]
Through the above picture, I thought the ./build.sh had been successfully executed. But I noticed that there are some errors in the output txt log files, but I do not know why the errors occur.
[colorado-latest.txt](https://github.com/user-attachments/files/17381948/colorado-latest.txt)

After that, I using the make launch to launch the carla and open the UW_OnRoadMainWidge plugin, and input the path to the generated map assets:
[image]
Then I run the osm world renderer by using the ./osmworldrenderer executable file:
[image]
Then I click the Config button, the osm world renderer start to compile the map files:
[image]
Then I click the Render button, the osm world renderer gave me the following error:
[image]
And the UW_OnRoadMainWidge still remain the following status:
[image]

Can any one tell me why the osm world renderer gave me the above error? And are there any methods to solve the error?

rohanNkhaire, 2024-10-16:
@JarvisLee0423 Hey, I encountered the same issue. I resolved it by following the instructions [here](https://github.com/carla-simulator/carla/issues/6913#issuecomment-2380016146). I hope that helps.

JarvisLee0423, 2024-10-24:
> @JarvisLee0423 Hey, I encountered the same issue. I resolved it by following the instructions [here](https://github.com/carla-simulator/carla/issues/6913#issuecomment-2380016146). I hope that helps.

Thanks for your kind reply! I successfully used the instructions in the link you shared to generate the map from the example OSM, like the images below:
[image]
However, I still face a problem, which is that, after generating the map, CARLA will give me two map assets, one named Colorado and another named Colorado_Tile_0_0, like the image illustrated below:
[image]
It seems that all the geometric assets will be saved in Colorado_Tile_0_0 map, and all the world parameters assets will be saved in Colorado map, like the world light setting and playable setting. In this case, if I load the Colorado_Tile_0_0 map, the windows will be all dark, till I put a direct light component inside the map manually. This situation is shown in images below:
Directly load Colorado_Tile_0_0 map:
[image]
After I put a direct light component manually:
[image]
Meanwhile, the error will occur if I push the play button, cause the playable component is not in the Colorado_Tile_0_0 map.
After that, if I load the Colorado map, the window will only show the default sky and does not contain any buildings or roads, like the image demonstrated below:
[image]

Therefore, I would like to know have you ever faced this problem before? If yes, could you tell me how to fix it? Or are there any methods to merge these two maps? It seems a little bit complex and stupid if I need to add all the world parameters settings (light, playable, etc) for all maps generated by OSMWorldRenered manually.

joeyleehk, 2025-02-03:
> erated by OSMWorldRenered.

The same issue I also faced, I didn't find a feasible way to resolve this.
But I check the blueprint of OSMWorldRenered, it seems that no 3d models are generated. Everything generated in the Tile_0_0 map is the combinations of existed assets. 
Any thoughts to fix this issues? Thanks a lot.

IrfanAli91170, 2025-02-19:
> [@JarvisLee0423](https://github.com/JarvisLee0423) Hey, I encountered the same issue. I resolved it by following the instructions [here](https://github.com/carla-simulator/carla/issues/6913#issuecomment-2380016146). I hope that helps.

Hello, i have the same issue here, What have you done to solve this?

Because i also tried these instructions given, and nothing happen, it remains the same issue.

IrfanAli91170, 2025-02-19:
> Hello, i have the same issue here, What have you done to solve this?
>
> Because i also tried these instructions given, and nothing happen, it remains the same issue.

> Hi, I encountered a problem using the OSM world renderer to generate the map from the OSM.
>
> My Carla version is 0.9.15 My system version is Unbuntu 22.04
>
> I follow the tutorial in [...]

IrfanAli91170, 2025-02-19:
> Hi, I encountered a problem using the OSM world renderer to generate the map from the OSM.
>
> My Carla version is 0.9.15 My system version is Unbuntu 22.04
>
> I follow the tutorial in [...]

Hello, i also have the same issue here. I carefully follow the instructions but it remains the same. 

What have you done to solve this issue, please ? 

Thank you in advance !