carla-simulator/carla #8284: Cannot 'make launch' when building carla 0.9.15-dev fails with error 6

wenbralliance, 2024-10-16:


wenbralliance, 2024-10-16:
carla: 0.9.15-dev
platform: win11

D:\carla_unreal\carla>make launch
-[BuildUE4Plugins]: [Batch params]: --build --all
HEAD is now at 260273d Correct descriptor
-[BuildUE4Plugins]: STREETMAP has been successfully installed in "D:\carla_unreal\carla\Unreal\CarlaUE4\Plugins\"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --generator "" --all
-[Setup]: Asynchronous jobs:  32
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Generator:          "Visual Studio 16 2019"
-[Setup]: Install directory:  "D:\carla_unreal\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "D:\carla_unreal\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "D:\carla_unreal\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
已复制         1 个文件。
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "D:\carla_unreal\carla\Build\" --zlib-install-dir "D:\carla_unreal\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "D:\carla_unreal\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "D:\carla_unreal\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "D:\carla_unreal\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "D:\carla_unreal\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_gtest]: Not cloning Google Test because already exists a folder called "gtest-src".
    -[install_gtest]: Generating build...
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.22631.
CMake Deprecation Warning at googlemock/CMakeLists.txt:42 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at googletest/CMakeLists.txt:49 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Warning (dev) at googletest/cmake/internal_utils.cmake:239 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  googletest/CMakeLists.txt:84 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done (0.2s)
-- Generating done (0.4s)
-- Build files have been written to: D:/carla_unreal/carla/Build/gtest-src/build
    -[install_gtest]: Building...
用于 .NET Framework 的 Microsoft (R) 生成引擎版本 16.11.2+f32259642
版权所有(C) Microsoft Corporation。保留所有权利。

[... 281 lines cut ...]
  C:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\um\winnt.h(620): warning C4005: “TEXT”: 宏重定义
  D:\carla_unreal\UnrealEngine-carla\Engine\Source\Runtime\Core\Public\HAL/Platform.h(1082): note: 参见“TEXT”的前一个定 义
  D:\carla_unreal\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Vehicle\WheeledVehicleMovementComponentNW.h(120): warning C4996: 'UWheeledVehicleMovementComponent': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla_unreal\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Vehicle\WheeledVehicleMovementComponentNW.h(148): warning C4996: 'UVehicleWheel': PhysX is deprecated. Use the UChaosVehicleWheel from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla_unreal\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(72): warning C4996: 'AWheeledVehicle': PhysX is deprecated. Use the AWheeledVehiclePawn from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla_unreal\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(188): warning C4996: 'UWheeledVehicleMovementComponent4W': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla_unreal\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Sensor\Sensor.cpp(30): warning C4458: “Episode”的声明隐藏了类成员
[... cut ...] before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla_unreal\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(188): warning C4996: 'UWheeledVehicleMovementComponent4W': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
make: *** [CarlaUE4Editor] 错误 6

muhammeddincmdx, 2025-01-19:
did you solve it?