carla-simulator/carla #8285: Replace background vehicles

CoderXuans, 2024-10-17:
How can I replace the remaining vehicles on the road with a specified model when running the evaluation of the leaderboard? I modified get_actors().filter("*vehicle*") in the project to use my specified model instead of ("*vehicle*"), but there is no change in the simulation.

glopezdiest, 2024-10-18:
Hello @CoderXuans. If I understand correctly, you want to change the vehicles part of the background activity? The ones that go around the vehicles without being a specific scenario.

if so, those are governed by the [background_activity.py](https://github.com/carla-simulator/scenario_runner/blob/leaderboard-2.0/srunner/scenarios/background_activity.py) scenario file. Specifically, the spawning is done at lines _2130_ and _2152_.

Closing the issue, but feel free to reopen it if there are unresolved issues

closed 2024-10-18 by glopezdiest

CoderXuans, 2024-10-26:
@glopezdiest 
Hi,I found a method to change the vehicle, and I would like to consult about the removal of traffic signals. I plan to change the map to a water surface and want to remove the traffic lights. When I disable the traffic light layer and click Play in UE4, the hidden traffic lights still appear. Directly deleting the traffic lights causes UE4 to crash when I click Play. How can I safely and effectively remove the traffic lights?