carla-simulator/carla #8289: problem of make PythonAPI

SYSUCtlLab2024, 2024-10-17:
CARLA version: 0.9.15
Platform/OS: windows 11
Problem you have experienced: --build-dir "D:\carla\Build\" --toolset msvc-14.2 --version 1.80.0 -j 20
The system cannot find the specified batch label - arg-parse
Full log： (CARLA) D:\carla>make PythonAPI
-[BuildUE4Plugins]: [Batch params]: --build --all
error: RPC failed; curl 28 Recv failure: Connection was reset
fatal: expected flush after ref listing
HEAD is now at 260273d Correct descriptor
-[BuildUE4Plugins]: STREETMAP has been successfully installed in "D:\carla\Unreal\CarlaUE4\Plugins\"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --generator "" --all
-[Setup]: Asynchronous jobs:  20
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Generator:          "Visual Studio 16 2019"
-[Setup]: Install directory:  "D:\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "D:\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "D:\carla\Build\" --zlib-install-dir "D:\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "D:\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "D:\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "D:\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "D:\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "D:\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "D:\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "D:\carla\Build\recast-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "D:\carla\Build\" --toolset msvc-14.2 --version 1.80.0 -j 20
The system cannot find the specified batch label - arg-parse

-[Setup]:
   Ok, and error ocurred, don't panic
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] Error 1

xminjun, 2024-11-05:
same problem，How to solve it

einarin-pattern, 2024-11-08:
I encountered this problem. It occurs when `Util/InstallersWin/install_boost.bat` is checked out with Unix line endings instead of Windows line endings. Check your Git settings, or manually convert that file to have Windows line endings (CRLF).

SYSUCtlLab2024, 2024-11-14:
thx, I'll try it.

Sama-Yousef, 2024-12-01:
great, it solved my issue

Blyron, 2025-04-04:
If you are okay to close this issue, I will close it